MarlinFirmware/Marlin #2799: FWRETRACT not working in 1.1.0-RC2

m0n5t3r, 2015-11-25:
If I enable it:

* extrude commands do nothing, return "ok"
* retraction commands raise the Z axis o.O
* the extruder stepper stays disabled (I can turn it by hand easily)

The motherboard is MKS_BASE (RAMPS_EFB with an additional extruder, basically).

KiteLab, 2015-12-08:
If FWRETRACT is not defined, your system is working normal?
A link to your Configuration.h, Configuration_adv.h and your used pin-file could be helpfull.
Do you use a dual extruder configuration?

m0n5t3r, 2015-12-08:
Yes, the system works just fine without FWRETRACT. It's a single extruder setup.

Configuration: https://gist.github.com/m0n5t3r/cd5e664113cd72e151de
Pins: https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/pins_MKS_BASE.h

KiteLab, 2015-12-08:
I could not reproduce the errors with your configuration. (Who wonders if FWRETRACT is not defined)
So i tested a bit how you could have produced the problems.

* extrude commands do nothing, return "ok"

If you send a `G1 E2` the first time it will move a bit. If you send it the second time nothing will happen, because, normally, we are in absolute mode and we are already at 2 - we only see an OK but no move (and no stepper activation). The hosts use a trick. They send G91 before the G1 E2, and a G90 behind, to go in and out relative mode. (just a guess)
* retraction commands raise the Z axis o.O

If you defined RETRACT_ZLIFT > 0 that is the wanted behaviour. With a G11 they will go down the same amount.
* the extruder stepper stays disabled (I can turn it by hand easily)

Maybe you defined DISABLE_E true and a very low DEFAULT_STEPPER_DEACTIVE_TIME.

m0n5t3r, 2015-12-08:
going to try again (only changing the FWRETRACT value); back when I got
this I tried with the extruder in relative mode (M83), and with whatever
octoprint is doing when you press extrude / retract

m0n5t3r, 2015-12-08:
all right, tested again: with the source I had when I reported this[1], the behaviour is what I described earlier (extrude 5mm => no effect, retract 5mm => z raises by 5mm) 

I switched to the RC branch and updated to the latest code[2] and regular extruder moves work, but G10 and G11 don't do anything (tried both by hand and with an actual print - "use firmware retraction" in Slic3r).

[1] FIRMWARE_NAME:Marlin 1.1.0 RCBugFix-RC2-18-g4b57b22 SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:m0n5t3r's i3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000

[2] FIRMWARE_NAME:Marlin 1.1.0 RC-RC2-22-g5a54204 SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:m0n5t3r's i3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000

KiteLab, 2015-12-09:
Is your EEPROM active?

m0n5t3r, 2015-12-09:
yes

KiteLab, 2015-12-09:
Then you may have a initialization problem.
You could try a M502 to get the values from your configuration. And a M500 to store them in the EEPROM.
That could explain different behaviour with unchanged code in the FWRETRACT area.

KiteLab, 2015-12-09:
Or expedite set values with the menu or M207, M208 and sore them in the EEPROM.

KiteLab, 2015-12-09:
Somewhere in the code could be a mm/min - mm/s - factor 60 problem
In the config RETRACT_FEEDRATE claims to be in mm/s.
`retract_feedrate = RETRACT_FEEDRATE;` init.
`    EEPROM_WRITE_VAR(i, retract_feedrate);`
`      EEPROM_READ_VAR(i, retract_feedrate);`
`    MENU_ITEM_EDIT(float3, MSG_CONTROL_RETRACTF, &retract_feedrate, 1, 999);`
`    SERIAL_ECHOPAIR(" F", retract_feedrate * 60);` output M503 in mm/min matching M207
`    if (code_seen('F')) retract_feedrate = code_value() / 60;` input M207 in mm/min
`      feedrate = retract_feedrate * 60;` feedrate in mm/min - OK

So everything is ok. 
In the g-codes every F-parameter should be in mm/min.
In the examples for WITBOX and Hephestos the default value with 80*60 could be a bit to high.

thinkyhead, 2016-02-24:
This appears solved. Please re-open if any issues remain.

closed 2016-02-24 by thinkyhead