MarlinFirmware/Marlin #2800: SD support for REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER?

gabrielcr78, 2015-11-25:
Hello, i just started doping some testing with this new version of Marlin 1.1.0 rc2 on a Prusa i3 with ramps 1.4

with the previous version of Marlin i had no issue to navigate in the LCD with my SD card, in fact, if i flash the previous version again it recognizes the SD card immediately, however with this new version, it looks like SD card is not supported?
I un commented the line for this display just like in my previous Marlin version:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

but not sure if now there is something else tht i have to configure for SD support?

thanks!

Gabe

KiteLab, 2015-11-25:
>//#define ULTRA_LCD  //general LCD support, also 16x2
//#define DOGLCD  // Support for SPI LCD 128x64 (Controller ST7565R graphic Display Family)
//#define SDSUPPORT // Enable SD Card Support in Hardware Console
// Changed behaviour! If you need SDSUPPORT uncomment it!

gabrielcr78, 2015-11-25:
uhhh thanks a bunch!

closed 2016-02-22 by thinkyhead