MarlinFirmware/Marlin #2804: CoreXY home XY problem

selimmeric, 2015-11-26:
Hi.. when I did Home X, and  Home Y its working and stoping 
"G28 X" => working
"G28 Y" => working
"G28 XY" => stucking near home position, not stopping ...

Wurstnase, 2015-11-26:
If you are using max endstops and the old release, that is a known issue. Please try then the RC.

selimmeric, 2015-11-26:
#define STRING_VERSION "1.0.2"

I am not using max endstop and 
there are a code in configration.h file so its impossiable to use maxendstops I think .. 

Line ~280: #define DISABLE_MAX_ENDSTOPS 

Line ~284:#if defined(COREXY) && !defined(DISABLE_MAX_ENDSTOPS)
Line ~285:  #define DISABLE_MAX_ENDSTOPS
Line ~286:#endif

I will resarch more ..

closed 2015-11-26 by selimmeric

Wurstnase, 2015-11-26:
Maybe you should use ```G28 X Y``` instead of ```G28 XY```

selimmeric, 2015-11-26:
I try it with RC .. yes it work... I need more time to transfer my old  configration to new release...