MarlinFirmware/Marlin #2807: Please help fix this line since the update it broken

Potasio101, 2015-11-27:
Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\Marlin_main.cpp:2667:36: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]

         LCD_MESSAGEPGM(MACHINE_NAME" "MSG_OFF".");

                                    ^

In file included from sketch\dogm_lcd_implementation.h:38:0,

                 from sketch\ultralcd.cpp:39:

sketch\dogm_font_data_marlin.h:12:25: fatal error: utility/u8g.h: No such file or directory

 #include <utility/u8g.h>

                         ^

compilation terminated.

exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Wackerbarth, 2015-11-27:
I don’t know what version of Marlin you are attempting to compile.

The only line in the current version of Marlin_main.cpp that is similar to that line is #4178.

Please switch to either the RCBugFix branch of this repository or to the master branch (in the MarlinDev repository)

> On Nov 27, 2015, at 11:32 AM, Potasio101 <notifications@github.com> wrote:
> 
> Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> sketch\Marlin_main.cpp:2667:36: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]
> 
>      LCD_MESSAGEPGM(MACHINE_NAME" "MSG_OFF".");

Potasio101, 2015-11-27:
Marlin 1.6.6

Well i just delete the Library in My document files from the update i receibe

that fix everything

Wackerbarth, 2015-11-27:
@Potasio101 -- There is no Marlin 1.6.6.

But I am glad to learn that you have gotten to a configuration that you can use.

KiteLab, 2015-11-27:
He tried to compile `Release`.

closed 2015-12-02 by Wackerbarth

AlfredoCJ, 2016-06-30:
Hi,
  I had the same error, I resolved it with blank space... like this:

> LCD_MESSAGEPGM(MACHINE_NAME " " MSG_OFF ".");

Bye

Blue-Marlin, 2016-06-30:
a) Its not an error but a warning.
b) It's fixed since a long long time - but not in the antiquarian old branches.