MarlinFirmware/Marlin #2810: expected initializer before 'MARLIN_PROGMEM'

SLYSCORPIO, 2015-11-28:
I have been searching and can not find a solution to this error message. I have a Prusa Mendel with a Rambo. Any help will be appreciated. 

Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Marlin.h:23:0,

                 from sketch\BlinkM.cpp:5:

sketch\pins.h:2366:0: warning: "X_MS1_PIN" redefined [enabled by default]

 #define X_MS1_PIN 40

 ^

In file included from sketch\Marlin.h:23:0,

                 from sketch\BlinkM.cpp:5:

sketch\pins.h:7:0: note: this is the location of the previous definition

 #define X_MS1_PIN -1

 ^

In file included from sketch\Marlin.h:23:0,

                 from sketch\BlinkM.cpp:5:

sketch\pins.h:2367:0: warning: "X_MS2_PIN" redefined [enabled by default]

 #define X_MS2_PIN 41

 ^

In file included from sketch\Marlin.h:23:0,

                 from sketch\BlinkM.cpp:5:

sketch\pins.h:8:0: note: this is the location of the previous definition

 #define X_MS2_PIN -1

 ^

In file included from sketch\Marlin.h:23:0,

                 from sketch\BlinkM.cpp:5:

sketch\pins.h:2374:0: warning: "Y_MS1_PIN" redefined [enabled by default]

 #define Y_MS1_PIN 69

 ^

In file included from sketch\Marlin.h:23:0,

                 from sketch\BlinkM.cpp:5:

sketch\pins.h:9:0: note: this is the location of the previous definition

 #define Y_MS1_PIN -1

 ^

In file included from sketch\Marlin.h:23:0,

                 from sketch\BlinkM.cpp:5:

sketch\pins.h:2375:0: warning: "Y_MS2_PIN" redefined [enabled by default]

[... 397 lines cut ...]

ConfigurationStore.cpp:434: error: 'tmp1' was not declared in this scope

     axis_steps_per_unit[i] = pgm_read_float(&tmp1[i]);

                                              ^

ConfigurationStore.cpp:435: error: 'tmp2' was not declared in this scope

     max_feedrate[i] = pgm_read_float(&tmp2[i]);

                                       ^

ConfigurationStore.cpp:436: error: 'tmp3' was not declared in this scope

     max_acceleration_units_per_sq_second[i] = pgm_read_float(&tmp3[i]);

                                                               ^

exit status 1
expected initializer before 'MARLIN_PROGMEM'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Wackerbarth, 2015-11-29:
> On Nov 28, 2015, at 4:38 PM, SLYSCORPIO <notifications@github.com> wrote:
> sketch\pins.h:2366:0: warning: "X_MS1_PIN" redefined [enabled by default]
> 

I don’t know what source you are trying to compile.
The current pins.h file has only 257 lines.
Branches earlier than 1.1 (aka RC, RCBugFix, etc.) do not compile properly on modern Arduino IDE releases.

Please switch to the RCBugFix for the latest prerelease firmware, or the “dev” branch in the MarlinDev repository for the latest buildable firmware.

SLYSCORPIO, 2015-11-29:
Okay I downloaded the Marlin RCBugFix and it compiled. I will transfer some code and recompile to make sure we are good to go but it appears that has fixed my problem. I appreciate your assistance.

SLYSCORPIO, 2015-11-30:
After running into some problems trying to print, I decided to download Arduino IDE version 1.0.6. That allowed me to compile the older version of Marlin and fixed my problem.

closed 2015-11-30 by SLYSCORPIO