MarlinFirmware/Marlin #2811: G29 Fails with Kossel Mini with FSRs

Sberfield, 2015-11-29:
I have a Kossel Mini with FSRs. I have the probe offsets set to X0.0 Y0.0 Z0.1

When I issue a G29, the printer lams upward, crashing the endstops repeatedly then cruises down until  it crashes into the build plate. 

I have all the usual measuremetns locked in properly. Have looked at the coe, can't see why this would be happning.

brainscan, 2015-11-29:
Not sure if this is the only problem but offsets must be negative.

brainscan, 2015-11-29:
Sorry z must be negative.

KiteLab, 2015-11-29:
@Sberfield 
Please describe again what exactly happens when you send a G28 G29.

Is the nozzle touching the bed at [0,0,0]? (It should not!)
Is your `DELTA_PROBABLE_RADIUS` small enough?
What is your setting for `ENDSTOPS_ONLY_FOR_HOMING`? FSRs have to be disabled after homing!

@brainscan 
For FSR-probes the z-offset has to be positive. With them ether the bed is pressed down or the nozzle is pressed up until they trigger. If that works with Marlin is an other question. The comment in Configuration.h was written with not having FSR-sensors in mind.

KiteLab, 2015-11-29:
Related:
https://github.com/MarlinFirmware/MarlinDev/issues/285 Q5
https://github.com/MarlinFirmware/Marlin/issues/2722

brainscan, 2015-11-29:
@kitelab
I spent a while trying to make a probe that needed a positive z offset and was advised at the time it wouldn't work unless I tweaked the bed levelling stuff. My math is pretty bad so I didn't know where to make the changes so for me it was easier to redesign the probe. That was more than a year ago so things could have changed, but I thought it still said it configuration.h that z offset must be negative.

Roxy-3DPrintBoard, 2015-11-29:
I think the new probe_pt() routines tolerate positive Z_PROBE_OFFSET_FROM_EXTRUDER numbers now.   I don't know that for a fact and I don't have the hardware to check it out.   But for some reason, I have that in my head.

> When I issue a G29, the printer lams upward, crashing the endstops repeatedly then cruises down until it crashes into the build plate.

Crashes the end stops repeatedly?  Are you talking about the normal homing procedure where everything goes up until one of the endstops gets hit, and then X backs off and slowly moves back to the endstop, then Y does the same thing, and then Z?    You do realize you should do a G28 first before a G29, right?

For starters, maybe you should back your FEEDRATE numbers off so things move much slower?   That will let you see exactly what is happening and give you time to protect your hardware if the wrong thing is happening.

Next, after G28 it might be good to use your hand to trigger the endstop well above the build plate so you can verify the positive Z-Offset is working.   While you are changing the FEEDRATE numbers, change the manual home position to be half as big as it really is so you have room to work and manually trigger the probe to see what kind of numbers you are getting.

brainscan, 2015-11-29:
> I think the new probe_pt() routines tolerate positive Z_PROBE_OFFSET_FROM_EXTRUDER numbers now. I don't know that for a fact and I don't have the hardware to check it out. But for some reason, I have that in my head.
Maybe I'll try and find the old probe I was working on and see what happens. It was a long time ago and I've moved house since so not sure how lucky I'll be but I really wanted to get it working.

Sberfield, 2015-11-30:
Thanks guys -
Answering some of the questions:
1) Nozzle is not touching at 0,0,0 it is a paper thickness above the plate
2) Delta probable radius is set to DELTA_PRINTABLE_RADIUS-10 which comes out to 75
3) The FSRs are running using a board that produces a very reliable on/off signal. Why would I turn them off when trying to use them to auto level? I am confused on this one.
4) I know the difference between normal homing and repeatedly slamming into the top end-stops with a shuddering that seems like it should break something. 
5) I also know to home before doing anything else.
6) When I briefly had a different version of Marlin running, it was successfully reading the probe values. 7) I have verified the FSR switching is working and that the top endstop switches are consistent. 
The behaviors looks a lot like some seriously bad value is getting generated and not caught as an error.

Roxy-3DPrintBoard, 2015-11-30:
@Sberfield  which version of Marlin do you have loaded?

Sberfield, 2015-11-30:
@Roxy-3DPrintBoard 1.1.0 RC2

Sberfield, 2015-11-30:
I am fully aware that this is probably something stupid I have missed or done :)

Sberfield, 2015-12-02:
OK - learned a little here. When I issue a g29 after a g28, the head tries to move to x 0, y 80, Z339.5??
My Max Z is 260.8. The shuddering appears to be the gathering of probe samples? At least this is what I get in the console:
>>> g29
SENDING:G29
G29 Auto Bed Leveling
-14.89 49.98 49.98 49.98 49.98 20.04 5.07 10.06 64.95
20.04 35.01 74.93 69.94 64.95 59.96 54.97 84.91 99.88
129.82 79.92 84.91 89.90 94.89 99.88 104.87 109.86 49.98
149.78 144.79 139.80 134.81 129.82 124.83 119.84 114.85 109.86
149.78 154.77 159.76 164.75 169.74 174.73 179.72 184.71 189.70
229.62 224.63 219.64 214.65 209.66 204.67 199.68 194.69 189.70
289.50 229.62 234.61 239.60 244.59 249.58 254.57 259.56 209.66
239.60 254.57 284.51 279.52 274.53 269.54 264.55 304.47 319.44
274.53 329.42 334.41 319.44 289.50 289.50 289.50 289.50 354.37

I am guessing I have something defined wrong, but darned if I know what. I'm attaching the configuration file for reference. 

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/49447/Configuration.txt)

Video at: https://youtu.be/ONqiOfJrPII

AnHardt, 2015-12-02:
Your display indicates the target is X 0 Y -80 Z 50 (a bit unsharp in your video) what is the same value i calculated from your config for the first point to probe at Z_RAISE_BEFORE_PROBING.

If we do not have a probe what must be deployed, like your FSRs the first move for a G29 is done in `probe_pt(`. `probe_pt(` splits the move in a vertical and a horizontal component, and is doing the vertical first, then the horizontal. In you video we can't see a vertical move before the (bowlshaped) horizontal. That indicates a triggered z-min endstop/probe! what is not expected here and a for that messed up tower coordinate system.
Can the vibrations of your moving motors trigger the FSRs? You could make a test. Home the machine with a `G28`, send a `M120` to activate the endstop testing manually, try to move down `G1 Z50`. Will it move down? Do you get a message on the console?

Roxy-3DPrintBoard, 2015-12-02:
This is the same answer is AnHardt gave but using different words.   The probe_pt() routine does a Z_RAISE_BEFORE_PROBING and then tries to probe the point.  If the Z-Probe is triggered, it will then do a Z_RAISE_BETWEEN_PROBES and raise some more.

And then the G29 will go on and try to probe the next point.  If the Z-Probe is already triggered, it will immediately do another Z_RAISE_BETWEEN_PROBES.  It will do this for every probe attempt and will result in the nozzle going up and up and up.

This is very similar to what happens with a manually deployed Allen Key probe.   If the probe is not deployed, it shows up as triggered.  And when you try to do the G29 the effector starts going up and up and up.

Sberfield, 2015-12-02:
Thanks - this gives me some things to look at. I will do some checking on the FSRs. Possible the endstop board is flakey. I don't think they are that sensitive normally - and they worked OK when I had the Cattel version installed - though obviously that is a totally different chunk of code. I will do some testing tonight and see if I can sort it out. At least now I know where to look. Thanks!

Roxy-3DPrintBoard, 2015-12-02:
Start with a M119 in the middle of the bed halfway up.    See if it says 'Triggered'.   If so, there may not be anything wrong with the hardware.  It might just be a configuration issue.

Sberfield, 2015-12-03:
Got it!
Because it is a Delta, there are no physical min end-stop switches. But I can't have DISABLE_MIN_ENDSTOPS defined, because I am using Z-Min for the FSR end-stop board input. I have the Z_MIN inverted as required by the FSR board, but since there are no switches for X & Y, I hadn't inverted them. So X & Y were triggered all the time. I inverted the signal and the reoutine now works perfectly. Many thanks!

closed 2015-12-03 by Sberfield