MarlinFirmware/Marlin #2813: X-Axis Homing Stops working when inverted

bholmes451, 2015-12-01:
While setting up a new printer, I determined the X Axis was mirrored.  So on the hardware side I moved the X-Min Endstop to the X-Max Endstop pins.  I then changed #define INVERT_X_DIR from true to false , and #define X_HOME_DIR from -1 to 1.  All of my MAX_ENDSTOP_INVERTING parameters are set to true.

Issue: The X-Axis will no longer home. The G28 command will test the endstops for the Y and Z but not the X.

```C++
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
const bool X_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MIN_PROBE_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
#define DISABLE_MAX_ENDSTOPS
//#define DISABLE_MIN_ENDSTOPS

// If you want to enable the Z probe pin, but disable its use, uncomment the line below.
// This only affects a Z probe endstop if you have separate Z min endstop as well and have
// activated Z_MIN_PROBE_ENDSTOP below. If you are using the Z Min endstop on your Z probe,
// this has no effect.
//#define DISABLE_Z_MIN_PROBE_ENDSTOP

// For Inverting Stepper Enable Pins (Active Low) use 0, Non Inverting (Active High) use 1
// :{0:'Low',1:'High'}
#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define E_ENABLE_ON 0 // For all extruders

// Disables axis when it's not being used.
// WARNING: When motors turn off there is a chance of losing position accuracy!
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false

// @section extruder

#define DISABLE_E false // For all extruders
#define DISABLE_INACTIVE_EXTRUDER true //disable only inactive extruders and keep active extruder enabled

// @section machine

// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false
#define INVERT_Y_DIR true
#define INVERT_Z_DIR true

// @section extruder

// For direct drive extruder v9 set to true, for geared extruder set to false.
#define INVERT_E0_DIR true
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false

// @section homing

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

#define min_software_endstops false // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops false  // If true, axis won't move to coordinates greater than the defined lengths below.

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS -3
#define X_MAX_POS 200 // 205 for full bed
#define Y_MAX_POS 220
#define Z_MAX_POS 273
```

brainscan, 2015-12-01:
Wouldn't it have been easier to plug the stepper in the other way?

bholmes451, 2015-12-01:
That's probably also why this hasn't been uncovered before. ;-)  I'll give that a try.  The reason I didn't do that is because the movement was correct and it was printing, just the items being printed were mirrored.

bholmes451, 2015-12-01:
Ok, tried that.  While it switched the direction of the motor...  It still doesn't home.  So I moved the endstop to the other end of the X-Axis and switched '#define X_HOME_DIR' from a 1 to a -1 (default) and it now homes again.  So it appears to specifically be a problem with having the X_HOME_DIR set to a positive 1.

bholmes451, 2015-12-01:
Ok, I found the real problem after undoing a number of config file changes I made last night...  It seems there is a new feature switch I missed before...

'''
#define DISABLE_MAX_ENDSTOPS
'''

I commented out that line and everything works the way I expect.  I hope these comments will help someone else with a similar issue.

closed 2015-12-01 by bholmes451

jairov4, 2017-02-19:
Thanks I made the same mistake today. years later XD