MarlinFirmware/Marlin #2814: RAMPS ANALOG PROBLEM

CoderSquirrel, 2015-12-02:
Hi guys, my problem is more hardware than software but in case anyone had it.
I was testing my joystick with arduino at analog inputs and it works perfect, but when i put the ramps on it it becames really strange.
I tried A3,A4,A9,A10,A11,A12 with my pin at the GND, so it's value must recieve only 0
But it's recieving:
`
1023.00
0.00
0.00
0.00
0.00
1023.00
0.00
0.00
0.00
0.00
0.00
0.00
0.00
0.00
1023.00
`

And it goes on and on this way.
I thought that the problem was that all of then, A3 to A12, are both Analog and Digital pin, according to the datasheet, but i tried with the A15, only analog and it didn't worked anyway.

So, for the analog pins at the RAMPS i need to make an specific analog declaration or something like it?

CoderSquirrel, 2015-12-03:
I tried to read the pin at marlin_main and there the reading is ok, reading the correct value, but in ultralcd.cpp still reading wrong.
Anyone knows what it may be?
It giving me a lot o headache

thinkyhead, 2016-02-22:
Did you get this sorted out? I'm closing this as an old issue, but please re-open if you are still stuck on this.

closed 2016-02-22 by thinkyhead