MarlinFirmware/Marlin #2816: Determine Endstop Hits when Homing

thesec, 2015-12-02:
Hi,

I just updated to 1.02 and wondered where to find the setting, if any, to allow for hitting the endstops twice when homing my Delta?

Thanks ahead.

KiteLab, 2015-12-04:
That's neither configurable in 1.0.2 nor later branches. Marlin should do always two bumps.
But there should be a backup from the endstops before setting the software endstops.

thesec, 2015-12-04:
It does not for me. How comes then?!

KiteLab, 2015-12-04:
In case this parameters:
```
//homing hits the endstop, then retracts by this distance, before it tries to slowly bump again:
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 5 // deltas need the same for all three axis
```
are to small to untrigger the endstops, no second bump will happen

thesec, 2015-12-04:
Seems to be called *_HOME_RETRACT_MM in 1.02 and is still set to 5 per default.
Yet the printer does not try to bump again on any axis.

What does determine a "too small value"?

KiteLab, 2015-12-04:
>  small to untrigger the endstops.
The carriages should move down that way to give the switch enough room to expand again, to release the contact.

There is an other possible cause for not moving down.
Please send a M119 to your printer and report the output here. Once in mid air, once right after homing.
Are any min-endstops TRIGGERED?

thesec, 2015-12-04:
Sry, that'll have to wait for the printer to be re-build ... am currently overhauling ...

Will get back when it is back up.

closed 2016-02-22 by thinkyhead