MarlinFirmware/Marlin #2818: Warning in configuration_store.cpp

indazoo, 2015-12-03:
Hi
I tried to compile the latest Marlin release candidate. It showed some warnings.
Perhaps its only me, but I don't like errors and warnings. 

Output:
configuration_store.cpp:In function 'void Config_ResetDefault()
configuration_store.cpp:545:17: error: unused variable 'e' [-Werror=unused-variable]
:const int e = 0; \\ only need to write once

So, if one changes only line 543  into a for loop, then the warning does not appear.
Other code structure is possible, but I don't wanted to change too much.
So this one liner may be accepted.

What do you think?

*****************************************************************
Existing code:
#if ENABLED(PIDTEMP)
    #if ENABLED(PID_PARAMS_PER_EXTRUDER)
      for (int e = 0; e < EXTRUDERS; e++)
    #else
      int e = 0; // only need to write once   <<<<<< Source of warning
    #endif
    {
      PID_PARAM(Kp, e) = DEFAULT_Kp;
      PID_PARAM(Ki, e) = scalePID_i(DEFAULT_Ki);
      PID_PARAM(Kd, e) = scalePID_d(DEFAULT_Kd);
      #if ENABLED(PID_ADD_EXTRUSION_RATE)
        PID_PARAM(Kc, e) = DEFAULT_Kc;
      #endif
    }

*****************************************************************
New Code
#if ENABLED(PIDTEMP)
    #if ENABLED(PID_PARAMS_PER_EXTRUDER)
      for (int e = 0; e < EXTRUDERS; e++)
    #else
      for (int e = 0; e < 1; e++) // only need to write once   <<<===  changed
    #endif
    {
      PID_PARAM(Kp, e) = DEFAULT_Kp;
      PID_PARAM(Ki, e) = scalePID_i(DEFAULT_Ki);
      PID_PARAM(Kd, e) = scalePID_d(DEFAULT_Kd);
      #if ENABLED(PID_ADD_EXTRUSION_RATE)
        PID_PARAM(Kc, e) = DEFAULT_Kc;
      #endif
    }

Wackerbarth, 2015-12-03:
This has already been corrected in the "dev" branch.

KiteLab, 2015-12-03:
The real clean-up would be to use Kp[0] instead of Kp (etc) in a single extruder configuration

indazoo, 2015-12-04:
Nice to hear that it has already been corrected! I looked at it and had the impression that correcting the warning by using "UNUSED(e)" is not really what happens in the code. For me it looks like (e) is being used. 
One can debate, that the compiler warning is incorrect because it somehow interprets the scope wrongly. But I have no intentions to debate endless about this because I'm no C expert. So, the original issue is adressed.

Wackerbarth, 2015-12-06:
Actually the parameter to the macro is unused because with only one extruder, the parameters are defined as scalars rather than as an array.  The nice thing is that the compiler optimizes all of the kruft away.

indazoo, 2015-12-09:
Ah! I see now why you do this! To save 20-36 bytes of RAM.
From temperature.h :
```
#if ENABLED(PID_PARAMS_PER_EXTRUDER)
    extern float Kp[EXTRUDERS], Ki[EXTRUDERS], Kd[EXTRUDERS], Kc[EXTRUDERS]; // one param per extruder
    #define PID_PARAM(param,e) param[e] // use macro to point to array value
  #else
    extern float Kp, Ki, Kd, Kc; // one param per extruder - saves 20 or 36 bytes of ram (inc array pointer)
    #define PID_PARAM(param, e) param // use macro to point directly to value
  #endif // PID_PARAMS_PER_EXTRUDER	
```

closed 2015-12-09 by indazoo