MarlinFirmware/Marlin #2827: Kossel Autoleveling moving up?

pistreamer, 2015-12-04:
Hi, I just updated my printer firmware (from 1.0) to newest release and have problems with autoleveling. The printer always moves up after homing during autoleveling. I am using a servo with endstop, which perfectly deploys and retracts after the process. All manual commands are working without any problems. What could be the reason, that autoleveling is working in the wrong z direction? Here is a snapshot of my debug .. end of G28 and beginning of G29:

KiteLab, 2015-12-04:
Please describe a bit more what happens.
You have homed the effector is at [0,0,215.80]
The servo deploys.
The effector is going down to [0,0, Z_RAISE_BEFORE_PROBING] 
Your  Z_RAISE_BEFORE_PROBING is 0.
Probably it will never arrive there, because your probe is already extended.
Set  Z_RAISE_BEFORE_PROBING to at least Z_RAISE_BETWEEN_PROBINGS to be sure not to touch the bed here.

Z_RAISE_BETWEEN_PROBINGS must be big enough to to avoid probe contact on all horizontal (bowl shaped moves). 
Z_RAISE_BEFORE_PROBING is ought to describe the minimal height above the bed for safely, deploying the servo probe, without touching the bed

Phaelz, 2015-12-04:
I am having the same problem here. I have a delta with no servo or anything that is deployed. The hotend itself is used as probe.

After homing, when I press G29, everything wants to go up, but the delta already is at its maximum height. With Marlin 1.0.2 or 1.0.4 I have no problem at all.

I tried to comment #define ENDSTOPS_ONLY_FOR_HOMING but then something really weird happens, it is like my steps/mm becomes 1, the printer barely moves.

Tried a lot of different options, no success.

KiteLab, 2015-12-04:
@Phaelz
I guess your probe is permanently triggered. Please test with M119.

Phaelz, 2015-12-04:
@KiteLab, it is not.

I tried several delta configurations, same thing.

Phaelz, 2015-12-04:
In fact, I triggered the endstop with my hand during the auto level several times and the behavior is the same.

KiteLab, 2015-12-04:
@Phaelz 
Please home G28, activate the endstops manually M120, go down G1 Z50.
Any messages on the console?

Phaelz, 2015-12-04:
@KiteLab , oh god hahahahah what is M120?

Here is the message:

echo:endstops hit:  X:488.28 Y:488.28 Z:488.28

I put G1 Z50 and it went down half centimeter only! What happened?

KiteLab, 2015-12-04:
@Phaelz
One of your endstops triggered.

Phaelz, 2015-12-04:
@KiteLab , I don't understand. I did active the Z_MIN_ENDSTOP with my hand before M120, and while it was activated I entered G1 Z50. It moved down just a little bit instead of moving 250mm.

And how the hell my endstops activated in 488.28? That's not even possible. I have Xmin and Ymin as triggered all the time, are they causing this? How do I deactivate them since the 1.1.0 marlin version removed the disable endstop option in config.h?

KiteLab, 2015-12-04:
@Phaelz
Arrrrrg.
```
const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
```
Change these.

KiteLab, 2015-12-04:
Sorry MIN!!!

KiteLab, 2015-12-04:
Golden rule: "No unused endstop shall be triggered."

Phaelz, 2015-12-04:
@KiteLab , I'm so sorry. Such a rookie mistake... I didn't know that Min endstops would interfere with the delta... everything is fine now... probably @pistreamer is facing something like this.

Thank you very very much! Everything is fine now, I'll keep with the debugging in this delta. I hate this kind of printer! haahhaha

pistreamer, 2015-12-04:
Hi! It seems that there are a lot of changes, between my old an new version ;) In my old version I didn't have to disable or invert min endstops for delta printer, which causes the problem. Now it works thank you very much!

closed 2015-12-06 by Wackerbarth