MarlinFirmware/Marlin #2831: Can't connect to Kossel Pro

KiteLab, 2015-12-06:
With yesterdays `RCBugFix` i could connect without problems. Today i can't get a connection. Has something changed? I can't find a PR!

Roxy-3DPrintBoard, 2015-12-06:
We did promote RCBugFix to be RC-3.   And RCBugFix became a repository for fixes to RC-3.    I am not aware of anything being put into it that would break communications.

KiteLab, 2015-12-06:
> not aware

Exactly that is the problem with direct commits. Without PR, you don't get a notice. https://github.com/MarlinFirmware/Marlin/commit/6bfcb74c25bfccfe602f5ea4bc2f786b200345c4

Roxy-3DPrintBoard, 2015-12-08:
@KiteLab 

> Exactly that is the problem with direct commits. Without PR, you don't get a notice. 6bfcb74
Baudrate changed.

Can you explain what we did wrong and how we should have done things so it doesn't cause problems?

KiteLab, 2015-12-08:
@Roxy-3DPrintBoard 
Congratulations. Nice new logo.
Are you familiar with the 'watching' funktionallity of GitHub? The little bell symbol on the top right of your GitHub window? On https://github.com/settings/notifications you can adjust how much and about what actions you want to be informed. 
```Watching
Updates to any repositories or threads you’re watching.

 Email    Web
```
With the setting above you'll get notification about __close to__ every changes to a repository you watch.
The two exceptions i know about are, a 'direct commit' and edits to issues/PRs, where you don't get a message for. 
In  https://github.com/MarlinFirmware/Marlin/commit/6bfcb74c25bfccfe602f5ea4bc2f786b200345c4 Richard was so kind, to try to "repair" the example 'Configuration.h' for the Kossel Pro in the 'RCBugFix' branch - with a direct commit. If he had done this with a PR, we for sure had a discussion about the change in the baud rate. We'd have a PR, where a interested person could search for changes.

If we have a rule about "Don't merge your own PRs (if not very urgent)" this is also, and especially, true for direct commits. Basically they are immediate merged PRs without notification to other interested persons.
I can easily accept when good/the_owner/you are changing text in the README.md on the direct way, but changes to the functionality should have a chance to be discussed and documented.

Roxy-3DPrintBoard, 2015-12-08:
Thank You for the explanation.  At a higher level, I understand what you are saying and why you are saying it.  (But I'm not a GitHub expert yet...)

>  If he had done this with a PR, we for sure had a discussion about the change in the baud rate. 

There was a long discussion about baud rates not needing to be set so high because it was causing a lot of CPU load that resulted in some printers stuttering.   I suspect this change was because of that discussion.   I guess I should go back and re-read that thread.

Can you give me your issues with lowering the baud rate so I have a more complete understanding?

KiteLab, 2015-12-08:
@Roxy-3DPrintBoard 
If the old baudrate used to be 25000 and you have used that for a year, you have a problem to connect with 25000 when someone changes the default without notice, until you find the change.

Roxy-3DPrintBoard, 2015-12-08:
> If the old baudrate used to be 25000 and you have used that for a year, you have a problem to connect with 25000 when someone changes the default without notice, until you find the change

Don't you mean 250,000 ?   I need to go back and re-read that thread.  But if I remember correctly, the stated baud rate doesn't really matter on a USB Serial connection.   I think the claim (and testing) was that this number could be lowered so the USB interrupt rate wasn't hammering the CPU as badly but everything would still connect without a problem.  

What I remember is having a higher burst rate wasn't beneficial because there were (almost) always some GCode commands buffered up and ready to act upon.   But having the higher burst rate meant that other important stuff could not get the processor because as soon as there was buffer space for the next command (because one just finished) the CPU got hammered pulling in yet another buffered command.    

I'm pretty sure the lower baud rate was tested and there was consensus it would not hurt anything.  But like I say...   I need to go back and re-read that thread.   If you had a problem, that implies we should have tested more.

KiteLab, 2015-12-08:
I do not doubt that 115200 is the better speed, but I'm annoyed by changing this thou the back door.

Roxy-3DPrintBoard, 2015-12-08:
> but I'm annoyed by changing this thou the back door.

Duly noted.

Wackerbarth, 2015-12-08:
@KiteLab -- And you are the one who advocates using a wiki (where ANYONE can change the published version of anything without discussion or notice)!

I will agree that we should not be changing the FUNCTIONAL code without prior discussion.
However, as you note, correcting descriptive material without going through the review process is somewhat different.

Personally, I have a major problem with the way Github (at least in the way we have been using it) modifies the git tree.   MERGES (that are not an append) are EVIL. They hide changes that are the result of merge conflicts and, generally, violate the concept that each commit should be concerned with only one aspect of the code.

Using Github's PR process is a cumbersome wrapper around the git mechanism.
It requires multiple web-based interactions to accomplish what has already been done in your local copy of the repository. I find it VERY inefficient.

This is particularly a problem where we have everything in one big file (Marlin_main.cpp) and almost every change affects it in some manner.  (But we hope to start addressing THAT issue in our future development)

Now, I will argue that the file that I changed is just a reference document (an EXAMPLE).
In the current configuration scheme, each user is REQUIRED to maintain their own version of their configuration files. My change did not affect YOUR file. You should have reviewed any changes and evaluated, for yourself, whether, or not, you wanted to make the corresponding change.

KiteLab, 2015-12-08:
diff and patch do a great yob - if there are no surprises.

Wackerbarth, 2015-12-08:
@KiteLab -- diff and patch are tools. I use then constantly within my local file system.
They are but a "workaround" in terms of decyphering obscured commits.  And, it is the obscurred commits whichare the issue. It is very wasteful, and error prone, to be required to re-examine commits repeatedly in order to understand but they are changing.

Further, those local tools have nothing to do with the workflow issues caused by the use of web interfaces in GitHub as a method of sharing git repositories.

KiteLab, 2015-12-08:
I use diff and patch to merge my_config.diff into the example_configurations. That way i usually have no need to look into the new_example_configuration - except something is going wrong.

closed 2015-12-18 by KiteLab