MarlinFirmware/Marlin #2832: Where to fork feature development from?

crashmatt, 2015-12-08:
I would like to develop a calibration system for the auto bed leveling probe.  This measures the offset between the probe and extruder.  MarlinDev seems the obvious choice as feature pull requests against RC are not accepted.  Keeping track of a fast moving codebase is hard work.

Is MarlinDev codebase a long way from Marlin RC?  If I develop from Marlin RC is my contribution junk?
What is the state of MarlinDev?  How reliable is it?  How stable is it?

What do you recommend?

Roxy-3DPrintBoard, 2015-12-08:
The purpose of MarlinDev is to separate development activities from end user use of Marlin.   So this discussion would probably be better to have on the other side of the fence.   MarlinDev is where we intend to get new features sorted out and working.   Marlin is where we intend to have stable branches and release candidates vetted.

How closely the RC's that are active in Marlin correlate to what is happening in MarlinDev's dev branch will vary.  But right now, at this point in time, the code base is nearly identical.   This might be over stated a little bit, but if you want your code to be accepted and integrated with the rest of Marlin, you should be doing the work over in MarlinDev/dev.

crashmatt, 2015-12-08:
Thanks.  I will go with MarlinDev.

I very much like the new user specific config organization.  Keep up the great work.

/Matt

closed 2015-12-08 by crashmatt

Wackerbarth, 2015-12-08:
"I very much like the new user specific config organization" 
Thanks, @crashmatt. It's still a work-in-progress. But I'm trying to accomplish to a few things by making the change:
• Provide a way for the user to maintain and use their personal configurations without editing/changing the distributed source tree
• Present the user configuration in a way that makes apparent just what entries are different from the "stock" configuration
• Apply the DRY principal to help keep all configurations synchronized with respect to features.