MarlinFirmware/Marlin #2835: RUMBA: EXTRUDER_0_AUTO_FAN_PIN explanation

mkubik, 2015-12-09:
Hi, 

I have a dual MK8 extruder, and would like to use the EXTRUDER_XX_AUTO_FAN_PIN feature.
However, when I set 
> #define EXTRUDER_0_AUTO_FAN_PIN 7
> #define EXTRUDER_1_AUTO_FAN_PIN 8

I get an error from the sanitycheck because FAN_PIN is also set to 7.

Now, when I set 
> #define FAN_PIN -1
> #define FAN1_PIN -1

in pins_RUMBA.h, this error still persists, so it seems that at least FAN_PIN is defined somewhere else.
I tried to work around this by simply disabling the sanity check. That worked for extruder 1 but not extruder 0.
How is this designed to work? What am I missing?

KiteLab, 2015-12-10:
Sorry i can't reproduce that with Arduino 1.6.6 an our RC.

https://github.com/MarlinFirmware/Marlin/wiki/errors_submitting

mkubik, 2015-12-10:
Appologies... really seems to compile now. Confused...

closed 2015-12-10 by mkubik