MarlinFirmware/Marlin #2836: Z axis does not move down during print when gcode says so.

AndKe, 2015-12-09:
When printing per-object (first completing one part, then the other) - there are commands to reduce Z
like in line 3421 in the attached gcode.

Please see the attached video for result - the Z is never lowered, and printing midair fails.  

[Pitot-mount-for-Observer_front.zip](https://github.com/MarlinFirmware/Marlin/files/57427/Pitot-mount-for-Observer_front.zip)

KiteLab, 2015-12-09:
Is your z-axis really that fast?
F4800.000 mm/min = 80mm/s.
Moving upwards is only a very short way, so with a 'normal' acceleration you'll never reach that speed. But the 3.8 mm downwards could be enough to exceed the limit - and loosing steps.

AndKe, 2015-12-10:
Wow good catch ! -  it's an obvious configuration error if it's trying to actually travel that in Z.
having huge problems figuring out where that "idea" comes from, can't find it in slic3r configuration below gcode.
thank you

closed 2015-12-10 by AndKe

KiteLab, 2015-12-10:
`; travel_speed = 80`

AndKe, 2015-12-10:
correct - the default for that is 130mm/s , and I cannot reduce it to anything realistic for Z (this is an extra big prusa mendel alike i had for years)  
Did not find any speed limit for z in Merlin
I think we can conclude that this is a limitation of Slic3r ?

KiteLab, 2015-12-10:
```
#define DEFAULT_MAX_FEEDRATE          {300, 300, 5, 25}    // (mm/sec)
```
Normally all moves should be limited to that.
But i'm still investigating, if the units are consistent everywhere in Marlin.

KiteLab, 2015-12-10:
The speed limit with `DEFAULT_MAX_FEEDRATE` -> `max_feedrate[}` works. 
There are only some comments that could be improved. Everything is consistent in mm/s
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/planner.cpp#L67 should be mm/s.
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/planner.h#L123 should be mm/s

AndKe, 2015-12-10:
I failed to find the DEFAULT_MAX_FEEDRATE because I searched for "_Z"
Anyway - it was at default 5mm/s)   - but testing that with gcode commands. proved that the stepper controller had much to low current to achieve that. A year ago, or so, I did reduce all to "top speed" - apparently did not had use for continuous high Z speed  before
  increased current, and it moves as expected.
Thank you again.