MarlinFirmware/Marlin #2839: Marlin Thermal Runaway retries should be added

nialaduannob, 2015-12-11:
I was having frequently thermal runaway with printers drived with a Gen6 or Gen6 deluxe boards when it was never happening with a Megatronics board, knowing that all my printers have the same All In One hotend.
After investigation I found that the cause was a bad temperature acquisition for wthich I don't know the reason.
I have modified the Marlin code by adding retries when coming in the Thermal Run Awway case in the following way:
"retry" parameter added in thermal_runaway_protection.
In temperature.h:
........................
void thermal_runaway_protection(int *state, unsigned long *timer, int *retry, float temperature, float target_temperature, int heater_id, int period_seconds, int hysteresis_degc);
.......................
static unsigned long thermal_runaway_timer[3]; // = {0,0,0};
static int thermal_runaway_retry[3]; // = {0,0,0};
...............
static unsigned long thermal_runaway_bed_timer;
    static int thermal_runaway_bed_retry;
.......................

In temperature.cpp I did the following changes:

thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], &thermal_runaway_retry[e], current_temperature[e], target_temperature[e], e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
..............................
thermal_runaway_protection(&thermal_runaway_bed_state_machine, &thermal_runaway_bed_timer, &thermal_runaway_bed_retry, current_temperature_bed, target_temperature_bed, 9, THERMAL_RUNAWAY_PROTECTION_BED_PERIOD, THERMAL_RUNAWAY_PROTECTION_BED_HYSTERESIS);
.........................................
void thermal_runaway_protection(int *state, unsigned long *timer, int *retry, float temperature, float target_temperature, int heater_id, int period_seconds, int hysteresis_degc)
{
  if ((target_temperature == 0) || thermal_runaway)
  {
    *state = 0;
    *timer = 0;
    *retry = 0;
    return;
  }
  switch (*state)
  {
    case 0: // "Heater Inactive" state
      if (target_temperature > 0) *state = 1;
      break;
    case 1: // "First Heating" state
      if (temperature >= target_temperature) *state = 2;
      break;
    case 2: // "Temperature Stable" state
      if (temperature >= (target_temperature - hysteresis_degc))
      {
        *timer = millis();
        *retry = 0;
      }
      else if ( (millis() - *timer) > ((unsigned long) period_seconds) * 1000)
      {
    	  *retry=*retry+1;
    	  if (*retry > 1) { // Hard coded, could be in configuration.h
			  SERIAL_ERROR_START;
			  SERIAL_ERRORLNPGM("Thermal Runaway, system stopped! Heater_ID: ");
			  SERIAL_ERRORLN((int)heater_id);
			  SERIAL_ERRORLN(" (millis() - *timer) = ");
			  SERIAL_ERRORLN((unsigned long)(millis() - *timer));
			  SERIAL_ERRORLN(" period_seconds = ");
			  SERIAL_ERRORLN((unsigned long)period_seconds);
			  LCD_ALERTMESSAGEPGM("THERMAL RUNAWAY");
			  thermal_runaway = true;
			  while(1)
				  {
				  disable_heater();
				 .....................
			  }
    	  } else {
    		  SERIAL_ECHO_START;
    		  SERIAL_ECHO("ABNORMAL TEMPERATURE DETECTED");
    		  SERIAL_ECHO("Heater_ID: ");
    		  SERIAL_ECHO((int)heater_id);
    		  *timer = millis();
    	  }
      }
      break;
  }
}

The number of retries is hadcoded to 1 but could be defined in configuration.h.
For me with only one retry it is working fine and I have no more the frustartion having a 3D print stop after 3 hours without reason.
This retry feature should be added to the standard Marlin code.

KiteLab, 2015-12-11:
Have you tried:
```
Potential Hardware Fixes

There have been a couple of instances of dirty 5v power causing instability with the ftdi and atmega
chips. If this affects you, a 1n 0805 cap soldered across R12 might help.
See http://www.mendel-parts.com/new_forum/phpBB3/viewtopic.php?f=19&t=381 for details. --Pelrun 17:53, 5 July 2011 (UTC)

I wasn't having this problem, but did solder a 1nF ceramic cap across R12, and it reduced the amount of
audible noise the stepper drivers made, and limited ranges of stepper actuation the noise would occur at.
Meaning if I hit 0.1mm on my X and Y in pronterface, each operating point would make noise, whereas
now, only some do. This mod also improved my thermistor readings, which previously would vary
20-30C, but are now pretty much dead stable. --Thav 00:46, 27 January 2012 (UTC)
```
from http://reprap.org/wiki/Generation_6_Electronics

KiteLab, 2015-12-11:
For a noisy temperature input, maybe a additional oversampling like:
```
float newtemp = (current_temperature[e] + analog2temp(current_temperature_raw[e], e)) /2;
current_temperature[e] = newtemp;
```
in `updateTemperaturesFromRawValues()`
could be the better solution.

nialaduannob, 2015-12-11:
I would prefer not having to solder on my board.
Modifying updateTemperaturesFromRawValues() will probably not solve the problem in my case because the read temperature is largely below the true temperature and the result will still lead to a thermal runaway.
The solution I am proposing is a protection against outliers, whatever the cause. If there is a true problem, it will be no risk for the hotend if you cut it only 40ms later.

KiteLab, 2015-12-11:
Ah. now i see why the code looks so 'new' to me. It's based on '1.0.x`or earlier - looks a bit different now.

An additional state could avoid the new variable.
The state and the timer array could be defined inside the function, to prevent the globals and lots of parameters.

Fixing an hardware error in software is never a good idea. We already have 16 times oversampled the measurement.
Having no denouncer here could be seen as a principal error

Wackerbarth, 2015-12-11:
@KiteLab  -- The code remains based on early Arduino code because of the MarlinSerial hack to allow the ISR to run "too long" which would otherwise cause serial characters to be dropped.

It is complicated by the need to support AT90USB chips as well as those processors which have a separate processor handling the USB communication.

I would love to see a simpler solution which still allows the very rapid stepper motion.
Perhaps you can make some suggestions.

nialaduannob, 2015-12-11:
The code I am using was downloaded last month and the file dates are 2015-11-02 05:02.
It is not the last stable release?

KiteLab, 2015-12-11:
@Wackerbarth 
I do not see the relation of your last comment to this thread.
Until now the stepper IRQ has to many blurry parts in my head. If i'll understand it one day as a whole, maybe i can find some improvements. The serial and the sd code are even blacker holes in my knollage.

@nialaduannob 
Yes, likely you downloded the 'Release' branch. That's code about a year old. A lot of changes have been made since then. MarlinFirmware/Marlin `RCBugFix` and MarlinFirmware/MarlinDev 'master' are the branches we are currently working on.

thinkyhead, 2016-02-24:
@nialaduannob Sorry this issue has languished so long. Do you need any further assistance on this, or want to add anything? You should try the pre-release version, currently 1.1.0-RC3, or download the `RCBugFix` branch and try that out. The sensor code has been changed somewhat. These changes might have a positive effect. In the long run, it might be good to insert some logging code and see what's going on, then we might be able to deal with it in a more targeted way.

closed 2016-04-27 by jbrazio