MarlinFirmware/Marlin #2841: G29 on Delta judders upwards (regression RC2->RC3)

yoink00, 2015-12-11:
There appears to be a regression in the G29 command between RC2 to RC3. Using identical configuration between the two the G29 behaves differently:

In RC2 the extruder performs the expected auto levelling function but in RC3 it judders upwards (and it not monitored -- like the first time it happened) will crash into the max endstops.

I have a github fork that I use for my RC3 config here: https://github.com/yoink00/Marlin The branch with my configuration is 'mykossel'.

Let me know if you need anymore information.

KiteLab, 2015-12-21:
Please let us see a M119 in mid air.
If that does not show __any__ triggered switches send M120 and move around in all directions. Any messages on the console?

IkeHayes, 2016-02-04:
I have encountered the same problem using RC3 on a Mini Kossel delta using a RAMPS 1.4 board with a Z_Probe connected to Z_min endstop. I've noticed that when running a M119 command there is no console output for the z probe to state whether it is triggered or not. This seems to indicate that HAS_Z_PROBE is not being set (since it is not triggering the IF statement in the M119 function) which may mean the Z probe state has a faulty status.

KiteLab, 2016-02-04:
@IkeHayes 
There are some settings with influence on this

```
...
//#define DISABLE_MIN_ENDSTOPS
...
// If you want to enable the Z probe pin, but disable its use, uncomment the line below.
// This only affects a Z probe endstop if you have separate Z min endstop as well and have
// activated Z_MIN_PROBE_ENDSTOP below. If you are using the Z Min endstop on your Z probe,
// this has no effect.
//#define DISABLE_Z_MIN_PROBE_ENDSTOP
...
  // Support for a dedicated Z probe endstop separate from the Z min endstop.
  // If you would like to use both a Z probe and a Z min endstop together,
  // uncomment #define Z_MIN_PROBE_ENDSTOP and read the instructions below.
  // If you still want to use the Z min endstop for homing, disable Z_SAFE_HOMING above.
  // Example: To park the head outside the bed area when homing with G28.
  //
  // WARNING:
  // The Z min endstop will need to set properly as it would without a Z probe
  // to prevent head crashes and premature stopping during a print.
  //
  // To use a separate Z probe endstop, you must have a Z_MIN_PROBE_PIN
  // defined in the pins_XXXXX.h file for your control board.
  // If you are using a servo based Z probe, you will need to enable NUM_SERVOS,
  // Z_ENDSTOP_SERVO_NR and SERVO_ENDSTOP_ANGLES in the R/C SERVO support below.
  // RAMPS 1.3/1.4 boards may be able to use the 5V, Ground and the D32 pin
  // in the Aux 4 section of the RAMPS board. Use 5V for powered sensors,
  // otherwise connect to ground and D32 for normally closed configuration
  // and 5V and D32 for normally open configurations.
  // Normally closed configuration is advised and assumed.
  // The D32 pin in Aux 4 on RAMPS maps to the Arduino D32 pin.
  // Z_MIN_PROBE_PIN is setting the pin to use on the Arduino.
  // Since the D32 pin on the RAMPS maps to D32 on Arduino, this works.
  // D32 is currently selected in the RAMPS 1.3/1.4 pin file.
  // All other boards will need changes to the respective pins_XXXXX.h file.
  //
  // WARNING:
  // Setting the wrong pin may have unexpected and potentially disastrous outcomes.
  // Use with caution and do your homework.
  //
  //#define Z_MIN_PROBE_ENDSTOP
...
```
```
...
#define Z_MIN_PIN          18
...
#if ENABLED(Z_MIN_PROBE_ENDSTOP)
  // Define a pin to use as the signal pin on Arduino for the Z_PROBE endstop.
  #define Z_MIN_PROBE_PIN  32
#endif
...
```
Experiment with them. There is a way to make it work.

thinkyhead, 2016-03-16:
Where does this issue stand with `RCBugFix` today?

AnHardt, 2016-03-16:
Usually this is a symptom of a permanently triggered z-min-probe/z-min-endstop. For deltas every permanently triggered min-endstop is bad - they usually do not have hardware-min-endstops. So inverting them, or even better undefining them - gives you a working system
.
Special handling for permanently triggered when stowed probes is required. Up to now they are only working when ENDSTOPS_ONLY_FOR_HOMING is defined. Some (unmerged) PRs have been made to fix that.

thinkyhead, 2016-03-18:
Endstop sanity check, eh?

canadaduane, 2016-03-22:
I was experiencing the same issue, and @AnHardt's diagnosis and solution worked for me. My endstop configuration now looks like this:

```
const bool X_MIN_ENDSTOP_INVERTING = true; // I don't have an X min endstop
const bool Y_MIN_ENDSTOP_INVERTING = true; // I don't have a Y min endstop
const bool Z_MIN_ENDSTOP_INVERTING = false;
const bool X_MAX_ENDSTOP_INVERTING = false;
const bool Y_MAX_ENDSTOP_INVERTING = false;
const bool Z_MAX_ENDSTOP_INVERTING = false;
const bool Z_MIN_PROBE_ENDSTOP_INVERTING = false;
```

thinkyhead, 2016-03-27:
@canadaduane Could you take a moment to check out #3065 and comment on whether it would help this kind of situation?

closed 2016-03-27 by thinkyhead

canadaduane, 2016-03-27:
That looks like an improvement to me. Thanks.