MarlinFirmware/Marlin #2845: Home Position Reset after G29

idzuwan, 2015-12-12:
Hi,

I have my bed leveling working fine few days ago on my mendelmax with proximity sensor, yesterday I changed my X carriage because the old one has worn out, the only change that was made is the position of the sensor, before it was on the left side with -45 X offset with the new X carriage I moved the sensor to the right with the same positive value offset (45), after initial testing G28 work fine home to the center but after G29 my X have at least -25 offset while my Y having -20 offset

my marlin version is  1.0.2, the weird thing is I used the same firmware files with the only change was made was changing 

  #define X_PROBE_OFFSET_FROM_EXTRUDER -45

to 

  #define X_PROBE_OFFSET_FROM_EXTRUDER 45

now evertime G29 is run my home reset!

fbarcenas, 2015-12-14:
Why you don't test 1.1.0 RC3, i have some problems like this with old versions.
With this new version all is perfect.

Also test a thing, write "+" before 45. So:
#define X_PROBE_OFFSET_FROM_EXTRUDER +45

Best Regards!

idzuwan, 2015-12-15:
I have try +45 failed even more :), I will download RC3 tonight and see how it go

KiteLab, 2015-12-15:
If any of you experiences a difference between 45 and +45 this disqualifies all of your observations.

If you have changed the sign of X_PROBE_OFFSET_FROM_EXTRUDER this effects the probe point coordinates.
```
  #if ENABLED(AUTO_BED_LEVELING_GRID)

    #define LEFT_PROBE_BED_POSITION 15
    #define RIGHT_PROBE_BED_POSITION 170
    #define FRONT_PROBE_BED_POSITION 20
    #define BACK_PROBE_BED_POSITION 170

    #define MIN_PROBE_EDGE 10 // The Z probe minimum square sides can be no smaller than this.

    // Set the number of grid points per dimension.
    // You probably don't need more than 3 (squared=9).
    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // !AUTO_BED_LEVELING_GRID

      // Arbitrary points to probe.
      // A simple cross-product is used to estimate the plane of the bed.
      #define ABL_PROBE_PT_1_X 15
      #define ABL_PROBE_PT_1_Y 180
      #define ABL_PROBE_PT_2_X 15
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 170
      #define ABL_PROBE_PT_3_Y 20

  #endif // AUTO_BED_LEVELING_GRID
```
If you don't correct them, in the newer Marlin versions you'll get an error during compilation, in 1.0.2 you'll probably hit an endstop silently.

idzuwan, 2015-12-15:
I have this previously in 1.0.2

 ```
#ifdef AUTO_BED_LEVELING_GRID

    // set the rectangle in which to probe
    #define LEFT_PROBE_BED_POSITION 50
    #define RIGHT_PROBE_BED_POSITION 170
    #define BACK_PROBE_BED_POSITION 170
    #define FRONT_PROBE_BED_POSITION 20

     // set the number of grid points per dimension
     // I wouldn't see a reason to go above 3 (=9 probing points on the bed)
    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // not AUTO_BED_LEVELING_GRID
    // with no grid, just probe 3 arbitrary points.  A simple cross-product
    // is used to esimate the plane of the print bed

      #define ABL_PROBE_PT_1_X 50
      #define ABL_PROBE_PT_1_Y 20
      #define ABL_PROBE_PT_2_X 180
      #define ABL_PROBE_PT_2_Y 20
      #define ABL_PROBE_PT_3_X 100
      #define ABL_PROBE_PT_3_Y 170

  #endif // AUTO_BED_LEVELING_GRID

  // these are the offsets to the probe relative to the extruder tip (Hotend - Probe)
  // X and Y offsets must be integers
  #define X_PROBE_OFFSET_FROM_EXTRUDER 45
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 7
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -0.2 
```

I did download the new RC3 using the same config (almost all of them) it compiled fine and pass the sanity check but now it does not detect my inductive sensor, on my last firmware I didn't remember changing anything special to make the sensor work, The sensor triggered manually but m119 still showing it open, Need to figure it this one out first before I'm able to do more testing

thinkyhead, 2016-03-02:
How is this looking with the current `RCBugFix`? I'm hoping to merge a few more PRs and release the new code as RC4 pretty soon, so it would be good to get some testing in.

idzuwan, 2016-03-02:
I'm still having the issues, but it running on old firmware, I can't test it on 1.1.0 RC3 because it wont detect my inductive sensor which is weird as the 1.0.2 can see the sensor with not problem, so instead of fixing the issues I encounter new issues with RC3 :)

thinkyhead, 2016-03-02:
> I encounter new issues with RC3

Ok, but what I asked was… How is this looking with the current `RCBugFix`? That's ahead of RC3.

nickbardom, 2016-03-11:
I have exactly the the same problem. Tested with RC3 and RCBugfix.

jbrazio, 2016-03-11:
AFAIK Marlin does not do anything fancy with the inductive sensor.. it's strange the sensor works in a previous version and not with RCBugFix. Did you check the board pin you're using file if some change could have affected you ?

idzuwan, 2016-03-14:
I have been playing around with the pin setting in the latest build but nothing avail, in 1.1 I noticed that probe and z endstop are now seperated previously the probe use Z end stop min,  I did try changing the probe pin to use my previous used pin (z min endstop) still nothing then I tried to use the max endstop z pin for probe and z min left alone and changed the probe pin again to the correct pin still it does not see the probe been triggered

thinkyhead, 2016-03-16:
@ldzuwan Before flashing the firmware enable the option `DEBUG_LEVELING_FEATURE`. Then before you do `G28` or `G29` use `M115 S32` to enable extra debug messages for leveling. This will give you very detailed output to the serial console when you do homing and leveling.

Be sure to get the very latest `RCBugFix`, as it is being patched several times a day right now.

idzuwan, 2016-03-16:
okay will do another git checkout later today see if I can get anywhere with it

paulusjacobus, 2016-03-17:
@thinkyhead  not sure if I should post it in this issue but I wonder if G29 has been fixed for Corexy. I'm buidling a corexy machine and close to testing. Should I use your fork "G29 debug" which has been my marlin for both of my Prusa I2 machines or the RCBugFix release?

thinkyhead, 2016-03-27:
@paulusjacobus The `RCBugFix` branch has gotten a lot of CoreXY improvements that _should_ make `G29` work exactly as on a regular cartesian. At the moment strange buffer overflow issues and a few other mysteries are still with us in `RC4` and `RCBugFix`. So I'd recommend waiting a little longer before putting RC onto many machines. At least hold off until `RC5` which will be another day or two.

paulusjacobus, 2016-03-28:
Cool @Thinkyhead will wait a bit. Still putting the last little hardware
bits into my Corexy (it's almost ready for its first print).

jhnphm, 2016-04-03:
W/ octoprint

Note that probe offset is x:40 y:-10 and M206 set to X:-10 Y:-20

Also note that before issuing G29, G1 is issued w/ X0 Y0, yet `Recv: > current_position: (25.00, -130.00, 20.00)` is printed.

```
Send: G1 Z30
Recv: ok
[...]
Send: G28
Recv: gcode_G28 >>>
Recv: setup_for_endstop_move > enable_endstops(true)
Recv: set_axis_is_at_home 0 > (home_offset[axis]==-10.00) > current_position: (-10.00, 0.00, 30.00)
Recv: set_axis_is_at_home 1 > (home_offset[axis]==-20.00) > current_position: (-10.00, -20.00, 30.00)
Recv: > QUICK_HOME > current_position 1: (-10.00, -20.00, 30.00)
Recv: > QUICK_HOME > current_position 2: (-10.00, -20.00, 30.00)
Recv: >>> homeaxis(0)
Recv: > enable_endstops(false)
Recv: > enable_endstops(true)
Recv: > TRIGGER ENDSTOP > current_position: (0.00, -20.00, 30.00)
Recv: set_axis_is_at_home 0 > (home_offset[axis]==-10.00) > current_position: (-10.00, -20.00, 30.00)
Recv: > AFTER set_axis_is_at_home > current_position: (-10.00, -20.00, 30.00)
Recv: <<< homeaxis(0)
Recv: > homeX: (-10.00, -20.00, 30.00)
Recv: >>> homeaxis(1)
Recv: > enable_endstops(false)
Recv: > enable_endstops(true)
Recv: > TRIGGER ENDSTOP > current_position: (-10.00, 0.00, 30.00)
Recv: set_axis_is_at_home 1 > (home_offset[axis]==-20.00) > current_position: (-10.00, -20.00, 30.00)
Recv: > AFTER set_axis_is_at_home > current_position: (-10.00, -20.00, 30.00)
Recv: <<< homeaxis(1)
Recv: > homeY: (-10.00, -20.00, 30.00)
Recv: > Z_SAFE_HOMING >>>
Recv: > Z_SAFE_HOMING > home_all_axis > current_position: (-10.00, -20.00, 30.00)
Recv: > Z_SAFE_HOMING > home_all_axis > destination: (55.00, 105.00, 30.00)
Recv: >>> homeaxis(2)
Recv: deploy_z_probe > current_position: (55.00, 105.00, 0.00)
Recv: > enable_endstops(false)
Recv: > enable_endstops(true)
Recv: > TRIGGER ENDSTOP > current_position: (55.00, 105.00, 0.00)
Recv: set_axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (55.00, 105.00, 0.30)
Recv: > AFTER set_axis_is_at_home > current_position: (55.00, 105.00, 0.30)
Recv: > SERVO_LEVELING > (stow_z_probe())
Recv: stow_z_probe > current_position: (55.00, 105.00, 0.30)
Recv: <<< homeaxis(2)
Recv: <<< Z_SAFE_HOMING
Recv: > (home_all_axis || homeZ) > final: (55.00, 105.00, 0.30)
Recv: ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
Recv: <<< gcode_G28
Recv: X:55.00 Y:105.00 Z:0.30 E:0.00 Count A: 8800 B:16800 Z:48
Recv: ok
[...]
Send: G1 X0 Y0 Z20
Recv: ok
[...]
Send: G29 B135
Recv: gcode_G29 >>>
Recv: G29 Auto Bed Leveling
Recv: setup_for_endstop_move > enable_endstops(true)
Recv: z_before = (before) 15.00
Recv: probe_pt >>>
Recv: > ProbeAction:1
Recv: > current_position: (25.00, -130.00, 20.00)
Recv: Z Raise to z_before 15.00
Recv: > do_blocking_move_to_z 15.00
Recv: do_blocking_move_to: (25.00, -130.00, 15.00)
Recv: > do_blocking_move_to_xy 0.00, 30.00
Recv: do_blocking_move_to: (0.00, 30.00, 15.00)
Recv: > ProbeDeploy
[... 89 lines cut ...]
Recv: run_z_probe > current_position: (140.00, 144.00, 0.30)
Recv: > ProbeStow (stow_z_probe will do Z Raise)
Recv: stow_z_probe > current_position: (140.00, 144.00, 0.30)
Recv: <<< probe_pt
Recv: > probing complete > current_position: (140.00, 144.00, 0.30)
Recv: clean_up_after_endstop_move > ENDSTOPS_ONLY_FOR_HOMING > endstops_not_homing()
Recv: Eqn coefficients: a: 0.00081845 b: -0.00328947 d: 0.91812067
Recv: planeNormal x: -0.000818 y: 0.003289 z: 1.000000
Recv: set_bed_level_equation_lsq > current_position: (142.00, -2.00, 0.18)
Recv: 
Recv: 
Recv: Bed Level Correction Matrix:
Recv: +1.000000 +0.000000 +0.000818
Recv: +0.000003 +0.999995 -0.003289
Recv: -0.000818 +0.003289 +0.999994
Recv: > BEFORE apply_rotation_xyz > z_tmp  = 0.18
Recv: > BEFORE apply_rotation_xyz > real_z = 0.30
Recv: > AFTER apply_rotation_xyz > z_tmp  = 0.37
Recv: > corrected Z in G29: (142.00, -2.00, 0.37)
Recv: do_blocking_move_to: (142.00, -2.00, 15.37)
Recv: <<< gcode_G29
Recv: X:142.00 Y:-2.00 Z:15.37 E:0.00 Count A: 22726 B:-330 Z:2478
```

On 1.1.0-RC5 , git rev 4d0d37aab735f5f1265bc7d06863aea5a487fd09

thinkyhead, 2016-04-10:
Since you're using the `M206` home offset you might be running into the issue where bed leveling fails to use the home offset when probing points. This is being patched now in #3446 so you should check that out and see how it affects leveling for you. It may not result in perfection, but it should eliminate at least some of the weirdness.

jhnphm, 2016-04-18:
W/ this and #3311 seems to work now. Thanks!

closed 2016-04-18 by thinkyhead