MarlinFirmware/Marlin #2849: Add a "with respect to" to the probe position

Louis-Cypher, 2015-12-13:
Dear Developers, 

from my point of view the offset to be given for the auto-bed-leveling probe is ambiguous. It would add clarity to have written something like "with respect to nozzle".

Thanks in advance,

Markus

Roxy-3DPrintBoard, 2015-12-13:
Well...   I guess it depends upon what you are talking about.   Within the code, most things are relative to the active nozzle.   But things like probe_pt() actually position the probe at the specified coordinates by using the X_PROBE_OFFSET_FROM_EXTRUDER &  Y_PROBE_OFFSET_FROM_EXTRUDER values.   Go ahead and make the changes that you think are appropriate and that can be the basis for  further discussion on the issue.

thinkyhead, 2016-02-24:
The verbiage in `Configuration.h` now reads:
```cpp
  . . .
  // Offsets to the Z probe relative to the nozzle tip.
  // X and Y offsets must be integers.
  #define X_PROBE_OFFSET_FROM_EXTRUDER -25     // Z probe to nozzle X offset: -left  +right
  . . .
```

closed 2016-02-24 by thinkyhead

Roxy-3DPrintBoard, 2016-02-24:
If we are going to say this...  I think we need to say more.    My first printer is a good example.  Right now, I have the origin in the back right corner.   So everything is rotated about the Z-Axis 180 degrees.   in my case the comment needs to be 

// Z Probe to nozzle X offset: -right +left (when looking at it from the front)

thinkyhead, 2016-02-28:
How about:
```
// Z Probe to nozzle X offset: -left +right (when looking at it from your origin point)
```
or
```
// Z Probe to nozzle X offset, relative to origin: -towards +away
```

thinkyhead, 2016-02-28:
> Right now, I have the origin in the back right corner.

So when you home, your X and Y are something like 190, 190? That would be the proper way to set homing for a back-right corner.

jbrazio, 2016-02-28:
A picture may worth a thousand words..

```
 /*
  * Z Probe to nozzle (X,Y) offset, relative to (0, 0).
  * X and Y offsets must be integers.
  *
  * In the following example the X and Y offsets are both positive:
  * #define X_PROBE_OFFSET_FROM_EXTRUDER 10
  * #define X_PROBE_OFFSET_FROM_EXTRUDER 10
  * 
  *  -----------
  * |    (+) P  | <-- probe (20,20)
  * |           |
  * | (-) N (+) | <-- nozzle (10,10)
  * |           |
  * |    (-)    |
  *  -----------
  * (0,0)
  */
  #define X_PROBE_OFFSET_FROM_EXTRUDER 34  // X offset: -left  [of the nozzle] +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 15  // Z offset: -front [of the nozzle] +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below [the nozzle] (always negative!)
```

Louis-Cypher, 2016-02-28:
Great idea!

lrpirlet, 2016-02-28:
I like this idea.. In fact we could even keep the actual comment if we add in the drawing the left, right, back, front and O or origin as per the following...
```
 /*
  * Z Probe to nozzle (X,Y) offset, relative to (0, 0).
  * X and Y offsets must be integers.
  *
  * In the following example the X and Y offsets are both positive:
  * #define X_PROBE_OFFSET_FROM_EXTRUDER 10
  * #define X_PROBE_OFFSET_FROM_EXTRUDER 10
  * 
  *   ---BACK----
  *L |    (+) P  | R <-- probe (20,20)
  *E |           | I
  *F | (-) N (+) | G <-- nozzle (10,10)
  *T |           | H
  *  |    (-)    | T
  * O ---FRONT---
  * (0,0)
  */
  #define X_PROBE_OFFSET_FROM_EXTRUDER 34  // X offset: -left  [of the nozzle] +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 15  // Z offset: -front [of the nozzle] +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below [the nozzle] (always negative!)
```
Then we could also slightly change the code in G29 ABL to reflect where is the origin. This would help avoid having the "wrong hand coordinate system"
```
        // Show the Topography map if enabled
        if (do_topography_map) {

          SERIAL_PROTOCOLPGM(" \nBed Height Topography: \n");
          SERIAL_PROTOCOLPGM(" +-----------+\n");
          SERIAL_PROTOCOLPGM("Y|...Back....|\n");
          SERIAL_PROTOCOLPGM("^|Left..Right|\n");
          SERIAL_PROTOCOLPGM("y|...Front...|\n");
          SERIAL_PROTOCOLPGM(" O-----------+\n");
		  SERIAL_PROTOCOLPGM("O---x-->--X---\n");

```

jbrazio, 2016-02-28:
Marking the origin in G29 topography map is a good idea, but I believe marking x > X to indicate the direction X "grows" may not be clear, in fact I don't even know if that was your intention.

thinkyhead, 2016-02-29:
Please continue this discussion on #3053.