MarlinFirmware/Marlin #2850: Autobed leveling not working properly on core XY

Louis-Cypher, 2015-12-13:
Dear Developers,

my autobed leveling doesn't work anymore starting with RC2 (used release before). The Z- axis does run compensating motions but the compensation seems to be largely wrong. Printing on a tilted heatbed is impossible.

Thanks in advance
P.S.: I'll test whether it might be a sign error. I have the impression that ABL does compensate to the wrong direction.

Jo3yDaPh1sH, 2015-12-15:
I had the same issue with a printrbot simple metal setup on ramps 1.4 , going to upgrade to rc3 after this print to see if it is sorted. It seems if the autolevel  grid is fliped on the Y axis, smashed layers at rear of bed but barely touching at the front. strange

Roxy-3DPrintBoard, 2015-12-15:
Seriously....  RC-3 is very close to right.

Louis-Cypher, 2015-12-15:
Ok, what exactly does this statement say? ;-/

Since you a) haven't seen my printer and b) don't suggest any solution 
besides us believing what we see is "close to right" (no it isn't, it is 
at least 0.4mm off!) it isn't really helpful, right?

Maybe we try to find out what the difference in our setups is. My system 
is a core XY, probe location is front right from probe. What is your setup?

Can anyone point me to where in the code the math is done to get an idea 
what is wrong with my setup? To my understanding even in case my 
coordinate system is completely messed up the correction should be 
working since it reads wrong and it writes wrong, right?

Best,

Markus

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Roxy-3DPrintBoard, 2015-12-15:
It was a suggestion to use RC-3 instead of RC-2.   Is your Z-Probe-Offset negative?

Louis-Cypher, 2015-12-16:
OK, got it. Thanks for that. I already use RC 3. Yes, my offset in y is negative. Best, 

Markus

Louis-Cypher, 2015-12-21:
I tried to find out whether the correction appears to be inverted but it isn't. The compensation is always not enought on the lefthand side. Maybe I have to move my probe points even further off the edge of the heatbed.

Roxy-3DPrintBoard, 2015-12-21:
Is this a Cartesian machine?  Or is it a Delta?

Louis-Cypher, 2015-12-21:
It is a cartesian core xy.

Jo3yDaPh1sH, 2015-12-22:
I went to rc3 and used the mesh bed leveling for my warped 350mm x 290mm bed. Now my whole bed is printable!! The mesh leveling is a godsend in my application.

epatel, 2015-12-22:
@Jo3yDaPh1sH Happy to hear it helped your machine :)

Roxy-3DPrintBoard, 2015-12-22:
@epatel  Not to go too far off topic...   We need the Mesh Leveling to Auto Probe the bed, save the settings to EEPROM, and allow easy editing of a given point.    And depending upon how that goes, we may still want to add an automatic fine tune probe to make adjustments on a per print basis.  (I'm wondering if we may need to tilt the generated mesh over time, kind of like Auto Bed Leveling does.)

I'm very willing to help with this coding if it interests you enough to keep pushing forward!

Wackerbarth, 2015-12-22:
@Roxy-3DPrintBoard -- I think that we will be moving toward a scheme that creates a pipeline of transformations in the conversion from Object Coordinates to Frame Coordinates.
There is no reason why we cannot first apply a mathematical correction, either linear or non-linear, followed by a bed_mapping (mesh) correction before finally driving the actuators to the proper position.

By configuration option, either of these transformational corrections can be omitted and, mathematically, replaced with an identity transform.

epatel, 2015-12-22:
@Roxy-3DPrintBoard I made kind of an auto probe feature, though very simple as a I don't have a "normal" probe setup. See here

http://www.thingiverse.com/thing:1120142

Louis-Cypher, 2015-12-22:
I completely support that. Didn't know that it isn't done that way yet 
(to be honest that was the reason for my request to point me to where 
the math is done).

That's the way we do it in robotics because it is much less error prone 
than doing the math seperately everywhere you need it. Is it really 
necessary/meaningful to have a non linear transform? I'd guess having a 
linear transformation should be sufficient and maybe better, because a 
nonlinear one deforms the object as well.

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Roxy-3DPrintBoard, 2015-12-22:
> I think that we will be moving toward a scheme that creates a pipeline of transformations in the conversion from Object Coordinates to Frame Coordinates.  There is no reason why we cannot first [...]

Duly noted.   If I start playing with any of this, I'll make sure the transformations are isolated and can easily be moved (or changed) to what ever the final code path is.

> By configuration option, either of these transformational corrections can be omitted and, mathematically, replaced with an identity transform.

I understand.   Quick question...   Can these transformational corrections be collapsed into a single correction?   Rather than do lots of math at each point in the process, if multiple corrections are in play can this easily be collapsed into a single correction to speed up the math?

>That's the way we do it in robotics because it is much less error prone than doing the math separately everywhere you need it. Is it really necessary/meaningful to have a non linear transform? 

Yes.  It is really necessary.   Any errors in the measurements and calculations for Delta's results in a bowl or dish shaped bed.   So tilting the bed to 'level' it doesn't achieve what is needed.   A Mesh based approach seems to be the right answer.  But our current Mesh based 'leveling' for Delta's is still pretty crude.   It needs a lot of improvement.

Louis-Cypher, 2015-12-22:
I do not get that completely. The transformation of the delta itself is 
nonlinear, but the transformation for the tilted bed still is a linear 
one. If it isn't you are deforming the object, right?

> That's the way we do it in robotics because it is much less error
>     prone than doing the math separately everywhere you need it. Is it
>     really necessary/meaningful to have a non linear [...]

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Roxy-3DPrintBoard, 2015-12-22:
> I do not get that completely. The transformation of the delta itself is
nonlinear, but the transformation for the tilted bed still is a linear
one. If it isn't you are deforming the object, right?

Yes.  That is correct.  And I think what Wackerbarth is saying is we need a general purpose pipeline of corrections that can correct one thing and pass the results on to the next stage of the process to correct what it knows how to do.

In the case of Delta's, the movement across the bed is not flat once you get a little ways from the center of the bed.  That would be a non-linear correction.  Currently that is done with a Mesh based approach.   But it is also easy to envision the Delta's bed being tilted.  That would be a linear correction.

Louis-Cypher, 2015-12-22:
So, what I try to say is that there is a linear transform (correction) 
from the tilted COS to the cartesian TCP COS (nozzle). Additionally 
there is a transform from joint angles to TCP COS (cartesian) or the 
other way round. The latter is dependent of the setup (delta, 
core-whatever), the first isn't.

Hence, my thinking would be do the correction transformation identically 
for any printer (which means only once) and then transform it to joint 
angles via the kinematics dependent tranformation.

Doing so you would end up having two tranformation matrices or whatevere 
you want to use in the pipeline. Doing all in one is again pretty error 
prone to my understanding. Any ideas about that?

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

KiteLab, 2015-12-22:
Every time a Frame Coordinate is calculated it is a multiplication of the Object coordinate with the matrix. If any offset, tilt angle, ... changes, the matrix is recalculated.

@Louis-Cypher 
It's a question about your perspective.
Normally we assume our printer is perfect and our bed is tilt but flat. Thats a thing we can correct with a simple rotation matrix.
If our measurements tell us about a not flat, but (for example) saddle shaped bed, but we know this piece of glass is flat, the assumption about the perfectness of the geometry of our printer is wrong. Then distorting the Object space makes sense to get undistorted objects..

Louis-Cypher, 2015-12-22:
@Louis-Cypher <https://github.com/Louis-Cypher>

> It's a question about your perspective.
> Normally we assume our printer is perfect and our bed is tilt but
> flat. Thats a thing we can correct with a simple rotation matrix.
> If our measurements tell [...]
Agreed, but isn't the object then distorted by the tranformation itself? 
Meaning you get a perfect first layer but the rest of the object is also 
completely deformed (in whatever shape!). Is it a good idea to do so? 
The object might be massively distorted the further you get away from 
the bed.

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Wackerbarth, 2015-12-22:
"Agreed, but isn't the object then distorted by the transformation itself?" -- No, the object is not distorted.
What you are distorting is the coordinate system of the nozzle. And the "distortion" that you are applying is to cause a correction in the actuator coordinate system so that the nozzle is positioned properly in the global space.

Louis-Cypher, 2015-12-22:
How do you realise a transformation that deforms the target coordinate 
system but not the object? If the object is planar it doesn't fit onto a 
warped heatbed, right? Hence, the transformation is multiplied with the 
object cooridnates, right?

Is it a translation that you are using for correction or a full 
transformation (meaning one that maps a plane to the warped heatbed)? 
The first would just transform any layer to the shape of the heatbed, right?

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

thinkyhead, 2016-03-11:
How is this looking today? We should be at a much better place with `RCBugFix` at this time. Now is the time to test it, because we're trying to get another release candidate out to the world, and it would be great to have all known issues cleared up.

bkjnky, 2016-03-16:
Could this bug apply to machines that have the print bed moving in Z rather than just CoreXY machines?  I seem to be having a similar with bed leveling on my printer.  What information can I provide to assist in identifying?

Blue-Marlin, 2016-03-16:
My first question would be, if your printer has a right handed coordinate system.
The second one would be about your version/branch of Marlin.

No. It does not matter if the nozzle of the bed is moved.

whosawhatsis, 2016-03-16:
@bkjnky The controller actually has no idea which axis is moving what in a vanilla cartesian bot. Whether your platform is going up or your extruder is going down, it all looks the same to the firmware. To anthropomorphize, the printer always "thinks" that the extruder is moving relative the the platform, regardless of which is actually hooked up to the motor(s). Any time the platform moves, the printer "thinks" that the platform is remaining still and the tool is moving in the opposite direction, so "machines that have the print bed moving in Z" is not a distinction it makes.

Blue-Marlin, 2016-03-16:
@Louis-Cypher 
You are right. 
But if the distortion really matters - use a flat bed.
If the error is not too big, here is an other way to make it right. One could use a raft, printed with a z-error dependant over/under-extrusion. But - who wants a raft?

bkjnky, 2016-03-16:
@Blue-Marlin 

0,0 is Front left 
Homing is x,y,z min

version is 1.1.0-rc3

bkjnky, 2016-03-16:
@whosawhatsis 

understood.  I think the confusion for me is originating from the odd behavior, coupled with little understanding of the full complexities of how leveling and homing/homing direction interact to produce the new plane in the firmware.

Roxy-3DPrintBoard, 2016-03-16:
@whosawhatsis   Do you realize your icon looks like 'Tom' in the BlackList TV Series?

thinkyhead, 2016-03-27:
Please do all testing with `RCBugFix` which has been patched a lot since `RC3`. If this is largely just an issue of fine-tuning, adjustment, and testing, then feel free to close this issue.

closed 2016-04-28 by jbrazio