MarlinFirmware/Marlin #2851: X stepper direction on dual X carriage issue

barus93, 2015-12-13:
hi, 
I update to this release but i have a problem with the stepper direction.
when i switch tool the first X stepper's movement is performed in the opposite direction.
I solve changing the 567 line in stepper.ccp from
  if (current_block->direction_bits != out_bits) {
To 
if (current_block != NULL) {
taken from an older working version

KiteLab, 2015-12-15:
__If__ `if (current_block != NULL) {` is the right condition here, it is already tested in line 618 (`if (current_block)`) and can be omitted.
`if (current_block->direction_bits != out_bits) {` is mend to avoid setting the direction output pins when they are already in the target state. The problem here is, that `current_block->direction_bits` contains information about one X, one Y, one Z and one Extruder axis, but not which X-stepper or E-stepper is mend. In other words - that can't work if the x-stepper or e-stepper was changed.
I suggest deleting the `if (current_block->direction_bits != out_bits) {` decision and execute 
```
    out_bits = current_block->direction_bits;
    set_stepper_direction();
```
unconditional.

bmarl, 2015-12-18:
Removing the condition did not change the situation for me

thinkyhead, 2016-03-02:
@bmarl There have been a lot of patches —some related to Dual X— that have been recently committed to `RCBugFix`. Give that branch a try, if you haven't already.

bmarl, 2016-03-02:
My problem was resolved a while ago. it was actually my own error, i interpreted the parameters incorrectly.

thinkyhead, 2016-03-05:
I'm lgad to hear this is resolved. Documentation should be improving over time. Sorry for the inconvenience!

closed 2016-03-05 by thinkyhead