MarlinFirmware/Marlin #2852: Why isn't there a Marlin folder at RepRap.Org???

Roxy-3DPrintBoard, 2015-12-13:
How can this be???   If you go to:

http://forums.reprap.org/index.php?140

There is no Marlin folder.   Who knows somebody over there to get a folder added?

MichaelBrock, 2015-12-14:
They have a separate set of folders for firmware. Still nothing specific to Marlin (or any other firmware for that matter):

http://forums.reprap.org/index.php?145

Roxy-3DPrintBoard, 2015-12-14:
> They have a separate set of folders for firmware. Still nothing specific to Marlin (or any other firmware for that matter):

>http://forums.reprap.org/index.php?145

Oh!!!   I saw Repetier in that list and just assumed it was for the firmware side of that.   And at that point, it was like "Wait a minute!!!"

ntoff, 2015-12-19:
http://forums.reprap.org/index.php?145 and I would assume it'd be in "mainstream"

thinkyhead, 2016-03-02:
Marlin folder now at http://forums.reprap.org/list.php?415

closed 2016-03-02 by thinkyhead