MarlinFirmware/Marlin #2853: Question: what should Z axis position be after G28

MichaelBrock, 2015-12-14:
I am having auto-bed leveling issues with my Corexy.  It either isn't compensating, or isn't doing it very well.   I'm trying to figure out the best way to debug it but I"m having trouble understanding some of what I'm seeing.   For example: G28 re-sets the auto-bed leveling matrix so no compensation is applied.   After I do G28 (without specifying axes, so all three are homed), the Z-axis position is reported as -1.3  (via M114).   Shouldn't this be 0, or the Z_PROBE_OFFSET_FROM_EXTRUDER (which is -3.7)?

Roxy-3DPrintBoard, 2015-12-14:
Well...  Part of the problem is we don't have a detailed specification to define the behavior and especially not when we start talking about innovative machines that do things differently.   Pretty much, the G28 numbers have always varied from the G29 numbers.   It would seem they should be in agreement but they never have been.  

At the root of the problem is G28 uses the HOMEAXIS() macro that translates into a homeaxis(Z_AXIS).  This function does not do any 'corrections' but it does modify the position by the home offsets.   G29 doesn't really care about the Z Position other than it is going to set the nozzle to be at Z_PROBE_OFFSET_FROM_EXTRUDER.

These GCodes are doing different things.   But if it bothers you...  You can use M428 to get your Z correct between the two functions and save the offsets with a M500.   I've never done this, but looking at the code it would appear that will take care of your issue.

MichaelBrock, 2015-12-14:
I will look into M428, thanks!     Let's assume that I am not doing auto-bed leveling.  After I do a G28, what should the z level be set to?  I.e. what does it represent?   My expectation would be that it finds the zero point by probing and finding the location at which the probe is triggered and then setting that position to z=0 (and I assume that that this is the position in which it rests at the end of the probe) then adjusting that value by the Z_PROBE_OFFSET_FROM_EXTRUDER.    So a subsequent M114 would show the current position as the probe offset.

Roxy-3DPrintBoard, 2015-12-14:
Remember...  We don't have a detailed specification that defines everything.  Originally, G28 homed all of the axis's.   Printers used limit switches to do the home instead of a Z-Probe.   G28 never has had the concept of a Z-Probe and a Z-Probe-Offset.

I would say when you home the machine and then slowly lower the nozzle so it just barely touches the bed, that should be 0.00 mm.    It is very possible other people are going to disagree with this and it might be interesting to hear why they disagree.

MichaelBrock, 2015-12-14:
I just noticed that in the code.  The Z_PROBE_OFFSET_FROM_EXTRUDER is added to the Z home value only if auto bed leveling is enabled.   So clearly, not intended that the Z-probe is used as an end stop.

Roxy-3DPrintBoard, 2015-12-14:
> I just noticed that in the code. The Z_PROBE_OFFSET_FROM_EXTRUDER is added to the Z home value only if auto bed leveling is enabled. So clearly, not intended that the Z-probe is used as an end stop.

Yes, from a historical perspective that is true.  But the code has evolved to allow the Z-Probe to be used as an endstop and in fact, you can have 'dual' Z Axis endstops.   I've never used those modes.  And you have to be careful when you set them up.  But they do work.

KiteLab, 2015-12-15:
It's a 'cause and effect' problem (like on many other places in Marlin).
When the ABL was introduced a probe was needed. All the probe related stuff was made dependent on 'AUTO_BED_LEVELING_FEATURE' (or whatever the name was then). That was a acceptable design decision then. Since we have an alternative use of the probe today (homing) - from today's perspective 'AUTO_BED_LEVELING_FEATURE' should be only possible if a probe is defined.
Today sorting out what code is 'only' probe related and what is 'only' the use of the probe is a mayor task. Some ideas how to go in that direction are around (unified probe handling - somewhere in the 'Hardware Endstops' thread https://github.com/MarlinFirmware/MarlinDev/issues/285). But soon implementation is unlikely.

AnHardt, 2015-12-16:
Back to the title.

From set_axis_is_at_home(AxisEnum axis)
```
...
  {
    current_position[axis] = base_home_pos(axis) + home_offset[axis];
    min_pos[axis] = base_min_pos(axis) + home_offset[axis];
    max_pos[axis] = base_max_pos(axis) + home_offset[axis];

    #if ENABLED(AUTO_BED_LEVELING_FEATURE) && Z_HOME_DIR < 0
      if (axis == Z_AXIS) current_position[Z_AXIS] -= zprobe_zoffset;
    #endif
  }
...
```
`base_home_pos` set by HOME_POS`
`home_offset` set by EEPROM, M206, M428
`zprobe_zoffset` set by Z_PROBE_OFFSET_FROM_EXTRUDER, EEPROM, M851

So, maybe you have some unexpected values in the EEPROM.

thinkyhead, 2016-03-02:
There were some bugs in CoreXY / CoreXZ related to homing and endstop position. These have been patched in `RCBugFix` and will be included in `1.1.0-RC4`. Give `RCBugFix` a try and see if you get better results.

closed 2016-05-04 by jbrazio