MarlinFirmware/Marlin #2855: Issue with AUTO_BED_LEVELING_GRID

joshuarowley42, 2015-12-15:
I have been banging my head against a wall over this for 2 days now and would very much appreciate some advice. With any luck fixing my issue will be something trivial, either way Marlin could either handle the problem a little better or needs a patch somewhere. 

I have created a repo for the exact environment that I am using. Hopefully this will help show the differences between stock configuration and mine. This can be found at: https://github.com/joshuarowley42/MarlinAutolevelBug

## The setup is as follows: 
* Standard Cartesian Bot
* 200 x 200 Build Volume
* Z-Probe Exactly coincident with nozzle in X & Y (not in reality, but for sake of this test)
* Z-Probe works anywhere in build area, including 0,0

## Symptoms:
### G28
When homing using G28, the motion system homes in X and Y and stops without homing Z, showing a seemingly random XY location on the LCD. I later discovered that the random XY location was actually the destination the printer was attempting to move to (bed center) before homing Z. 

The move to the `Z_SAFE_HOMING_X/Y_POINT`  is placed in the `destination[X/Y_AXIS]` which is reflected on the LCD, however the axis do not move. Setting this point to 0,0 as well as the probe offset (as seen in the repo) makes the problem go away - however when using a G29 the issue resurfaces....

### G29
After homing with G28, issuing a G29 command has no effect. As before, the LCD reflects the last probing location, but the axis did not move. 

## Debugging Steps Taken:
I have read through a fair bit of the code to try and figure out where this is happening. I got as far as `do_blocking_move_to(x,y,z)` in Marlin_main.cpp and added some debugging messages there. The output of which can be seen here:

[image]

Clearly the `plan_buffer_line` and `st_synchronise` commands are being executed, but for some reason they are not actually materialising into machine motion. Looking at the time-stamps in the screenshot it can also be seen that they take no time!

The final display on the LCD after this flurry of activity is X170, Y20, Z1, however the motion system has not moved.

## Other Testing Info
Doing a G28 followed by G1 X100 Y100 moves the axis to the expected location. 
If once at X100 Y100 G29 is issued, the axis move to Y0 then stop, again with the LCD showing X170, Y20. 

Note the delay highlighted in Red when the XY Move from Y180 -> Y20.
[image]

To me it is very mysterious, and the fact that no one else has been seeing this issue makes me want to think it is a hardware problem or some stupid configuration mistake. Maybe it is something to do with endstops? Any pointers would be very much appreciated indeed.

brainscan, 2015-12-15:
> [Uploading RepetierOutput2.png…]()

You haven't defined bed levelling and why is your baud rate so strange? Also might need to disable max end stops and min software end stops.

joshuarowley42, 2015-12-15:
Thanks Brinscan - you have saved what is left of my hair!

I can confirm that `DISABLE_MAX_ENDSTOPS` did the trick. 

Initially I also set `min_software_endstops` and `max_software_endstops` to false. But re-enabling them did not cause the levelling to break. 

I had disabled `AUTO_BED_LEVELLING_GRID` to see if it would help anything when I was debugging (it didn't). It's back on now. 

Baud rate - highest supported as standard by Linux kernels. I stand to be corrected on that, but I don't see it as a problem. 

It would have been great for there to have been some more debugging information about what caused those motion commands to fail. Some end-stop errors I am sure would have enabled me to figure the problem out without having to pester you guys!

Roxy-3DPrintBoard, 2015-12-15:
> Baud rate - highest supported as standard by Linux kernels. I stand to be corrected on that, but I don't see it as a problem.

The problem with a super high USB Baud rate is the processor gets hammered every time there is space for a new GCode command.   By lowering the baud rate you let other stuff get a small amount of processor time to do things.   And because the GCode buffer is multiple commands deep, the processor never runs out of commands to do.   It depends on the printer, but many printers are better off with the lower baud rate.

> It would have been great for there to have been some more debugging information about what caused those motion commands to fail. Some end-stop errors I am sure would have enabled me to figure the problem out without having to pester you guys!

Did you see:

uint8_t marlin_debug_flags = DEBUG_INFO | DEBUG_ERRORS  | DEBUG_LEVELING ;

In Marlin_main.cpp?   If you turn on the debug stuff you get a fair amount of control over what stuff gets printed out.

Wackerbarth, 2015-12-15:
Rather than unconditionally turning on debug flags at compile time, you should be using the M111 command to change the value at run time.

Roxy-3DPrintBoard, 2015-12-15:
> Rather than unconditionally turning on debug flags at compile time, you should be using the M111 command to change the value at run time.

Agreed...   That comes from an over-reaction to the fact I have to make sure DEBUG_LEVELING_FEATURE is turned on in the source.   So if I need it, I just brute force it.

brainscan, 2015-12-15:
Glad I could help.

joshuarowley42, 2015-12-15:
Thanks for the suggestions.

I had not noticed that debug option to be honest. I thought that I had all
the info I was going to get with the debug option set for auto_levelling. I
will turn it on next week and see if it would have helped in my particular
situation.

KiteLab, 2015-12-15:
[image]
All the needed info to debug was there.
endstops hit on all 3 axis moving from [0,0,somewhere] to [mid,mid,Z_RAISE_BEFORE_HOMING], means triggered max endstops while far away from them - so probably always on.

A single M119, in mid air, would have told the complete story.

__No unused endstop shall be triggered.__

Roxy-3DPrintBoard, 2015-12-15:
@KiteLab 

Good Job!  Everybody missed that!!!

joshuarowley42, 2015-12-16:
Indeed! Makes me feel particularly silly, but as they say - hindsight is
20:20.

thinkyhead, 2016-03-02:
I'm going to add some more dire warnings that users should enable `DISABLE_MAX_ENDSTOPS` unless they actually have one or more Max Endstops. Perhaps in future it would be better to specify what all the endstops and probes are in a more explicit way, then the code can be far smarter about how to handle them. For example:
```cpp
#define X_MIN_ENDSTOP_EXISTS
//#define X_MAX_ENDSTOP_EXISTS
#define Y_MIN_ENDSTOP_EXISTS
//#define Y_MAX_ENDSTOP_EXISTS
//#define Z_MIN_ENDSTOP_EXISTS
#define Z_MAX_ENDSTOP_EXISTS
```
Then these troublesome options can go away forever:
```cpp
//#define DISABLE_MAX_ENDSTOPS
//#define DISABLE_MIN_ENDSTOPS
```

closed 2016-03-02 by thinkyhead