MarlinFirmware/Marlin #2856: Extrude while option os true

CoderSquirrel, 2015-12-15:
I made an callback item at the menu and while this callback'd be true the printer must keep extruding, but it only moves one time.
I'm using plan_buffer_line to change the extruder positon and make it move and i'm making it at the ultralcd.cpp.
MY first shot was use enquecommand_P but it make a querie of commands, and make  a delay when i turn the option off, so i'm using  plan_buffer_line because once it stops it stops, at least i think.
How could i make this work? :/

KiteLab, 2015-12-16:
Pardon.
I don't understand the question. Please explain again. For what purpose you want to do what?

CoderSquirrel, 2015-12-16:
i'm helping a friend to make an extruder, he makes filaments, and what he needs is:
a option, like a callback that is on and off
when its on it must keep extrusing until turns off again
but when it turns off it must straight stop.
I cound make it thru the menu using enquecommand_P with G1, but it keeps a buffer and only stops once the buffer its empty.
So i saw the way it works to make a move (like in move step) and saw that it uses plan_buffer_line, so i tried to use it, but it didn't worked  at all.
It's more clear?

CoderSquirrel, 2015-12-16:
I managed to make it now doing
`
 START_MENU();
  MENU_ITEM(back, MSG_CHANGE_FILAMENT, lcd_change_filament);

  MENU_ITEM_EDIT_CALLBACK(bool, "Iniciar", &extrudeFilament, lcd_slow_filament);

  END_MENU();
  /*
    Funciona mas da alok */
 

   if (extrudeFilament) {
    current_position[E_AXIS] += 50;
    plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], manual_feedrate[E_AXIS] / 60, active_extruder);

  }
`
it works and keep extruding, don't stop, but it freeze the lcd

Roxy-3DPrintBoard, 2015-12-16:
I'm not sure Marlin is the right starting point for what you are trying to do.  But the LCD Panel only gets updated when the lcd_update() function is called.   You need to call it periodically to get new information on it.   For example, the idle() function gets called when there is nothing happening.  It calls lcd_update().
~~~~
void idle() {
  manage_heater();
  manage_inactivity();
  lcd_update();
}
~~~~

KiteLab, 2015-12-21:
This has some interesting aspects.
Using a single long move is difficult to stop.
Using a single short move will fill up the planner buffer - will also take a while to stop.
Using a single short move with stsync() will have accelerations/decelerations between the moves

while (do_extrude && movesplaned() < 3 ) do_a_short_move();

somewhere in the main loop could do a good job.

thinkyhead, 2016-03-02:
@KiteLab That's the solution we keep coming back to for all axis motion from the LCD menu. We should remove all calls to `plan_buffer_line` from the LCD menus and instead set a `manual_destination` global that gets checked in the `idle()` routine.

CoderSquirrel, 2016-04-18:
Hey @KiteLab @Roxy-3DPrintBoard @thinkyhead  here i am at this again.
My meaning with this question is, is not from the lcd, its from marlin_main. 
But the thing is, what is the correct way to move the motor, by myself.
I was using gcode, as "G1 E1 F3" with the enquecommand but i don't want to use the gcode.
I tried
```
 plan_buffer_line(destination[(X_AXIS)], destination[(Y_AXIS)], destination[(Z_AXIS)], destination[E_AXIS+10], feedmultiply/60/100.0, active_extruder);
  st_synchronize();
```
But has no effect at the motor

CoderSquirrel, 2016-04-18:
i change to
```
 current_position[E_AXIS] += 5.0;
  plan_buffer_line(current_position[(X_AXIS)], current_position[(Y_AXIS)], current_position[(Z_AXIS)], current_position[E_AXIS], 100.0, active_extruder);
  st_synchronize();
```

closed 2016-04-18 by CoderSquirrel

thinkyhead, 2016-04-19:
`G1 E1 F3` would be awfully slow. How about something more reasonable like `G1 E1 F180`? Feedrate is in _millimeters per minute_.