MarlinFirmware/Marlin #2857: Thermal Runaway always on same spot

pbrgle1, 2015-12-15:
I'm having this thermal runaway issue and it's driving me crazy. I'm trying to print this LCD Housing with my dual X Carriage printer using 2 materials and the print always stops on the same spot. I am using the second extruder for infill and the thermal runaway happens always about 30s into the first Solit Top Layer which is using the main extruder. 

At first i thought it was an issue with Octoprint but now the same happened using Simplify3D host. I am very close to disabling the thermal runaway protection alltogether... Is there a known fix for this issue? I see a lot of people are having thermal runaway issues. 

Here are some pictures:
[image]
[image]

Roxy-3DPrintBoard, 2015-12-15:
Before you disable it...   Try changing the Configuration_adv.h settings to relax things a little bit.  On my printer I had to change:
~~~~
//  #define WATCH_TEMP_PERIOD 16                // Seconds
//  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius
~~~~
to 
~~~~
  #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
~~~~

KiteLab, 2015-12-16:
Dear @Roxy-3DPrintBoard 
again you are suggesting to tune the wrong parameters.

Tune WATCH_TEMP_PERIOD and WATCH_TEMP_INCREASE against "Heating failed".
Tune THERMAL_PROTECTION_PERIOD and THERMAL_PROTECTION_HYSTERESIS against "Thermal Runaway"

Roxy-3DPrintBoard, 2015-12-16:
Yeah...  OK...  I'll be more careful.    But the truth is, that fixed my problem.  When my 2nd extruder turned off, it starts dropping in temperature.   When I turn it back on, it shows that it is still dropping for a while.  There is some lag between when the heater is turned on and the thermistor sees an increase in temperature.

closed 2016-03-02 by thinkyhead