MarlinFirmware/Marlin #2858: marlin configurator

erteclas, 2015-12-16:
can't parse boards.h when i upload the file

KiteLab, 2015-12-16:
The configurator is not ready yet - and it's questionable it it will be finished, since development stopped.
For an alternate way for a (simplified?) configuration see MarlinFirmware/MarlinDev.

thinkyhead, 2016-01-29:
I've moved development to: https://github.com/thinkyhead/MarlinConfigurator

With an all-new configuration system being developed for Marlin 1.2.x, I'm not sure what the fate of this "configurator" will be. It was off to an interesting start, and I'd like to continue to develop the core functions which read C++ compiler directives, make sense of them, and then wrap them up in a GUI. Once that core stuff is more complete, it will be easier to apply to any future compiler directive-based configuration system that Marlin might use.

closed 2016-01-29 by thinkyhead