MarlinFirmware/Marlin #2859: Z Homing not working (motor humming)

TheNuubi, 2015-12-16:
First: Its not a hardware problem, the movements work fine, motor is strong enough.
I am having weird issues with my z axis homing. I can go to the endstop manually (not homing) without problems. However if i home (repetier host) the motor just humms. Endstop and movement are callibrated correctly (like x and y which are working). I tried disabling safe homing and zprobe (i dont have one).

KiteLab, 2015-12-16:
Probably your
```
#define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)
```
is set to high.

Kriechi, 2015-12-20:
I had the very same problem with my Prusa i3 - the default speed was just too high. I reduced it to `2.5*60` and now it homes properly.

thinkyhead, 2016-03-02:
I see this question a lot. Both the Z homing and manual feedrates need to be set to a reasonable value or the Z steppers can't handle the load.

closed 2016-03-02 by thinkyhead