MarlinFirmware/Marlin #2861: Dual X - Direction for T1 travel incorrect

bmarl, 2015-12-18:
This is a support request.
Using RC3 i try to use dual x carriage in mode 1. Home does work for both carriages, x1 homes to x-min and x2 homes to x-max. Also X1 travels correctly when extruder 0 (T0) is selected. If i switch to extruder 1 (T1) it tries to travel in positive direction from it's home position (260mm). It should travel in negative direction.

KiteLab, 2015-12-18:
#2851

bmarl, 2015-12-20:
Any further ideas? The linked issue 2851 is either a different problem, or the solution suggested does not work. I am grateful for any help, i am stuck in the middle of a bigger project due to this.

caenan0, 2015-12-21:
After tool change, if you tell it to move in the positive direction and then move in the negative direction again, does that fix the problem?

bmarl, 2015-12-21:
Thanks caenan0 for the suggestion.
I changed to mode 0 to get full control and disabled endstops only for homing.
This is what happenes.
After starting the system with X0 and X1 at their home positions (XMin and XMax)
T0 then G28 X0
First X1 moves away from x-max and back, then tries to move beyond xmax and loses steps until the command finishes. Afterwards the same happens for X0.
T0 + G0 X10 correctly moves X0 to position 10
T1 + G0 X10 results in no movement and error endstop hit at position X20.01
The X position is now shown as 20.01
G0 X10 results again in the error endstop hit at position X20.01
X position is now shown as beeing 10
T0 + G1 X5 correctly moves X0 to position 5

It seems to me the initial behaviour after the first correct homing moves is the main symptom.

pbrgle1, 2015-12-22:
I'm having a similar issue. Everytime i start a new print the Z axis moves down (+) even though there are no instructions in the gcode for that. Sometimes it moves 100mm sometimes 200mm and sometimes even the Y moves in the + direction and then goes back again. The problem is that when the Z moves down it doesn't come all the way up again and leaves a small gap between the nozzle and bed. Also the tool starts in the wrong X direction. After restarting the same print it works fine but there is still a slow Y movement before the print starts (but it immediately goes back to 0).

bmarl, 2015-12-22:
pbrgle1: I did not have Z connected before, but after i did i have no problem similar to yours. Z and Y axis behave completely normal and correct, it is only X that causes any trouble.
So maybe the two problems are not related

bmarl, 2015-12-22:
Maybe someone can clarify. If i switch from X1 to X2 (T1), should the positional information (M114) change to the current position of X2 or shall it still show the old position of X1? For me it always only shows the same position.

caenan0, 2015-12-22:
@pbrgle1, this happens sometimes and I haven't gotten around to nailing down the root cause yet, but in the meantime, a full reset before starting the next print will solve the issue. 

@bmarl, any reason you're using mode 0 instead of 1? My printer is busy for the next 30 hours so I can't do testing but I've never had the M114 not changing issue, and strangely what fixed the direction issue for me is properly setting the "Travel limits after homing" under config.h. I defined X min as a small negative value instead of 0.

bmarl, 2015-12-22:
I switched to mode 1 and it seems that it over all behaves more predictable. X1 basically works, but X2  will not move due to it beeing in xmax position while showing position 0. 
If i hard set (G92) x to a realistic value (Xmax or lower) i can move X2 using G0 X60 or something like it.
The single remaining problem seems to be that the X2 position is simply remembered incorrectly after switching the extruder.

bmarl, 2016-02-01:
Some searching later i found line 2400 in marlin_main.cpp, which assigns a value to the x position of the inactive extruder that makes no sense at all. The line seems out of context. I removed it and to my surprise i now get the correct value for the x position of carriage 2 after homing.
Can anyone clarify what line 2400 does? 
`       inactive_extruder_x_pos = current_position[X_AXIS];`

bmarl, 2016-02-02:
Thanks to the help of Lucca Baruzzo i am now fairly sure that the problem is indeed somewhere in my settings and not in the software itself. Is there any place where i can find more information regarding how the settings for Dual X Carriage work?

bmarl, 2016-02-02:
Problem solved, it was the fact that i enabled EXTRUDER_OFFSET_X
The issue can be closed

closed 2016-02-03 by Wurstnase