MarlinFirmware/Marlin #2865: Dual Extruder Prints in wrong location after tool change

SuperSleeper, 2015-12-19:
Hi Folks,
New to 3D printing and not exactly sure how to word all of this as I don't have all the lingo down.  I happen to know I'm running Marlin on a Delta Mini Rockstar Q2s Pro.  Unfortunately, I don't have a reference to the version I'm running, but I downloaded it in mid October 2015 and files are dated 10/21/2015.  Unfortunately the notes section of the Marlin.ino file contains no version specific information.

Issue I'm having is only when executing code for both extruders.  I can print perfectly fine with either extruder alone without issue, it just tends to go off to lalaland (reverses Z axis) and print in the air only after switching between extruders.  If I pause the print and re-home the printer, it will then start printing on the second extruder where it's supposed to.

I'm using Repetier Host, and after running Slic3r, I see both filaments aligning properly in the print preview.  Therefore, it leads me to believe there must be some firmware issue.

Thanks in advance for ANY direction here.

docpayce, 2016-01-03:
I cant find your printer online, can you give a source on it? Is it this one? http://www.geeetech.com/wiki/index.php/File:Dual_extruder.jpg

In the meantime, pls check your Extruder settings in Slic3r: Switch to expert settings, printer tab and check the first and second extruder offsets. Also, did you try another Slicer? Kisslicer for instance?

Also, you could try to change TOOLCHANGE_PARK_ZLIFT and TOOLCHANGE_UNPARK_ZLIFT in configuration_adv.h. Set them to 0. I dont *think* thats the problem tho.

Roxy-3DPrintBoard, 2016-01-04:
@SuperSleeper  Can you move your Configuration.h file over to the current Release Candidate (RC-3) and try it there?   If it happens there, it will get fixed ASAP.

SuperSleeper, 2016-01-04:
Thanks for the tip, I was considering testing RC-3.  I'll test it and let
you know.

thinkyhead, 2016-03-11:
Hi @SuperSleeper, is this still an issue for you? We're working a new release candidate and want to make sure bugs are being fixed.

SuperSleeper, 2016-03-11:
Thanks for the follow up. My PC was offline for a while and hadn't had a
chance yet to upload the new firmware.

thinkyhead, 2016-03-24:
Sorry we didn't get this followed up in time for RC4, but give that new release a try when you get a chance. https://github.com/MarlinFirmware/Marlin/releases

closed 2016-04-28 by jbrazio