MarlinFirmware/Marlin #2866: RRD Fan Extender rapms 1.4

erteclas, 2015-12-20:
How to configure with ramps 1.4 EEB board selected for use rrd fan extender as fan layer.

KiteLab, 2015-12-21:
https://github.com/MarlinFirmware/MarlinDev/issues/284

erteclas, 2015-12-26:
i read all thread and no solution with marlin 1.02 or 1.1.0 rc2... with m42 p6 or p11 s0 or 255 nothings happend..
always on

thinkyhead, 2016-03-11:
@erteclas Did you have any luck? I haven't read this yet, but check out this Instructables article on [Configuring and using a RepRap RAMPS 1.4 RRD Fan Extender with Marlin](http://www.instructables.com/id/Configuring-and-using-Reprap-Ramps14-RRD-Fan-Exten/)

Let me know if you need any help integrating the code from the article with the newer Marlin.

SrGeek, 2016-04-06:
The article helps explains how to setup the RRD fan so that the fans turn on after the hotend or any other component of your choice reaches a designated temperature.  However you cannot control the fan with the 12864 LCD reprapdiscount smart controller or repetier host.  Unless you manually enter speed commands in gcode.

thinkyhead, 2016-04-08:
That is correct. Extruder Auto Fans are only automatic, and intended for cooling the hot end above the heat break. They are not the same as the GCode-controlled PWM fans which are intended to cool the printed object, the speeds of which are set through your slicer options.

closed 2016-05-04 by jbrazio