MarlinFirmware/Marlin #2867: Extruder values go out of range

gmossy, 2015-12-22:
Testing Marlin RC3BugFix,  a print was going for 2hours, and started clicking.  I canceled the print from cura and when to play with the Extruder control on the LCD, and the values were out of range, and funny characters,  i.e..such as *?--.    I have seen this problem a number of times now, also on RC2.  
3UP printer, RAMPS 1.4, Marlin RC3BugFix, EDV6 Extruder. 
Glenn

gmossy, 2015-12-23:
Additional erratic behavior of Extruder control, after a print was stopped, Extruder read +660 then I rotated the extruder to remove filament and Marlin reloaded.

bradanlane, 2016-01-21:
I may be seeing similar:

My print ran for approximately 20 minutes and extruder started clicking. The flow became erratic and under extruded. I will run the test again and check the LCD values. 

If I run the same gcode print with Marlin 1.0.? (jcrocholl) I complete the print without the extruder clicking.

bradanlane, 2016-01-22:
I ran two tests today and I am not seeing the corrupted extruder values that you mentioned. I did see that the values reported through the LCD did not match what I had in my configuration. This is the second case where I have seen EEPROM values not match what were set in the configuration.h file. (I do have `#define EEPROM_SETTINGS`)

bradanlane, 2016-01-25:
@gmossy - when you mentioned "**funny characters, i.e..such as *?--.**" were refering to the entire LCD panel or just extruder settings?

I ask because I was printing with RC3BugFix yesterday and - while the print completed successfully - the LCD panel showed the following display for the at least last hour (of a two hour print)

[image]

gmossy, 2016-01-25:
The characters were only on the Extruder motion adjustment screen.  ?_.? etc.  and I've only seen it on RC2 so far.  Never had any problem the LCD panel using RC3BugFix so far.  
As far the EEPROM settings, I think there is an issue and I'll examine it closer.  Mine didn't match the configuration.h, I had set the values to lower values before I flashed the firmware to RC3BugFix and then the new firmware didn't update them.  I manually did a memory default and save from the LCD screen, but I had assumed that was expected behavior.  I believe uploading a new arduino sketch won't clear the eeprom and I"m familiar with having to run an eeprom sketch to clear out the memory.  Question to the developers here what behavior is expected for the EEPROM.

bradanlane, 2016-01-25:
I am also seeing the EEPROM settings not updating. I'v reverted to using an **auto0.g** file on the SD card to insure all important settings are correct for the printer.

KiteLab, 2016-01-25:
@gmossy 
The number to string conversion currently used for displaying the E axis is `ftostr31()`.
This is good for numbers between -999.9 and +999.9. If the current_position[E_AXIS] value is smaller or larger garbage may appear on the display. That does not explain other malfunctions.
current_position[E_AXIS] is a float - can be nearly infinite big - is nan absolute value - but usually is reset to 0 by a G92 per layer by the slicer. Nerveless it can be much larger than 999.9.

There was a real error in that area of the menu, fixed in the current RCBugFix.

In the current RCBugFix you could additionally try to replace `ftostr31(` with `ftost51(` in the lines 836 and 873 in ultralcd.cpp

@gmossy & @bradanlane 
The EEPROM setting overrule the config. settings by purpose. To store your new config. setting in the EEPROM use:
>M502 - Revert to the default "factory settings". You still need to store them in EEPROM afterwards if you want to.
M500 - Store parameters in EEPROM

Than check with 
>M501 - Read parameters from EEPROM (if you need reset them after you changed them temporarily).
M503 - Print the current settings (from memory not from EEPROM). Use S0 to leave off headings.

bradanlane, 2016-01-25:
Thanks. My problem was misinterpreting the comment in the configuration file.

```
// EEPROM
// The microcontroller can store settings in the EEPROM, e.g. max velocity...
//define this to enable EEPROM support
```

I read the comments to mean "'set the #define to store configuration.h settings in EEPROM".

KiteLab, 2016-01-25:
@bradanlane 
Usually you can restore a corrupted status screen by clicking the encoder twice. Entering the menu contains a software reinitialisation of the display. 
Usually the corruption, your picture shows, is caused by, static electricity, or noise on the display lines, or a to tight timing, or a combination of these.
In some cases it needs a hardware reset, or power down, to come out of this.

bradanlane, 2016-01-25:
I will try to "double tap" if I see this again.

thinkyhead, 2016-03-11:
This should be all fixed now in `RCBugFix`. We had a stack corruption issue associated with axis movement from the LCD. Please give the latest code a try and let us know if it's still an issue.

closed 2016-04-28 by jbrazio