MarlinFirmware/Marlin #2870: Anormal change of engine speed

htaxil, 2015-12-23:
Hi, I'm using Marlin RC 1.1.0 RC2 with my Delta printer. I notice a problem of change of regime of engines during the ascents and the descents of carriers. I think hiccups ((short fall of regime) occur at regular intervals, listen the audio on this vidéo : https://youtu.be/l0B-ooFcSps
I upgraded to Marlin RC 1.1.0 RC3, but it has not solved anything.
I thought of a hardware problem, so I changed the electronics (Arduino, Ramps card and stepper drivers) unsuccessfully. I tried different intensity settings for the motors without success.
So I suspect a software problem ... any idea ?

htaxil, 2016-01-02:
nobody has an idea ?

KiteLab, 2016-01-02:
Lots off ideas here. But all pure speculation without more data about your configuration.
The about 1Hz rhythm lets me suppose a full graphic controller with 8 stripe refresh, like a VIKI2.

htaxil, 2016-01-04:
Many thanks for your answer. You're right, if I disable the LCD, the problem disappears.
Is there any solution? I do not find by browsing the issues list.

KiteLab, 2016-01-04:
Dispaytype?

htaxil, 2016-01-04:
I'm using a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

KiteLab, 2016-01-04:
Hmm, that's already the most optimized driver (needs only two stripes).
Maybe you can reduce the load by decreasing DELTA_SEGMENTS_PER_SECOND,
or a lower micro step setting.

htaxil, 2016-01-04:
Oh, that is not a good news.
I decreased DELTA_SEGMENTS_PER_SECOND, passing from 200 to 100. This greatly improves things, but I still have hiccups while printing curves. I will further decrease a bit, but I hope it will not degrade a print quality . I am very surprised ( and disappointed ) by this issue especially since you seem to say that this is one of the most optimized drivers. It might cause problems to many people.

AnHardt, 2016-01-12:
@htaxil 
Maybe you are interested in some graphic display speedup.
https://github.com/AnHardt/Marlin/pull/13

thinkyhead, 2016-03-11:
A Graphical Display plus Delta calculation pushes the limits of modest AVR boards. The only way to make a computer "go faster" is actually to make it do less. So I suggest checking out the work at https://github.com/AnHardt/Marlin/pull/13

closed 2016-03-11 by thinkyhead