MarlinFirmware/Marlin #2871: Why is EEPROM disabled by default in Configuration.h?

RetireeJay46, 2015-12-24:
This is not so much of a bug report or a request for a feature, but merely a question: why is the usage of EEPROM disabled by default?  I went through Configuration.h with a fine-tooth comb to set up Marlin to match my machine, but I never imagined that EEPROM would be disabled, so I didn't read that part of Configuration.h closely enough to realize I needed to change it.  (This is true both of the Release version right here and also of the MarlinFirmware/MarlinDev repository.)

Roxy-3DPrintBoard, 2015-12-24:
Well...     Unless you are changing and saving settings in EEPROM, you don't need the EEPROM to be active.   And the truth is, the EEPROM causes a lot of support problems where people will change this or that in their Configuration.h file but the change never shows up because the value stored in the EEPROM over rides the change.  This happens especially when people are trying to get their Z_PROBE_OFFSET_FROM_EXTRUDER number dialed in.

RetireeJay46, 2015-12-24:
I can see that this discussion is philosophical, not technical, and there's no "right" answer, and I appreciate you explanation.  Now that I know to look out for this, it's now part of my standard operating procedure.  And I've documented it for other users on the Printrbottalk forum.

But IMHO, this is exactly what EEPROM is for.  When you are repeatedly testing and modifying to "dial in" a setting, isn't it easier to just plug the number into the Active Value and save it with M500 than to edit the source code, save it, compile it, and re-flash the whole thing?  You have a good built-in record of what works well, so once you've got everything dialed in you can easily go back and give your source code a single, complete update.

I'm very active on the Printrbottalk forum, with a few thousand members, and I guess the total number of users who OFTEN edit and compile their own version of Marlin is less than a dozen.  Most, including me, either edit & compile the source code rarely or never.  But we definitely use EEPROM all the time.  I've seen a few brand new users who on their first day with their new printer tend to think that they need to re-compile the way you suggest, but we quickly steer them to the EEPROM instead.  The vast majority of our printer users just want to use the printer; programming is not a hobby or a delight for them.

Wackerbarth, 2015-12-24:
@RetireeJay46 -- I think the philosophy is that you should use the EEPROM.
But that you should do so only after you make conscious effort to enable it.

The question becomes one of deciding WHO makes that effort.
In particular, who is going to answer the "customer support" issues.

As things stand right now, in OUR user level documentation there is not an adequate explanation of the use (and misuse) of the EEPROM to avoid support issues which are caused by the unexpected effect of having the EEPROM enabled. So, to avoid those issues, the developers have chosen to default to disabling the feature. If a user chooses to enable the feature, then they should be aware that they have done so and, hopefully, will not forget that in doing so, they are changing the power-up behavior of their printer.

If you would like to assist us in improving our documentation, then, under those changed circumstances, we can revisit the issue.  The documentation is in the wiki so that knowledgable users, such as yourself, can easily contribute improvements.

ntoff, 2015-12-26:
I just put all my calibrated values in auto0.g on an SD card. Saves having to re-flash the firmware or mess about with it at all.

ghuser2, 2016-01-03:
Can this "auto0.g" option be explained more?

KiteLab, 2016-01-03:
https://github.com/MarlinFirmware/Marlin/wiki/Autostart

closed 2016-02-07 by thinkyhead