MarlinFirmware/Marlin #2875: G30 doesnt work

marvin42blue, 2015-12-27:
I'm using the latest 1.1.0-RC3 on a Kossel Mini (Arduino M2560+Ramps1.4). Motion control works so far and also the endstops (according to M119). I have a Z probe connected to the Z-MIN this also seems to work (open/triggered) accordint to M119.

But whenever I issue a G30 command, the steppers do only one single Z-step (approx 0.02 mm) then stop. 

I believe the endstop handling in stepper.cpp somehow aborts the run_z_probe motion and I'm trying to understand the code : can anyone please explain how exactly a change in the endstops aborts a current motion ? It would be nice if there was any explanation in the source code about this.

thanks a lot, matthias

brainscan, 2015-12-28:
Can you post your configuration.h please.

Roxy-3DPrintBoard, 2015-12-28:
So...   This is a some what uninformed opinion...   But I just used G30 on a different matter with RC-3.  G30 works.   My guess is you have a conflict with trying to use Z-Min and Z-Probe together.   It is very important to follow the directions exactly on getting that configured.  As BrainScan suggests, posting your Configuration.h would help.   And if that is what it turns out to be, don't feel bad.   I had a similar problem.    We probably need to get rid of that option or else add a whole bunch of sanity checks for it.

I

marvin42blue, 2015-12-28:
...blame it on me. I disabled Z_PROBE_ALLEN_KEY since I'm using a manual Z-probe (which is not an allen key, but rather a feeler gauge) and I don't wanted the auto-deploy/retract to happen - so in this case G30 wouldn't do too much

Attached is the configuration.h - I think a sanity check should report when bed leveling is enabled, but no probe mechanism is active
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/73094/Configuration.h.txt)

On the other hand, it looks like I cannot have a simple manual Z-probe (whatever that might be) without modifying the code.
I'd like to suggest a different approach:
- configure whether Z-probing is enabled or not
- configure which method shall be used for bed-levelling
- configure which trigger input shall be used (Z-min, analog for FSR, or whatever)
- configure auto-deploy/stow of Z-probe

marvin42blue, 2015-12-28:
Hello,

sorry for bothering you directly - let me know if you think that's inappropriate - but I'd like to suggest an idea:

I'm new to 3D-printing and just created a delta machine based on Kossel mini design. My profession are industrial robots for over 20 years.

I'm used to robots and kinematic transformations, so for me the "effector" of the Kossel is just the flange of a delta kinematic. Since I want to use the delta design not only for 3D-printing but also for milling and together with my laser (0.2W) head, I'm considering the print head to be a "tool" and the tip of the print head is what I call a TCP (tool center point). For a robot, the TCP is just = flange + tool (where the '+' operator is basically a 4x4 matrix product). With this idea in mind, any Z-probe (allen.key, sled, whatever) is also just another tool, with a different TCP, thats all.

So I'd like to suggest to have a more modular approach in order to use Marlin firmware also for these applications.
This approach would also work when having multiple print heads (for eg different colors) etc

In case you think I should talk to anyone else or/and post this into a forum, please let me know,

best regards
Matthias Dehof

  ----- Original Message ----- 
  From: Roxy-3DPrintBoard 
  To: MarlinFirmware/Marlin 
  Cc: marvin42blue 
  Sent: Monday, December 28, 2015 4:31 AM
  Subject: Re: [Marlin] G30 doesnt work (#2875)

  So... This is a some what uninformed opinion... But I just used G30 on a different matter with RC-3. G30 works. My guess is you have a conflict with trying to use Z-Min and Z-Probe together. It is very important to follow the directions exactly on getting that configured. As BrainScan suggests, posting your Configuration.h would help. And if that is what it turns out to be, don't feel bad. I had a similar problem. We probably need to get rid of that option or else add a whole bunch of sanity checks for it.

  I

Roxy-3DPrintBoard, 2015-12-28:
> ...blame it on me. I disabled Z_PROBE_ALLEN_KEY since I'm using a manual Z-probe (which is not an allen key, but rather a feeler gauge) and I don't wanted the auto-deploy/retract to happen - so in this case G30 wouldn't do too much

This is in the Development branch right now.   You can have a Z_PROBE_ALLEN_KEY and there is an extra qualifier that says it is MANUALLY_DEPLOYED.   Probably we will end up making the MANUALLY_DEPLOYED part of it apply to any Z-Probe.   The Development Branch is almost identical to the RC-3 branch so you can switch without much concern.

> Attached is the configuration.h - I think a sanity check should report when bed leveling is enabled, but no probe mechanism is active

There are sanity checks that complain if you have Auto Bed Leveling enabled but no Z-Probe.

AnHardt, 2016-01-18:
If you don't like the moves to deploy/stove the Allen Keys - configure them away.

```
    // don't do extra moves
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_X current_position[X_AXIS]
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Y current_position[Y_AXIS]
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_Z Z_RAISE_BEFORE_PROBING
    #define Z_PROBE_ALLEN_KEY_DEPLOY_1_FEEDRATE XY_TRAVEL_SPEED
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_X current_position[X_AXIS]
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_Y current_position[Y_AXIS]
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_Z Z_RAISE_BEFORE_PROBING
    #define Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE XY_TRAVEL_SPEED

    #define Z_PROBE_ALLEN_KEY_STOW_DEPTH 0
    // Move the probe into position
    #define Z_PROBE_ALLEN_KEY_STOW_1_X current_position[X_AXIS]
    #define Z_PROBE_ALLEN_KEY_STOW_1_Y current_position[Y_AXIS]
    #define Z_PROBE_ALLEN_KEY_STOW_1_Z Z_RAISE_AFTER_PROBING
    #define Z_PROBE_ALLEN_KEY_STOW_1_FEEDRATE XY_TRAVEL_SPEED
    // Move the nozzle down further to push the probe into retracted position.
    #define Z_PROBE_ALLEN_KEY_STOW_2_X  current_position[X_AXIS]
    #define Z_PROBE_ALLEN_KEY_STOW_2_Y  current_position[Y_AXIS]
    #define Z_PROBE_ALLEN_KEY_STOW_2_Z  Z_RAISE_AFTER_PROBING
    #define Z_PROBE_ALLEN_KEY_STOW_2_FEEDRATE XY_TRAVEL_SPEED
    // Raise things back up slightly so we don't bump into anything
    #define Z_PROBE_ALLEN_KEY_STOW_3_X  current_position[X_AXIS]
    #define Z_PROBE_ALLEN_KEY_STOW_3_Y  current_position[Y_AXIS]
    #define Z_PROBE_ALLEN_KEY_STOW_3_Z  Z_RAISE_AFTER_PROBING
    #define Z_PROBE_ALLEN_KEY_STOW_3_FEEDRATE XY_TRAVEL_SPEED
```
UNTESTED!

AnHardt, 2016-01-18:
By the way:
```
          // Move to engage deployment
          if (Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE != Z_PROBE_ALLEN_KEY_DEPLOY_1_FEEDRATE)
            feedrate = Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE;
          if (Z_PROBE_ALLEN_KEY_DEPLOY_2_X != Z_PROBE_ALLEN_KEY_DEPLOY_1_X)
            destination[X_AXIS] = Z_PROBE_ALLEN_KEY_DEPLOY_2_X;
          if (Z_PROBE_ALLEN_KEY_DEPLOY_2_Y != Z_PROBE_ALLEN_KEY_DEPLOY_1_Y)
            destination[Y_AXIS] = Z_PROBE_ALLEN_KEY_DEPLOY_2_Y;
          if (Z_PROBE_ALLEN_KEY_DEPLOY_2_Z != Z_PROBE_ALLEN_KEY_DEPLOY_1_Z)
            destination[Z_AXIS] = Z_PROBE_ALLEN_KEY_DEPLOY_2_Z;
          prepare_move_raw();

          #ifdef Z_PROBE_ALLEN_KEY_DEPLOY_3_X
            if (Z_PROBE_ALLEN_KEY_DEPLOY_3_FEEDRATE != Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE)
              feedrate = Z_PROBE_ALLEN_KEY_DEPLOY_3_FEEDRATE;

          // Move to trigger deployment
          if (Z_PROBE_ALLEN_KEY_DEPLOY_3_FEEDRATE != Z_PROBE_ALLEN_KEY_DEPLOY_2_FEEDRATE)
            feedrate = Z_PROBE_ALLEN_KEY_DEPLOY_3_FEEDRATE;
          if (Z_PROBE_ALLEN_KEY_DEPLOY_3_X != Z_PROBE_ALLEN_KEY_DEPLOY_2_X)
            destination[X_AXIS] = Z_PROBE_ALLEN_KEY_DEPLOY_3_X;
          if (Z_PROBE_ALLEN_KEY_DEPLOY_3_Y != Z_PROBE_ALLEN_KEY_DEPLOY_2_Y)
            destination[Y_AXIS] = Z_PROBE_ALLEN_KEY_DEPLOY_3_Y;
          if (Z_PROBE_ALLEN_KEY_DEPLOY_3_Z != Z_PROBE_ALLEN_KEY_DEPLOY_2_Z)
            destination[Z_AXIS] = Z_PROBE_ALLEN_KEY_DEPLOY_3_Z;

          prepare_move_raw();
```
```      // Move the nozzle down to push the Z probe into retracted position
      if (Z_PROBE_ALLEN_KEY_STOW_2_FEEDRATE != Z_PROBE_ALLEN_KEY_STOW_1_FEEDRATE)
        feedrate = Z_PROBE_ALLEN_KEY_STOW_2_FEEDRATE;
      if (Z_PROBE_ALLEN_KEY_STOW_2_X != Z_PROBE_ALLEN_KEY_STOW_1_X)
        destination[X_AXIS] = Z_PROBE_ALLEN_KEY_STOW_2_X;
      if (Z_PROBE_ALLEN_KEY_STOW_2_Y != Z_PROBE_ALLEN_KEY_STOW_1_Y)
        destination[Y_AXIS] = Z_PROBE_ALLEN_KEY_STOW_2_Y;
      destination[Z_AXIS] = Z_PROBE_ALLEN_KEY_STOW_2_Z;
      prepare_move_raw();

      // Move up for safety
      if (Z_PROBE_ALLEN_KEY_STOW_3_FEEDRATE != Z_PROBE_ALLEN_KEY_STOW_2_FEEDRATE)
        feedrate = Z_PROBE_ALLEN_KEY_STOW_2_FEEDRATE;
      if (Z_PROBE_ALLEN_KEY_STOW_3_X != Z_PROBE_ALLEN_KEY_STOW_2_X)
        destination[X_AXIS] = Z_PROBE_ALLEN_KEY_STOW_3_X;
      if (Z_PROBE_ALLEN_KEY_STOW_3_Y != Z_PROBE_ALLEN_KEY_STOW_2_Y)
        destination[Y_AXIS] = Z_PROBE_ALLEN_KEY_STOW_3_Y;
      destination[Z_AXIS] = Z_PROBE_ALLEN_KEY_STOW_3_Z;
      prepare_move_raw();
```
These compares only take valuable progmem and take more time than the simple assignment. The zero length moves will be sorted out by prepare_move_raw().
More sense could make to test if the coordinates are #defined at all.

Wackerbarth, 2016-01-18:
"These compares only take valuable progmem and take more time than the simple assignment." -- Actually, they don't. You forget that the compilers now compress constant expressions and eliminate unreachable code.

With in more modern compilers, it is possible to avoid using some of the preprocessor conditionals that used to be required to eliminate the extra code and, as a result, write conditional blocks in normal "C" expressions which are just as efficient, but also allow more flexibility in the formatting of the code.

However, you do raise a very necessary issue. Is it really worthwhile to be excessively worried about some resources?  In particular, the computational efficiency of these movements is "garbage". They are seldom executed, and when they are, the mechanical movement is the limiting factor. A few extra microseconds (here) will have NO impact on the performance of the printer.

It is also unclear to me just how much value we should place on saving a few bytes of progmem. Although we cannot be as cavalier as we would with a larger (eg ARM) processor, are we really saving enough PROM to offset the human cost associated with the effort to create the potential savings?

I know that in the case of the printers that I have, using a few hundred extra bytes of PROM would still leave plenty to spare. Perhaps we are being "Penny wise, but Pound foolish".

Roxy-3DPrintBoard, 2016-01-18:
> It is also unclear to me just how much value we should place on saving a few bytes of progmem. Although we cannot be as cavalier as we would with a larger (eg ARM) processor, are we really saving enough PROM to offset the human cost associated with the effort to create the potential savings?

I'm of the opinion we don't want to be blatantly wasteful of memory (or processing power).   But right now, we have enough extra to give us head room until the 3D-Printer community switches to ARM processors.    (I'm just guessing the bulk of the community will move to ARM because it has support from the same tools and company.)

Wackerbarth, 2016-01-18:
I wasn't suggesting that we be "wasteful" when there are clear efficiencies to be had. I will concede that we are pushing the ISR processing requirements to the limit. Anything that helps in that area has a significantly better cost/benefit ratio. But, with a few exceptions, over the years it has been learned that effort spent on improving algorithms is more cost effective than attempting to squeeze the last little bit out of some piece of code. As an example, rewriting two paths so that they can actually share some common code might be more beneficial than attempting to squeeze out a few bytes by rearranging code order.

As I allude, there will (probably soon) become a point where it no longer makes sense to attempt to add additional features to memory-limited printers. Let the older designs do what they can and add the new features (which require additional resources) only to the newer, larger, faster processors.

Roxy-3DPrintBoard, 2016-01-18:
I know you were not suggesting we be wasteful.

My thinking is right now we have plenty of ROM space and 3+ KB of RAM.   At some point we are going to have to freeze the features for the older AVR boards.   I'm just guessing we still have a year before that happens.   With a little bit of luck, the 32 bit ARMS will be everywhere by then.

AnHardt, 2016-01-18:
Until now there are several 64k/4k boards out there. Very careful feature selection is required with them.

Roxy-3DPrintBoard, 2016-01-18:
@AnHardt   What is a 64k/4k board?   Can you explain that to me?    With my LCD Panel it burns up a bunch of the program memory but I still have lots of space left.

AnHardt, 2016-01-18:
Boards based on processors with 64k progmem / 4k RAM.
See the Makefile. List is from line 74 on. Every board with atmega644, atmega644p, at90usb646.

Wackerbarth, 2016-01-18:
And, what combination of options make a real difference?  Some are sufficiently small. Others are, clearly, too large to fit. But what configurations are "marginal"?

AnHardt, 2016-01-19:
@Wackerbarth 
I suppose

SDSUPPORT
DOGLCD
ULTIPANEL
AUTO_BED_LEVELING_FEATURE
MANUAL_BED_LEVELING

are the most memory consuming - in that order. Maybe you want to test yourself. Just activate a feature and compile.

Oh, ähh - DELTA needs a bit, but i don't know how to order.

closed 2016-04-23 by jbrazio