MarlinFirmware/Marlin #2877: make Z safe homing independent of auto bed leveling

MacTheApple, 2015-12-28:
I have a modified bed on my printer that is not in line with the X and Y endstops, so I need to be able to tell it to move to a specific spot to home Z. Right now, in order to do that, I would have to enable auto-bed leveling, which is incompatible with my printbed. I would rather use mesh bed leveling, but there appears to be no way to manually set the XY position for Z homing in that mode. Is there a specific reason that the firmware will not compile if z safe homing is enabled outside the auto bed leveling section, and could it be made to operate independently of any specific bed leveling feature?

Roxy-3DPrintBoard, 2015-12-28:
I think this might be OK.   I haven't looked at the code real carefully to verify that.   But the history of the Safe Homing was because the probe offset from the nozzle was causing problems.   If you homed the Z-Axis in a location where the Z-Probe would miss the bed, really bad things happened.

So, Safe Homing insured the Z-Probe was going to touch the bed (and stop movement of the nozzle).   Like I say, I haven't looked at the code real carefully to verify this, but this does seem like a reasonable request.

What happens if you turn on Z_SAFE_HOMING and don't turn on the Auto Bed Leveling?  I don't know that it will work, but it has a fair chance of working.   You would do this by going into Configuration.h and moving these lines:

~~~~
  #define Z_SAFE_HOMING   // This feature is meant to avoid Z homing with Z probe outside the bed area.
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
   #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
~~~~

in front of:

~~~~
#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
#define DEBUG_LEVELING_FEATURE
#define Z_MIN_PROBE_REPEATABILITY_TEST  // If not commented out, Z-Probe Repeatability test will be included if Auto Bed Leveling is Enabled.

#if ENABLED(AUTO_BED_LEVELING_FEATURE)
~~~~

At that point, you will have the #define's you need for Safe Homing defined even though the Auto Bed leveling is not turned on.    Please give that a try and let us know what happened.

MacTheApple, 2015-12-29:
ok, that didn't work, but it gave me a hint to something that at least compiles.

in front of 
```
//#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
```
I placed the following:
```
// Make z safe homing work with mesh bed leveling
#define Z_SAFE_HOMING   

  #if ENABLED(Z_SAFE_HOMING) && ENABLED(MESH_BED_LEVELING)
// Offsets to the Z probe relative to the nozzle tip.
  // X and Y offsets must be integers.
  #define X_PROBE_OFFSET_FROM_EXTRUDER -25     // Z probe to nozzle X offset: -left  +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -29     // Z probe to nozzle Y offset: -front +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -12.35  // Z probe to nozzle Z offset: -below (always!)

  #define Z_RAISE_BEFORE_HOMING 4       // (in mm) Raise Z axis before homing (G28) for Z probe clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case.

  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min.
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).

  #endif
```

Now all I have to do is test it out. I made the if statement unique so that it would not fire twice if auto bed leveling was on instead of mesh bed leveling.

MacTheApple, 2015-12-31:
Does not work, still attempts to home Z at X=0, and Y=0.
i tried looking at marlin.main, as that's where it was throwing errors before, but i dont really see the issue, although it does say this:

```
// NOTE: If current_position[X_AXIS] or current_position[Y_AXIS] were set above
// then this may not work as expected.
```

Perhaps that's part of the issue?

MacTheApple, 2016-01-01:
Ok, after looking at it a bit more, it might be better if there was just had a section for the type of z axis sensor used, with all the options for that sensor being in that section, If I'm using a prox sensor and I don't want to use auto bed leveling, I cant access the option to use an alternate z min probe pin either.

It would be mush more usable if the statements went:
```
//===========================================================================
//==============================Z Endstop Options ===========================
//===========================================================================
// Only one of these can be on at a time.
//#define Z_probe_Prox //enable if you have a prox sensor on Z
//#define Z_probe_Servo //enable if you have a servo sensor on Z
#define Z_probe_mech_endstop //leave enabled unless you have a special type of endstop

#If Enabled(Z_probe_Prox) 
< all prox settings go here (including alt z min probe, Z_safe homing, etc)>
 #endif
#If Enabled(Z_probe_Servo) 
< all servo settings go here(sled options, other servo stuff)>
 #endif
#If Enabled(Z_probe_mech_endstop) 
< all mech_endstop settings go here>
 #endif

//===========================================================================
//=========================== Manual Bed Leveling ===========================
//===========================================================================

//Can use any sensor type
<All manual/mesh bed leveling stuff goes here, and any variables needed for any of the probes are set here>

//===========================================================================
//============================ Bed Auto Leveling ============================
//===========================================================================

//Must use either servo or proximity sensor
<All auto bed leveling stuff goes here, and any variables needed for servos and prox sensors are set here>
```

Wackerbarth, 2016-01-01:
I agree with your general organization.

However, I have never seen why all of these defines are conditional at all.
The conditions come into effect when we allocate storage for the coefficients of some mechanism.
Until then, they are just defining preprocessor names.

So, I would eliminate all of the #if/#endif from the Configuration section

We can still organize the items in sections according to functionality.
But things would be less cluttered and not buried in obfuscating nested conditionsl blocks.

ESS-1, 2016-01-10:
The following configuration works on 1.1.0 RC3:

    // X and Y offsets must be integers
    #define X_PROBE_OFFSET_FROM_EXTRUDER -16
    #define Y_PROBE_OFFSET_FROM_EXTRUDER 19
    #define Z_RAISE_BEFORE_HOMING 0.5 // (in mm) Raise Z before homing (G28) for Probe Clearance.
                                  // Be sure you have this distance over your Z_MAX_POS in case
    #define XY_TRAVEL_SPEED 50*60     // X and Y axis travel speed between probes, in mm/min
    #define Z_SAFE_HOMING // This feature is meant to avoid Z homing with probe outside the bed area.
    #define Z_SAFE_HOMING_X_POINT (X_MAX_LENGTH/3)    // X point for Z homing when homing all axis (G28)
    #define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH/3)    // Y point for Z homing when homing all axis (G28)

Safe Z-homing area is defined by X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER (homing is allowed if the probe is within the printing area).
Z_SAFE_HOMING_X_POINT and Z_SAFE_HOMING_Y_POINT are used only when you home all axes with G28 command.

KiteLab, 2016-01-12:
@ESS-1 
The probable area:
```
  #if ENABLED(AUTO_BED_LEVELING_GRID)

    #define LEFT_PROBE_BED_POSITION 15
    #define RIGHT_PROBE_BED_POSITION 170
    #define FRONT_PROBE_BED_POSITION 20
    #define BACK_PROBE_BED_POSITION 170
```
is neither defined nor tested this way, but is required for homing z only.

ESS-1, 2016-01-12:
*_PROBE_BED_POSITION are used for auto-leveling only.
The only additional definitions required for safe homing in 1.1.0 RC3 are X_PROBE_OFFSET_FROM_EXTRUDER, Y_PROBE_OFFSET_FROM_EXTRUDER, Z_RAISE_BEFORE_HOMING, XY_TRAVEL_SPEED.

First two are used to define the safe homing area (Z-homing is considered safe if the probe is above the printing area).

Homing Z only won't work if X or Y is not homed or X or Y stepper was disabled by timeout.

Roxy-3DPrintBoard, 2016-01-12:
> Homing Z only won't work if X or Y is not homed or X or Y stepper was disabled by timeout.

Especially if the Delta flag is turned on.

thinkyhead, 2016-03-11:
This feature request should be kept alive, but re-opened in the MarlinDev issue queue. `Z_SAFE_HOMING` as a general option is something that we have discussed a lot and everyone agrees it can be helpful. We're just a little shy to mess with any homing or leveling code because it deals with a lot of different configurations and it can be kind of fragile. So we should work on this in the dev branch for 1.2 and perhaps it can also get into 1.1.1.

closed 2016-03-11 by thinkyhead

SirZylinder, 2017-05-31:
#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif

Where do I enter my values here? Sry im a scrub and cry every night but I really need to know :P

Roxy-3D, 2017-05-31:
If Z_SAFE_HOMING is turned on...  The safe X & Y point automatically get set to the middle of the bed.   If you want to hard code a location, you can do it by doing something like:
```cpp
#define Z_SAFE_HOMING_X_POINT (58) // X point for Z homing when homing all axis (G28).
#define Z_SAFE_HOMING_Y_POINT (137) // Y point for Z homing when homing all axis (G28).
```

SirZylinder, 2017-05-31:
Found it out myself already works perfect.
Trial and error is the tactic to go.
Thanks tho for the quick answer.

Blisk, 2018-08-23:
Z switch should be for safety, because inudctive sensors are not reliable, I have changed 5 inductive sensors because they crashed nozzle into my heatbed. I hope they will enable this in marlin.

thinkyhead, 2018-08-25:
@Blisk — One thing we did add is an option to stop the downward probe movement as soon as it goes past the expected trigger point, so it won't try to go 100mm below the bed… The option is called `Z_PROBE_LOW_POINT`.