MarlinFirmware/Marlin #2878: Initial overshoot on extruder temperature

frank16, 2015-12-29:
#define PID_FUNCTIONAL_RANGE 10

I think this is too small and causes overshoot. Even if I turn all my pid values to 0, I get at a fast initial slope and overshoot. I dont't see the point of it? Shouldn't pid do the work from the beginning?
Is it meant as a backup-feature (it does do some control even if the pid is way of...)

thinkyhead, 2016-03-03:
Have you done calibration with `M303`? It shouldn't overshoot so much with a properly calibrated heater.

Blue-Marlin, 2016-03-03:
@thinkyhead 
M303 tells you how to tune the PID parameters.
It the temperature is below (target - PID_FUNCTIONAL_RANGE) the heater is heating with 100% power - PID is not in effect so can not help here.

@frank16 
Hotends are different. That's the reason for PID_FUNCTIONAL_RANGE being configurable. 10 seems to work well with a lot of hotends.

The other question here is, if we can drop PID_FUNCTIONAL_RANGE at all.
PID_FUNCTIONAL_RANGE was used as a workaround to limit the I-therm in the PID calculation. It should prevent I from growing to infinity when the target temperature is far away. Since the i-therm is now limited internally, in the PID algorithm, this could be dropped.

A simple test is, to set PID_FUNCTIONAL_RANGE to a value in the magnitude of the target temperature (or larger). If then no critical overshot occurs. PID_FUNCTIONAL_RANGE can be dropped. 
For me that works.
Any other testers?

frank16, 2016-03-03:
That is the one that makes sense!
The I term going to high values during build-up of temperature and causing 
overshoot!

I've noticed that the hardware (temp sensor) is very much dampened, 
probably to reduce noise from the wiring, especially with the thermistor 
(100kohm is asking for trouble). Unfortunately that causes a delay in the 
system making it very succeptible to oscillations from the I value. I tune 
my hotend manually now, first P to an overshoot of 5deg max, then add a 
little I until it locks in without undershoot. Add a bit of D action if a 
disturbance (100% object fan) makes the hotend lose more than 2degC.  Too 
much D action makes the output to the heater very jumpy, so make sure to 
watch the temperature graph and keep it smooth. If you do it this way, 
because of low I value, PID_FUNCTIONAL_RANGE is not so critical anymore.
I switched to repetier firmware recently, same issues. That one has an 
option to add in a base PWM value for the output, also speeding up response 
a bit.
I have experimented with different combinations of voltage and heater, even 
24V in 12V element, the method I described above works quite well.
The only issue I have is that on subsequent runs of the printer initial 
overshoot gets higher, maybe the PID is not reset between runs? Maybe it's 
worthwhile checking how Marlin handles this.

jbrazio, 2016-03-03:
@Blue-Marlin I have set `#define PID_FUNCTIONAL_RANGE 300` and here is the result:
[image]

closed 2016-04-27 by jbrazio