MarlinFirmware/Marlin #2880: Full Graphic smart controller jog/push control inactive

picclock, 2016-01-01:
Hi
Have upgraded to latest rc3 and now the control knob on the graphics controller does not function. Graphics looks great, btw. Have checked connections to the Arduino boards with a meter, and replaced the arduino board in case of a fault. Neither the push button function or rotate seem to do anything.

I have uncommented the following lines in configuration.h -
#define SDSUPPORT // Enable SD Card Support in Hardware Console
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define ENCODER_PULSES_PER_STEP 1 // Increase if you have a high resolution encoder
#define ENCODER_STEPS_PER_MENU_ITEM 5 // Set according to ENCODER_PULSES_PER_STEP 
#define SPEAKER // The sound device is a speaker - not a buzzer. A buzzer resonates 
#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100 // the duration the buzzer plays the
#define LCD_FEEDBACK_FREQUENCY_HZ 1000         // this is the tone frequency the buzzer plays 

I could not find any other lines in config re the display control so any help much appreciated.

Windoze 7 64bit, Arduino IDE 1.6.6

Many thanks

Best Regards

picclock

picclock, 2016-01-01:
Have done additional testing. Measured voltage on push switch which is connected to gnd when pushed. Likewise on both direction inputs which are momentarily pulled to ground when switch is rotated. Also tried fitting reprap discount smart controller (non graphic), complete with ribbon cables and adapter card. I altered configuration.h to reflect this and recompiled. The encoder behaviour remains unchanged and I am unable to select any of the other menus.

Best Regards

picclock

picclock, 2016-01-01:
Sorted - but deceptive. Caused by disconnection pf thermistor. Problem is that on older version board does not stop running. May I suggest a minor software change. When mintemp for thermistors is exceeded display a message :- 
System shutdown due low temp detected.
This would inform the user that the system had stopped and that global warming had been reversed :-) or a thermistor wire was disconnected.

Best Regards

picclock

AnHardt, 2016-01-01:
Maybe something like:
[image]
```
17:15:12.654 : Error:MINTEMP triggered, system stopped! Heater_ID: 0
17:15:12.655 : Error:Printer halted. kill() called!
```
:-)

picclock, 2016-01-01:
Hi anhardt
Because my cards were enclosed seeing any flashing lights is a nono. So ATM I was greeted with an apparently running but unresponsive system. Your comments seem perfect to correct that misunderstanding. It was only when I changed the display I realised belatedly what was going on. Thankyou for responding. Best regards, picclock

closed 2016-03-11 by thinkyhead