MarlinFirmware/Marlin #2882: Small typo in configuration.h: Line 187&188: Comment should be "//  P=U^2/R"

docpayce, 2016-01-02:
In lines 187 and 188 there is a small typo: The comment should not be "//  P=I^2/R" (current) but either "//  P=U^2/R" or "//  P=V^2/R" (voltage), depending on your preferred flavor.

thinkyhead, 2016-03-11:
Patched!

closed 2016-03-11 by thinkyhead

CONSULitAS, 2016-03-15:
@thinkyhead 
Hi Scott,

perhaps you could state in your commits, if it fixes an issue:
https://help.github.com/articles/closing-issues-via-commit-messages/

This would have the advantage, that the commit is linked to the issue and makes the housekeeping easier, because the issue will be closed when the commit is merged.

What do you think?

Regards Jochen

BTW: Great work by you in the last weeks! :+1:

thinkyhead, 2016-03-16:
@CONSULitAS

Thanks Jochen! I wasn't aware of that Github feature. I will definitely use it when it's obvious.

After all this busy code-sweeping I hope all these weeks of work will give us a nice stable release!