MarlinFirmware/Marlin #2883: Software z endstop not operable with Delta

docpayce, 2016-01-02:
Using Marlin 1.1.0 RC3 my Delta is allowed to go below z = 0 (to negative z) despite
```
#define min_software_endstops true
...
#define Z_MIN_POS 0
```
in configuration.h being set. I can move down to z=-0.8 mm, which is exactly my 
```
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.80
```
with using an inductive distance sensor on a Kossel setup. Maybe the Z_PROBE_OFFSET_FROM_EXTRUDER messes up the software endstops? All other endstops work like a charm.

btw: Setting 
```
#define Z_MIN_POS -Z_PROBE_OFFSET_FROM_EXTRUDER 
```

is a functional workaround and will limit min movement to z = 0. But its kinda weird to set that. :)

thinkyhead, 2016-03-11:
Is this still the case with `RCBugFix` today? There have been a few changes to code in these areas since RC3.

Blue-Marlin, 2016-03-11:
@thinkyhead 
Software endstops for DELTA machines are completely broken.
Some thoughts in https://github.com/MarlinFirmware/MarlinDev/issues/304.

thinkyhead, 2016-03-23:
One question at MarlinFirmware/MarlinDev#304 is why the zmax isn't clipped in the same way as zmin. I don't see that it needs any special handling. Maybe you can think of a case…?

The other thing that seems off to me is the block in `clamp_to_software_endstops` that begins `if (home_offset[Z_AXIS] < 0) {`. To me that appears redundant. The Z position of the bed (on a delta too) should be `Z_HOME_POS + home_offset[Z_AXIS]`. The `min_pos[Z_AXIS]` value also already includes `home_offset[Z_AXIS]`. (Basically the whole coordinate system is offset by `home_offset[]`.)

Here's the function source for reference…
```cpp
void clamp_to_software_endstops(float target[3]) {
  if (min_software_endstops) {
    NOLESS(target[X_AXIS], min_pos[X_AXIS]);
    NOLESS(target[Y_AXIS], min_pos[Y_AXIS]);

    float negative_z_offset = 0;
    #if ENABLED(AUTO_BED_LEVELING_FEATURE)
      if (zprobe_zoffset < 0) negative_z_offset += zprobe_zoffset;
      if (home_offset[Z_AXIS] < 0) {
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (marlin_debug_flags & DEBUG_LEVELING) {
            SERIAL_ECHOPAIR("> clamp_to_software_endstops > Add home_offset[Z_AXIS]:", home_offset[Z_AXIS]);
            SERIAL_EOL;
          }
        #endif
        negative_z_offset += home_offset[Z_AXIS];
      }
    #endif
    NOLESS(target[Z_AXIS], min_pos[Z_AXIS] + negative_z_offset);
  }

  if (max_software_endstops) {
    NOMORE(target[X_AXIS], max_pos[X_AXIS]);
    NOMORE(target[Y_AXIS], max_pos[Y_AXIS]);
    NOMORE(target[Z_AXIS], max_pos[Z_AXIS]);
  }
}
```

closed 2016-04-28 by jbrazio