MarlinFirmware/Marlin #2886: Z_RAISE_BEFORE_PROBING is executed once for each axis in G28...

lrpirlet, 2016-01-02:
**Issue**

While working the code I found that the global variable destination[Z_AXIS] was set in G28 to execute line_to_destination(), but NEVER reset before a next line_to_destination(), 

I guess this is a bug...

This happens in the following code
```C++

      #elif DISABLED(Z_SAFE_HOMING) && defined(Z_RAISE_BEFORE_HOMING) && Z_RAISE_BEFORE_HOMING > 0

        // Raise Z before homing any other axes
        // (Does this need to be "negative home direction?" Why not just use Z_RAISE_BEFORE_HOMING?)
        destination[Z_AXIS] = -Z_RAISE_BEFORE_HOMING * home_dir(Z_AXIS);
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (marlin_debug_flags & DEBUG_LEVELING) {
            SERIAL_ECHOPAIR("Raise Z (before homing) by ", (float)Z_RAISE_BEFORE_HOMING);
            SERIAL_EOL;
            print_xyz("> (home_all_axis || homeZ) > destination", destination);
          }
        #endif
        feedrate = max_feedrate[Z_AXIS] * 60;
        line_to_destination();
		// LRP restore destination[Z_AXIS] for next line_to_destination() !!! global variable !!!
		// LRP NEED TO DO: make sure to have each axis destination initialized before any line_to_destination()
		destination[Z_AXIS] = current_position[Z_AXIS];
		// LRP 
        st_synchronize();
```
The solution is in the comment //LRP (3lines)

**Proposed solution**

The diff is

```diff
$ git diff
diff --git a/Marlin-RC/Marlin/Marlin_main.cpp b/Marlin-RC/Marlin/Marlin_main.cpp
index 1de6a9f..7652706 100644
--- a/Marlin-RC/Marlin/Marlin_main.cpp
+++ b/Marlin-RC/Marlin/Marlin_main.cpp
@@ -2323,6 +2323,10 @@ inline void gcode_G28() {
         #endif
         feedrate = max_feedrate[Z_AXIS] * 60;
         line_to_destination();
+               // LRP restore destination[Z_AXIS] for next line_to_destination() !!! global variable !!!
+               // LRP NEED TO DO: make sure to have each axis destination initialized before any line_to_destination()
+               destination[Z_AXIS] = current_position[Z_AXIS];
+               // LRP
         st_synchronize();

       #endif
```

AnHardt, 2016-01-14:
Under what circumstances (g-code sequence) can you observe this behavior? I can't reproduce this problem.
destination[Z_AXIS] is an absolute value.
```
destination[Z_AXIS] = -Z_RAISE_BEFORE_HOMING * home_dir(Z_AXIS);
```
When the move was performed the z-axis is at destination[Z_AXIS].
If called again, with the same destination, this will not result in a move.

lrpirlet, 2016-01-15:
Hmmm ok @AnHardt, I am using Repetier host to talk with the printer... I
need to verify what Gcode is sent...
Basically I turn on the printer, connect to it from Repetier and
issue debug on, (from memory M111 S100000) then G28... (no arguments).
I know that Repetier sends some gcode, I am NOT sure that I did customize
some start-up Gcode. I will verify what code is sent...

I am away from my device till Sunday, I will elaborate as soon as I am home
and as soon as I have time to do so.

For your info, I designed a thing attached to a servo. Made of a surface
mounted switch mounted on a servo. I have zero offset for X and for Y, a
constant offset for the Z switch that triggers when the hot-end tip
activates it by pressing it to the bed... The switch is wired in parallel
to the static z switch... I will use this setup to "motorize" the mesh
levelling... So, in my configuration.h, I only have the servo enabled and
NOT the ABL... I have not published this work as I can work on it very
intermittently...

This lead me to discover at least 2 "funny behaviours" and for each of
those I submitted "my" correction. The big problem is with the lack of
"specifications"... for example, should the servo move back to up position
when it did NOT move to the bottom position? Should the G28 end-up with the
hot-end tip "on the bed" at position x = y = z = zero to mimic the
behaviour observed with 3 "statics" switches? etc...
​

lrpirlet, 2016-01-15:
@AnHardt I got home earlier and got time today to answer...

First of all, please notice #2899  entry from @Kriechi who report same problem...
BTW @Kriechi  thanks for your input... And NO I have no idea how to format (not that I care, ever read obfuscatng C? this is valid C even when nobody can grasp easely...)

Well, I am not sure about what you are looking for, but here is the initialization after power-up with firmware before modification, and with debuging information in line_to_destination() displaying destination...

It gives all the Gcode from Repetier prior my M111 S100000 and my G28..

The  Raize Z before homing, value 4 is applied and shows up in ALL AXIS DESTINATION, note that move values are RELATIVE since intended no move value is ZERO...Also, current position displays as 0,0,0 even after a move... at least I understand it that way...

I highlighted the destination

Please note further down a second debug session where I included my correction...

```
22:30:03.709 : Printer reset detected - initalizing
22:30:03.709 : start
22:30:03.709 : echo: External Reset
22:30:03.709 : Marlin 1.1.0-RC3
22:30:03.725 : echo: Last Updated: 2015-12-01 12:00 | Author: LRP auto mesh 29-Dec-2015
22:30:03.725 : Compiled: Jan 15 2016
22:30:03.725 : echo: Free Memory: 2532  PlannerBufferBytes: 1232
22:30:03.725 : echo:V21 stored settings retrieved (428 bytes)
22:30:03.725 : echo:Steps per unit:
22:30:03.725 : echo:  M92 X80.00 Y80.00 Z4000.00 E402.00
22:30:03.725 : echo:Maximum feedrates (mm/s):
22:30:03.725 : echo:  M203 X200.00 Y200.00 Z1.00 E45.00
22:30:03.741 : echo:Maximum Acceleration (mm/s2):
22:30:03.741 : echo:  M201 X1000 Y1000 Z50 E5000
22:30:03.741 : echo:Accelerations: P=printing, R=retract and T=travel
22:30:03.741 : echo:  M204 P1500.00 R2000.00 T1500.00
22:30:03.741 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
22:30:03.741 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
22:30:03.756 : echo:Home offset (mm):
22:30:03.756 : echo:  M206 X0.00 Y0.00 Z0.00
22:30:03.756 : echo:Material heatup parameters:
22:30:03.756 : echo:  M145 M0 H160 B70 F0
22:30:03.756 : echo:  M145 M1 H220 B110 F0
22:30:03.756 : echo:PID settings:
22:30:03.756 : echo:  M301 P22.41 I1.56 D80.41 C100.00 L20
22:30:03.756 : echo:Filament settings: Disabled
22:30:03.756 : echo:  M200 D3.00
22:30:03.756 : echo:  M200 D0
22:30:03.756 : echo:Z-Probe Offset (mm):
22:30:03.756 : echo:  M851 Z0.00
22:30:03.866 : N1 M110 *2
22:30:07.663 : echo:SD init fail
22:30:07.663 : ok
22:30:07.663 : N2 M115 *4
22:30:07.772 : FIRMWARE_NAME:Marlin 1.1.0-RC3 From Archive SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:Pi3 3D LabteK EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
22:30:07.772 : ok
22:30:07.788 : N4 M114 *3
22:30:07.788 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0.00 Y:0.00 Z:0.00
22:30:07.788 : ok
22:30:07.788 : N5 M111 S6 *66
22:30:07.803 : ok
22:30:07.803 : N6 T0 *28
22:30:07.803 : echo:Active Extruder: 0
22:30:07.803 : ok
22:30:07.803 : N7 M20 *54
22:30:07.803 : Begin file list
22:30:07.803 : End file list
22:30:07.803 : ok
22:30:07.803 : N8 M80 *51
22:30:07.803 : ok
22:30:07.803 : N9 M220 S100 *72
22:30:07.819 : ok
22:30:07.819 : N10 M221 S100 *113
22:30:07.819 : ok
[... 127 lines cut ...]
22:44:15.400 : set_axis_is_at_home 1 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
22:44:15.415 : > AFTER set_axis_is_at_home > current_position: (0.00, 0.00, 0.00)
22:44:15.415 : <<< homeaxis(1)
22:44:15.415 : > homeY: (0.00, 0.00, 0.00)
22:44:15.415 : >>> homeaxis(2)
22:44:15.415 : deploy_z_probe > current_position: (0.00, 0.00, 0.00)
22:44:15.712 : > line_to_destination(), destination: (0.00, 0.00, -252.00)
22:44:21.729 : > enable_endstops(false)
22:44:21.745 : > line_to_destination(), destination: (0.00, 0.00, 2.00)
22:44:23.761 : > enable_endstops(true)
22:44:23.761 : > line_to_destination(), destination: (0.00, 0.00, -4.00)
22:44:29.511 : > TRIGGER ENDSTOP > current_position: (0.00, 0.00, 0.00)
22:44:29.511 : set_axis_is_at_home 2 > (home_offset[axis]==0.00) > current_position: (0.00, 0.00, 0.00)
22:44:29.527 : > AFTER set_axis_is_at_home > current_position: (0.00, 0.00, 0.00)
22:44:29.527 : > SERVO_LEVELING > stow_z_probe
22:44:29.527 : stow_z_probe > current_position: (0.00, 0.00, 0.00)
22:44:29.527 : Raise Z (after) by 2.00
22:44:29.527 : > SERVO_ENDSTOPS > raise_z_after_probing()
22:44:29.527 : do_blocking_move_to: (0.00, 0.00, 2.00)
22:44:31.840 : <<< homeaxis(2)
22:44:31.856 : > (home_all_axis || homeZ) > final: (0.00, 0.00, 2.00)
22:44:31.856 : ENDSTOPS_ONLY_FOR_HOMING enable_endstops(false)
22:44:31.856 : <<< gcode_G28
22:44:31.856 : ok
```

AnHardt, 2016-01-16:
Sorry. Now with the right branch and config. i can see the problem.
And i can also see your fix works. :-)

It does - but for the wrong reasons.
destination[Z_AXIS] is an absolute value - else we'd end up much higher.
The problem works this way:
Entering G28 we set
set_destination_to_current(); //what is all 0 directly after reset
destination[Z_AXIS] = -Z_RAISE_BEFORE_HOMING * home_dir(Z_AXIS);
line_to_destination(); // do the move;
// now current_position[Z_AXIS] is 0 and destination[Z_AXIS] is  -Z_RAISE_BEFORE_HOMING * home_dir(Z_AXIS)
st_synchronize(); // Wait until the move arrived

next is 
HOMEAXIS(X);
what is 
homeaxis(X_AXIS)
Here we set
current_position[X_AXIS] = 0;
and call
sync_plan_position();
what is
plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
Basically we tell the the stepper coordinate system here, where we are - based on current_position, what is not the same at this point, as destination.
Now our zero level is above our start point 
The next 
line_to_destination();
wants to go to the same height, but now sees a difference in the stepper coordinate system and rises.

You try to avoid the rise by setting
destination[Z_AXIS] = current_position[Z_AXIS]; // 0!

I mean the better way would be to say
current_position[Z_AXIS] = destination[Z_AXIS]; //  -Z_RAISE_BEFORE_HOMING * home_dir(Z_AXIS);
at the same place as you did. That would not touch the level. 

Anyway both ways are working because we finally home the z-axis.

If we, one day, want to home x or y only, with a minimal height of the z-axis, it will make a difference!

AnHardt, 2016-01-16:
As far as i know the end position of G28 is undefined.
For me that means - make no attempt to go anywhere special - make no assumption where the tool is. 
After the next G0/G1 we are where we want to be.

The problem is to sync Marlin and RepetierHost. Directly after a G28 Repetier does not know exactly where we ended. Because Repetier uses relative moves and respects its own sotware endstops it is sometimes a bit difficult to get both systems in sync (without a absolute move).

The solution would be to send the the actual endposition to Repetier wit the output of M114.
```
gcode_M114();
```
at the end of G28, G29, M48, or any other routine where we do not know where we end, solves the problem.

lrpirlet, 2016-01-18:
@AnHardt 
Thanks, I learned something.
Before I propose a better commit, and to check for comprehension, could I replace my inappropriate line ```destination[Z_AXIS] = current_position[Z_AXIS]; ``` with ```destination[Z_AXIS] = 0;```?
I feel that this would be more coherent with the rest of the code in G28... If we raise blindly Z by Z_RAISE_BEFORE_HOMING  while ```axis_known_position``` is still false, I do not see the need to record the position...
Also, this is what is done around line 2450 in the code run #if ENABLED(Z_SAFE_HOMING) that will (home_all_axis)...

AnHardt, 2016-01-18:
> with destination[Z_AXIS] = 0;?
would not move a second/third time upwards. But
'''
plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], 
'''
Would still shift the stepper coordinate system.

My code at the moment looks like
```
    #if Z_HOME_DIR > 0  // If homing away from BED do Z first

      if (home_all_axis || homeZ) {
        HOMEAXIS(Z);
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (marlin_debug_flags & DEBUG_LEVELING) {
            print_xyz("> HOMEAXIS(Z) > current_position", current_position);
          }
        #endif
       }

     #elif defined(Z_RAISE_BEFORE_HOMING) && Z_RAISE_BEFORE_HOMING > 0
    
      if (home_all_axis || homeX || homeY) {  // Raise if homing not only z alone downwards
        destination[Z_AXIS] = Z_RAISE_BEFORE_HOMING;
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (marlin_debug_flags & DEBUG_LEVELING) {
            SERIAL_ECHOPAIR("Raise Z (before homing) by ", (float)Z_RAISE_BEFORE_HOMING);
            SERIAL_EOL;
            print_xyz("> (home_all_axis || homeX || homeY) > destination", destination);
          }
        #endif
        feedrate = max_feedrate[Z_AXIS] * 60;
        line_to_destination();
        st_synchronize();
        current_position[Z_AXIS] = destination[Z_AXIS];
      }
    #endif
```

That nicely lifts for x or y only homings too.

thinkyhead, 2016-02-07:
How is this issue looking at this time? Is there a PR needed, or can we close this?

lrpirlet, 2016-02-08:
@thinkyhead. Thanks, this thread can be closed, there is a PR submitted against RCBugFix under #2899

closed 2016-02-08 by lrpirlet