MarlinFirmware/Marlin #2888: auto bed leveling using ramps 1.4

hadisalame, 2016-01-04:
Hi , I set auto bed leveling feature for my 3D printer but after that the Z axis started to have a problem , I think it's from the firmware ... the problem is ( when I raise it up some times the two motors go up together while other times they go in different directions ) Could you please help me ... it's a tough problem for me !
Do you want my firmware to send it to you using email ?
Thank you .

thinkyhead, 2016-03-11:
Hi @hadisalame. Sorry this issue has been sitting around for so long without a reply. Apparently while I was away there was a shortage of good help.

Is this still an issue for you, and have you tried the latest `RCBugFix` to see if the problem still exists? We're trying to get a new release together, so it would be good to know if this is still a problem.

hadisalame, 2016-03-11:
Thanks bro. for your kind reply, I fixed that problem since two weeks ago, and it is working properly now except a little problem, which is that before I start printing I should lift the Z-axis manually to avoid damaging the servo and I can't write it in the starting G-code because it maybe finished the latest shape at the maximum Z height and it will damage the printer !!!
Thanks again for your interest .

closed 2016-03-13 by thinkyhead