MarlinFirmware/Marlin #2890: Serial communication issue (echo:Unknown command: "echo")

alexrj, 2016-01-05:
I observed a weird issue in serial communication, and I'd like to have your feedback about it.
(Using Marlin da095d8a07c39397c7eedcbafb27f1714bdeb889)

```
>> N11586 M105*44
<< ok T:46.3 /275.0 B:98.9 /100.0 T0:46.3 /275.0 @:40 B@:127
>> N11587 M105*45
<< ok T:46.3 /275.0 B:98.9 /100.0 T0:46.3 /275.0 @:40 B@:127
>> N11588 M105*34
<< ok T:46.7 /275.0 B:98.9 /100.0 T0:46.7 /275.0 @:40 B@:127
>> N11589 G29*22
<< Bed x: 18.00 y: 0.00 z: 4.01
<< Bed x: 173.00 y: 0.00 z: 5.96
<< Bed x: 173.00 y: 157.00 z: 6.18
<< Bed x: 18.00 y: 157.00 z: 4.19
<< Eqn coefficients: a: 0.01 b: 0.00 d: 3.77
<< planeNormal x: -0.01 y: -0.00 z: 1.00
<< ok
>> N11590 M105*43
<< echo:endstops hit:  Z:4.19
<< echo:Unknown command: "echo"
<< ok
>> N11591 M105*42
<< Error:Line Number is not Last Line Number+1, Last Line: 11589
<< Resend: 11590
<< ok
```

Of course, client is coded so that no new line is sent before firmware sends `ok`.
Looks like Marlin sends `echo:endstops hit:  Z:4.19` and for some reason that line is re-injected into Marlin which refuses it. I see three possible causes for this:
1. Client has some kind of buffer overflow and mixes read/write buffers, sending wrong data to Marlin.
2. Marlin has a buffer bug, mixing write/read buffers.
3. USB/serial communication error.

Option 3 looks possible. What do you think? Have you seen this kind of issues?

I think that for robustness clients should drop the assumption that Marlin can't request lines to be resent after they've been acknowledged with an `ok`. So they should probably keep the last 2 (or more) sent lines since Marlin may request them.

Wackerbarth, 2016-01-05:
(3) Is a very distinct possibility.

We need to know a lot more about your hardware and how the firmware was compiled.

alexrj, 2016-01-05:
Hardware is a RAMBo board attached via USB to a MacOS X 10.6 host, firmware was compiled with the Arduino IDE 1.0.6. Does this ring any bell?

a4jp-com, 2016-01-05:
I'm also using a RAMBo board with some new code from @Wackerbarth and the Marlin team. 

@alexrj if you like I can try the file you have and tell you if I get the same problem on a Windows 10 machine.

alexrj, 2016-01-05:
Nah, it's not file dependent.

I think that most host software keep the entire line history in memory, so they are able to resend any old line, so issues like this are handled gracefully. Still I really wonder about what the root cause could be...

a4jp-com, 2016-01-05:
Okay. 

I've been printing with a newer version of Marlin and everything is going very well now. 

I don't get any problems like the one you are having.

I just get a small blob at the end of prints if I let Slic3r get the retraction data from Marlin. 

Also after the flow-rate/feed-rate numbers on the display are tweaked we might be in for a new version or RC4.

a4jp-com, 2016-01-05:
PS @alexrj thanks for all the work you did in Slic3r.

alexrj, 2016-01-19:
Another interesting excerpt:

```
>> N19723 G1 X57.857 Y88.711 E16.75779*95
<< ok
>> N19724 G1 X57.739 Y89.006 E16.76426*85
<< ok
>> N19725 G1 Z1.750 F7800.000*0
<< ok
>> N19726 G1 E12.76426 F1800.00000*43
<< ok
>> N19727 G92 E0*77
<< echo:Unknown command: ""
<< ok
>> N19728 G1 Z2.250 F7800.000*11
<< Error:Line Number is not Last Line Number+1, Last Line: 19725
<< Resend: 19726
>> N19726 G1 E12.76426 F1800.00000*43
<< ok
>> N19727 G92 E0*77
<< echo:Unknown command: ""
<< ok
>> N19728 G1 Z2.250 F7800.000*11
<< Error:Line Number is not Last Line Number+1, Last Line: 19725
<< Resend: 19726
>> N19726 G1 E12.76426 F1800.00000*43
<< ok
>> N19727 G92 E0*77
<< echo:Unknown command: ""
<< ok
>> N19728 G1 Z2.250 F7800.000*11
<< Error:Line Number is not Last Line Number+1, Last Line: 19725
<< Resend: 19726
>> N19726 G1 E12.76426 F1800.00000*43
<< ok
>> N19727 G92 E0*77
<< echo:Unknown command: ""
<< ok
>> N19728 G1 Z2.250 F7800.000*11
<< Error:Line Number is not Last Line Number+1, Last Line: 19725
<< Resend: 19726
```

alexrj, 2016-01-19:
Marlin devs, what do you think about adding an M-code for enabling a "strict serial mode"?
In that mode, Marlin would discard/reject any line which doesn't contain the `N` line number and the checksum. I think this would make the protocol much more robust in case of interferences...

thinkyhead, 2016-02-07:
We might try it preliminarily as an M115 bitflag. @AnHardt is tracking some code issues that may be contributing, and I'm advising we revisit the conversation that precipitated the introduction of the `ADVANCED_OK` feature, which is designed to address communication issues as well.

thinkyhead, 2016-03-26:
We still haven't done "strict" serial mode. Perhaps it's worth a revisit now that we're coming to another stable release. The Feature Request should probably go on the MarlinDev repository at this point, as we're freezing features for this release.

closed 2016-03-26 by thinkyhead