MarlinFirmware/Marlin #2891: Printing From SD Card when printer is turned On

Leed3, 2016-01-05:
Hi,
I have a Robo3d R1Plus. It ships with an SD card (no LCD Controller). I had the impression that if you create a file named auto0.g and placed it on an SD card, you could put it in the printer and turn the printer on (with no USB cable) and the printer would start to print. Is this correct?

 The only way I can get it to print is to have the card in, turn on the printer, then connect the USB cable and then start MatterControl.

 Is there a way to make the printer print from the SD card on a cold boot with no USB cable?

brainscan, 2016-01-05:
Does your board still have power without the usb? My set up needs the board connected to usb while printing from sd as I haven't installed the diode to have my PSU power everything.

Leed3, 2016-01-05:
Thank you for your reply. That is a very good question. I believe so. When I turn on the power supply with no cable attached, the lights come on the board. If I connect the USB cable but do not start Matter Control then the printer does not start. It is only after starting Matter Control and Matter Control shows connected (no print jobs in the Queue).

Hopes that helps,

Lee

brainscan, 2016-01-05:
I've never tried to use auto0.g but I'll give it a try when I get home and see if I can replicate what's happening. 
Maybe someone will jump in with a suggestion in the mean time. 

Lee

Leed3, 2016-01-05:
Here is a link to Cura talking about printing from the SD card. https://printrbot.dozuki.com/Guide/How+to+Auto-Start+a+Print+with+an+SD+Card+using+Cura/201 

This also shows how to print with auto0.g.  Note that the code is supposed to print files on the SD card with names auto0.g through auto9.g in order.

ntoff, 2016-01-09:
I use auto0.g to set up my steps/mm and PID settings, it just automatically runs whenever the printer gets turned on for me (though I do have an LCD screen).

Leed3, 2016-01-09:
Thank you for checking that. I am beginning to think that the only way auto0.g will work is if the LCD screen is attached.  I am not sure why.

atunguyd, 2016-01-20:
In cardreader.cpp the code is there so it should work however calls to it are wrapped around the SDSUPPORT definition so you need to ensure that SDSUPPORT is enabled in your configuration files.

thinkyhead, 2016-02-07:
While an LCD controller is not needed for auto#.g to work, if you have an LCD controller you can use the `MENU_ADDAUTOSTART` option to add a menu command that will run the auto0.g, auto1.g, auto2.g whenever you want.

closed 2016-02-07 by thinkyhead