MarlinFirmware/Marlin #2894: Does Marlin works for PortabeeGo printer?

pauloguerraf, 2016-01-07:
Any tips on how to use Marlin firmware on a Portabee Go printer would be greatly apprecitated. Thanks in advance!

ntoff, 2016-01-09:
scroll to the bottom of the documentation http://portabee3dprinter.com/support/portabee-kit-docs/

pauloguerraf, 2016-01-09:
Thank you! I have seen the documentation before, but was not sure if the PortabeeGO uses the same board as the Portabee 3D Printer Kit. Can both printers use the same firmware or is there any configuration detail that I should take into account?

pauloguerraf, 2016-01-09:
Thank you! I have seen the documentation before, but was not sure if the PortabeeGO uses the same board as the Portabee 3D Printer Kit. Can both printers use the same firmware or is there any configuration detail that I should take into account? Once again, thank you for your help.

thinkyhead, 2016-03-11:
Is the latest version of Marlin any more helpful with the PortabeeGo? I know we don't support this machine in any formal way, but maybe it's easier to set up now. Would it help if we integrated the changes proposed at http://portabee3dprinter.com/support/portabee-kit-docs/#k8 ?

closed 2016-05-04 by jbrazio