MarlinFirmware/Marlin #2895: I2C feature request

sh4jonny, 2016-01-07:
Hi everyone. Sending full command M150 <command> to I2C module it would be fine. also it would be desirable to receive the answer. thank you.

Grogyan, 2016-01-08:
Is there full I2C support? And by extension 1-wire?

1-wire especially, as the max31850, as an example, utilizes that protocol.
And allows for multiple devices to be chained together.
Many times where I see I2C is used, there is also a 1-wire version as well.

One of the limitations of RepRaps and 3d printers is the number of wires
going from an extruder to a central control board, and with I2C the number
of wires can be reduced.

Just my 2¢

sh4jonny, 2016-01-08:
I told about possibility of a parcel of all commands for BlinkM. Not only M150 r0 u0 b0. And about possibility of reception of the answer from the module.

sh4jonny, 2016-01-20:
any one?

DustinWoods, 2016-01-21:
You could expand M150 to talk to any i2c device.

Perhaps the parameters for M150 could be restructured:
* If _I_(new M150 parameter) is not defined, then set blinkm with standard R/U/B parameters
* If _I_ is defined, then:
Use _I_ as the I2C address, and split _current_command_args_ at each space and send each over Wire.write.

thinkyhead, 2016-02-07:
I'm not sure how you would begin to implement what you need, but take a look at the [source code of digipot_mcp4451.cpp](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/digipot_mcp4451.cpp) for the implementation of the `DIGIPOT_I2C` feature, which does a simple send, but no receive. Presumably you can use the same technique to talk to any I2C device.

jbrazio, 2016-02-28:
@sh4jonny what is your expectation ? When the device responds back on the I2C bus how should Marlin react ?

thinkyhead, 2016-03-11:
I think this might be too general a feature to include in Marlin. We need some use-cases to explain what it's good for. Does someone want to cover every inch of their RepRap in NanoPixel pushbuttons?

Grogyan, 2016-03-11:
@thinkyhead I'm already thinking about having a go at getting I2C thermocouple cold junction compensators working.  Which would enable the SPI port for use with the SD card only!
Other suggestion are to get mini external display remotes using this, which would then enable people to make/buy ethernet cables, instead of unshielded cables.  Unshielded cables for SPI makes me cringe.
Ultimachine has the right idea of using digipots for motor driver current management, an I2C interface would be better.

jbrazio, 2016-03-11:
That could be a good use for TWI, but what thinkyhead meant was that the initial request was too generic. If the plan is to implement a particular device then it should be feasible..

cabbagecreek, 2016-03-18:
You where searching for a Use-Case.. I have one
I have developed a Filament Sensor tha count the filament used and the filament diameter.
Because I dont want to add more logic to The Marlin Software I have implemented the Filament sensor as a "smart sensor". It off-load the Marlin processor to do what its supposed to do: Print

I have made 2 interfaces to the sensor: I2C and Rs232/USB. 
Sound like I have done alot of work.. ;-) I have not. The sensor is based on a standard Arduino so the I2C and Rs232 is a no-brainer.

The arduino is setup as a I2C slave. The Marlin will be its Master. (Wow!! Impressive)
he marlin needs to send commands like 
M151 adress bytes ( M151 = Send bytes to I2C address)
The Filament sensor prepares the data back to Marlin
M152 adress number-of-bytes (M152 = Receive number-of-bytes from I2C address)
When Marlin receives the bytes it reponds to the Serial port
SERIAL_PROTOCOLLN(MSG_M152_REPORT); and the received bytes

Then its up to the GCODE sender to read and do actions on the Received bytes. Marlin actually dont do any action on the received I2C bytes

The functions that the Filament sensor has that has no responce is (= Only M151 needed):
* Reset the Filamentroll counter = I have inserted a new filament roll.
* Reset the Print counter = I have started a new print
* Do printer "Kill signal" = Somthing is wrong! = Filament sensor connected to Kill-Pin or Reset

The functions that the Filament sensor has that has responce is (=Needs additional M152):
* Show Filament counter = What is left on the filament roll.
* Show Last print Counter = Compare the actual filament usage compared with slicer prediction
* Show Filament diameter 
* Show "Out-Of-Filament" = Reason to do M151 Kill

User Scenario:
The user wants to calibrate the extruder.
1. Heats up the hotend
2. Reset the Print counter 
3. Extrude 100mm
4. Show Last print Counter
5. Compare Actual Filamentlengt with 100mm
- Adjust the Firmware (configuration.h)
- Adjust "Extrusion multiplier" or similar in GCODE softwre.
6. Redo 2-5 until perfection.

Future scenario:
* Marlin wants to now actual filament length and diameter for near real-time correction of feedrate
* Marlin checks "Out-Of-Filament" in a timed manner. GCODE Software inserts M151 commands. 
* Marlin checks "Out-Of-Filament" by itsself and do Kill
The data that the Filament sensor responce back is:

cabbagecreek, 2016-03-18:
With this new M151 the M150 could be obsolite.
Sending M151 with adress to the BlinkM and RGB data as 3 bytes is the same as the M150

Roxy-3DPrintBoard, 2016-03-18:
I understand the motivation for keeping this simple.   But I would actually put the support logic for the filament change in Marlin's firmware just because it will be too hard and take too long to get all the hosts to accommodate it.  

And rather than add a new M151 command that pretty much replaces the existing M150, why not make the existing M151 a little smarter?   If you don't send (or add) any parameters to the M150 command, it could behave as it does now.  If you do add parameters, it behaves like your new M151 command?  

Can you publish the schematics for your filament sensor?  This is a really interesting approach!

Nandox7, 2016-03-18:
This would be handy for some features like the Filament sensor.
No longer needed to convert from I2c to digital and later to analog.

-Nando

sh4jonny, 2016-03-18:
hi! where i can read about M151 ?

cabbagecreek, 2016-03-19:
The M151 is only a suggestion.
Schematics fr the filament sensor is an arduino 5 wires and a old mouse containing a ADNS2050E chip.
The 5 wires are GND, 5V, SDIO, SLCK, And NegativeEnable.
The arduino uses wire library and act as a slave. RS232 is the normal Serial port

Commands can be sent from both I2C and Serial. The syntax for control that you first send a command and then reads the value for the previous command.

This makes it possible to calibrate the sensor from a PC and also connect it to the Marlin Firmware. I have made a program called 3DprinterTools that uses the filament sensor to calibrate the extruder feedrate. The program can both update the firmware by using Arduino IDE in background or with values for "Extrution Mulitpier"(=Simplify3d name) for GCODE software..

jbrazio, 2016-03-20:
@cabbagecreek I understand where you want to go and it seems a nice step forward for spool/filament handling but this is not a trivial change, in fact only the handling of how much filament is left in the spool is not an easy, trivial, task.

Nevertheless this feature set is something that, if implemented, should not be directly tied with the I2C bus.

cabbagecreek, 2016-03-22:
There is no way that you can predict the length of filament on a spool. There is a way that you can weight a print and with the Filament sensor value for filament usage do some calculation to predict how much filament there is on the spool. 1000gr = ( last print filament usage / last print weight) * length of filament on spool.

All the calculation is done within the FilamentSensor/Arduino so the Marlin firmware will not know or take any "RAM-usable-for-printing" space.

The function for I2C Send is almost implemented in digipot_mcp4451.cpp

>  static void i2c_send(byte addr, byte a, byte b)
> {
> 	Wire.beginTransmission(addr);
>     Wire.write(a);
>     Wire.write(b);
>     Wire.endTransmission();
> }

Creating a I2C.cpp with
>  static void i2c_send(byte addr, byte [] data)
> {
> 	Wire.beginTransmission(addr);
>     TODO: Loop thru byte[] and do Wire.write(for-each-byte-in-byte-array)
>    
>     Wire.endTransmission();
> }

The current digipot_mcp4451.cpp uses the I2C.cpp and sends only 2 bytes = byte a and byte b and that will reduce the code in digipot_mcp4451.cpp and make it possible for M151 to use the same code/ !2C.cpp class.

During the day I will look at a possibility to ReadFromI2C as a generic function. I have not found any current function that reads byte(s) from I2C.

I need someone to help me with coding. Its very complicated to try to reverse-enginer functions and dependencies. This is my first addition to Marlin so I will be grateful for all help I can get ;-)

cabbagecreek, 2016-03-22:
The other part that I have not done is to create a Syntax for the M151/M152 parameters.
To make it useful for both use of parameters in decimal form and Hex
A suggestion is  
M151 A 8 D 065066067  (Send 3 bytes A B C in Decimal  to I2C decimal address 8)
and/or
M151 A 8 H 0x410x420x43  (Send 3 bytes A B C in Hex  to I2C decimal address 8)

Is there some other GCODE that sends bytes/strings to Marlin Firmware??

Roxy-3DPrintBoard, 2016-03-22:
> M151 A 8 H 0x410x420x43

This is going to be problematic.   You can't have letters embedded in the numeric part of a parameter.   You are going to have to settle for something like:

M151 A 8 H 01410042043

jbrazio, 2016-03-22:
@cabbagecreek 

We do all agree Marlin will be the BUS master, so there are two functions we must implement:
 1. Write to a slave and expect nothing back
 2. Write a request to a slave and expect X bytes response

It's important to note that slave-to-master initiated communication shall not happen, slave replies must only come if the mater request them.

The only TWI implementation on Marlin is for the BLINKM which uses the first scenario with `M150`. I would suggest to change the proposed `M151` to `M155` to leave more free space between functionality (we do not know if people in the future want more blinking stuff). `M155` would be the equivalent to **1.** and `M156` the equivalent to **2.**.

With a custom parser for `current_command_args` we could have hex on the message:
`M155 D<two byte address> S<size of the message> M<message>`
(**D**estination, **S**ize and **M**essage)

As we should keep the `message` under the "ASCII visible range" in order to stay *G-CODE compatible* so we would require each byte of the message to be "zero padded two char" i.e. "4142430D" for "ABC\n" (0x41, 0x42, 0x43, 0x**0**D)

The same could be applied to `M156`:
`M156 D<two byte address> S<number of bytes requested>`
(**D**estination and **S**ize)

> I need someone to help me with coding

I'll try to mock up something dirty tonight.

cabbagecreek, 2016-03-22:
Marvelous.. 
[image]
[image]
Yesterday I was skiing in Verbier and today its my birthday.. so I will not be able to do any work today. Tomorrow I belive I will start my Estern-holliday and have about 3-4 days that I can work on this idea. I will try to make a wiki page about the work I have done. My idea is that everything should be alble to DIY and the filament sensor will be totaly open source. The filament sensor is a reinvented mouse that "feels" the filament move and size. I used a very old Logitech mouse that gives 1000dpi.

cabbagecreek, 2016-03-23:
+jbrazio made wonderful suggestions.. I really like them

I have looked at jbrazio suggestion for the M15x syntax. I belive the data bytes needs to be 3 digits long to cover the range from 0-256.

If the Data part of the syntax contains Data In Decimal format, The argument D feels right. (D for Data and Decimal). Data can be both a command and a message. Data is more generic
* Is the adress 2 bytes? I belived it was in the 1 byte range (0-256)
* The first argument is Address.. Could not the Argument A feel more natural?
* Do the size of the message be needed? You will have to loop thru each byte anyway.
* Do the Data/Message need to be null-terminated?

> M155 A<two byte address> D<message>

example M155 A 8 D 041042043   => Sends ABC to I2C at address 8

A = 041 decimal
B= 042 decimal
C= 043 decimal

On the receiver the syntax could be changed to

> M156 A<two byte address> N<number of bytes requested>

A= Address and N=Number of byte

The recieved bytes are relayed to Serial

> SERIAL_PROTOCOLPGM("I2C data=");
> SERIAL_PROTOCOL( received I2C Bytes );
> 

Then the Sending GCODE software will have to handle the recived bytes. Nothing will be processed within Marlin.

Then printeing from SD the recived bytes still be sent to the Serial Port. (like the temp reports) 
If a Software is connected to the 3D printer the connected software can do someting with the received data. Otherwise the data is lost in space.

jbrazio, 2016-03-23:
> Is the adress 2 bytes? I belived it was in the 1 byte range (0-256)

Arduino's wire lib works only with 7bit addresses, thus the max address will be 127. The issue working with literal hex on the serial is that each byte you receive on serial will be "half-byte" of your constructed value, imagine you want to send the value 255 in hex on the serial, you'll need send something like FF taking "two serial bytes".

> The first argument is Address.. Could not the Argument A feel more natural?

Yes let's change it. I would also propose to change the Message (M) to Data (D) in order not to have on the same GCODE line multiple M codes.

> Do the size of the message be needed? You will have to loop thru each byte anyway.

You're right, but this can be our poors-man-checksum.

> Do the Data/Message need to be null-terminated?

No, raw data.

thinkyhead, 2016-03-23:
Because of the strictures of GCode, it may be best to send one byte per GCode command instead of trying to send them all on a single line. If there's worry that the bytes might not be sent rapidly enough (due to the short delay in receiving each GCode command) then it could work something like this:
```
M151 P1 ; Select and clear I2C send buffer number 1
M151 V1 ; Add a byte to the selected send buffer
M151 V103
M151 V0
M151 V6
M151 V11
M151 V86
M151 V42
M151 V125
M151 S1 P1 ; Send all bytes in I2C buffer number 1
M152 ; Wait for I2C reply, if needed (with timeout?)
```
The other advantage to this technique is that you can store your commands ahead of time, then invoke them many times with a single command.

jbrazio, 2016-03-23:
@thinkyhead when you speak about "strictures" it means GCODE standards do not allow a parameter to have a variable length or do you foreseen possible issues with the parser ?

Roxy-3DPrintBoard, 2016-03-23:
I like the approach ThinkyHead just suggested if we want to keep looking and feeling like GCode.  But I'm not sure we need the limitations that GCode brings.   Another path might be to declare a new command type besides G, M and T.    We could have the S(mart) command type that parsed parameters in a sane way.

jbrazio, 2016-03-23:
This is the current parser I have:

```
Printer is now online.

>>> M155 A3F L07 D6a6272617a696f
SENDING:M155 A3F L07 D6A6272617A696F
Address: 63
Len: 7
Data: jbrazio

>>> M155 A3F D6a6272617a696f
SENDING:M155 A3F D6A6272617A696F
Address: 63
Len: 0
Data: jbrazio

>>> M155 A3F D6a6272617a696f  L07
SENDING:M155 A3F D6A6272617A696F  L07
Address: 63
Len: 7
Data: jbrazio

>>> M155 D6a6272617a696f L07 A3F
SENDING:M155 D6A6272617A696F L07 A3F
Address: 0
Len: 0
Data: jbrazio
```

Marlin is parsing and responding with serial echo the `M155` command. As you can see position of the args is irrelevant (and also the Len arg).

Roxy-3DPrintBoard, 2016-03-23:
And obviously...  Instead of defining a new command type that gets parsed differently.  We could just let each command decide which parsing method it wants to use.   (Which what you are doing here)   

This is creeping elegance:   But it would be nice to be able to parse quoted strings such as D 'jbrazio' and longer parameter names such as -Data 'jbrazio'

cabbagecreek, 2016-03-23:
I will concentrate next 2 days on the filament sensor. I will publish my testprogram (3DprinterTool) on github. My solidworks files and stl for the filament guide made for a logitech mouse. I will try to find the modelnr.  
My test will be concentrated to the Arduino Software and calibration of the sensor data. 
I will be using RS232 interface to the Arduino. 
I want to do more test to find out how accurate the sensor is and if the accuarncy changes with different filamenttypes.

My goal is to have the filament sensor accurate close to what a steelruler will get ( within 0.5mm)

cabbagecreek, 2016-03-23:
+jbrazio 
Marvelous work done.. 
The M155 looks like it could work for the Filament sensor
I will just use 1 byte to do the commands

Next step is to do the M156 command

I will try to do a mockup of the Arduino so a PC can act as a Filament Sensor and do the responce thru the Arduino. I am a C# programmer so the program will be for Microsoft OS.

Then anyone with a arduino and a PC cant help with the I2C interfacing and it will be a perfect testing platform..
I publish the Arduino code for the FilamentSensorMockup on GitHub with some basic fritzing pictures

jbrazio, 2016-03-23:
@Roxy-3DPrintBoard 

> Another path might be to declare a new command type besides G, M and T

Aren't we deviating from the standard.. how far are we willing to go ?

> would be nice to be able to parse quoted strings such as D 'jbrazio'

Instead of hex data ?

@cabbagecreek 
I'm getting a bit confused, so in one side we have the I2C sensor thing, and on the serial side we need a standalone programe ? Or this is just for testing ?

What would be your ideia about how pieces interact in the final "product" ?

cabbagecreek, 2016-03-23:
This is how the ADNS2050Sensor is wired to the Arduino.
[image]

cabbagecreek, 2016-03-23:
The standalone C# software is just for testing. Its a bit difficult to do Unit-test on a working filamentsensor.

The other part of the C# software is to act as a filamentsensor-mockup. The PC dont have I2C so I use a Arduino as Pc-to-I2c interface. Then it will be possible to test the Marlin I2C communication thru the Arduino and show resuts on the PC GUI. The C# GUI will make it possible to create data to the Arduino so when Marlin is requesting data the arduino have some data to send.
This is a testing tool that can be used to test the filamentsensor without even having to buy/buid one.

I think I will have this up and running in a couple of hours

cabbagecreek, 2016-03-23:
I think I decribed a comment that no longer exist.. Anyhow I an working of  C# testing GUI.

jbrazio, 2016-03-23:
@Roxy-3DPrintBoard decided to speak with my avatar so edited my previous comment. :-P

But how will Marlin interact with the arduino-plugin/sensor ?

cabbagecreek, 2016-03-23:
Marlin just act as a command Relay.. 
Getting the M155 command from the Slicing Software, parse the command and send it to I2C
Getting the M156 command will receive data from I2C, The data is relayed to the outgoing Serial port back to the Slicing Software.

Marlin will not have any processing on the data to or from I2C.

thinkyhead, 2016-03-24:
As to the flavor of GCode that Marlin uses, it's almost entirely following the "de facto" standard set by its predecessor Sprinter. There have been some extensions to its behavior, specifically to allow commands like `M117 and `M0` to take a string argument, and to allow `M23` to take a filename argument. The only requirement we impose is that the string be the last argument. So, `M155` could conceivably follow this approach.

However, in that case I do not recommend using hexadecimal, because A-F may be interpreted as command letters in some cases. Our current command parser is a bit wasteful on cycles, but it allows us to be picky about which arguments we look for. But the GCode command parser in Matt Roberts' RepRap Firmware pre-scans for all arguments in advance, and follows a less picky approach. Command parsers in other firmware may face similar issues if we try to stretch GCode too far, then we face an uphill battle asking the community to adopt a divergent methodology.

I've run up against resistance before, even for seemingly innocuous changes. When I submitted the Mixing Extruder feature to MarlinDev I got flak for "hijacking" `A`-`D` and `G`-`H` parameters. So I looked at how Repetier Host implements mixing extruders in GCode and they follow a method like the one I recommended above. So I added their method to Marlin (keeping the Pià Taubert method for reference). They do this out of respect for the existing "standard" and don't presume to expand it on everyone's behalf just to fit one feature.

Part of the reason for keeping it simpler and avoiding new special cases is precisely because GCode is intended to be very simple to parse, so simple that even a modest embedded processor can parse it ("with no storage" as they say). Thus GCode is meant to be parsed in character order, and preferably in advance of calling the code handler. Marlin and Sprinter are interesting in that they defer looking for command arguments until the handler. But as I point out, some firmware is probably using a "reference implementation" with minor changes. And those are quite conservative.

Hence, the reason I propose the solution I described is technical, not philosophical. It will be maximally compatible with existing parsers. If we do not truly need to extend GCode (and we don't for this feature) then we should avoid it.

jbrazio, 2016-03-24:
This is the latest test version.

```
>>> M155 A3F L07 D6a6272617a696f
SENDING:M155 A3F L07 D6A6272617A696F
Address: 63
Len: 7
Data: jbrazio
```

[image]

----

```
>>> M155 A3F L06 D4d61726c696e
SENDING:M155 A3F L06 D4D61726C696E
Address: 63
Len: 6
Data: Marlin
```
[image]

jbrazio, 2016-03-24:
I've implemented @thinkyhead suggestion for the command parser, here are the results.
You may check the code base [here](https://github.com/jbrazio/Marlin/tree/experimental/twi-implementation), latest changeset is jbrazio/Marlin@a16cc3aa755a5b1f4ffaa495a01c54126c4e8594.
The `twibus::foo stuff` are just debug lines.

```
; "t1" test macro
; This macro send the string "jbrazio"
; It's a one liner M155 with multiple B<base 10> args
M155 A63                              ; Target slave address
M155 B106 B98 B114 B97 B122 B105 B111 ; j b r a z i o
M155 S1                               ; Send the current buffer
```

Serial output:
```
>>> t1
SENDING:M155 A63
SENDING:M155 B106 B98 B114 B97 B122 B105 B111
SENDING:M155 S1
twibus::sendto: 63
twibus::addbyte: 106
twibus::addbyte: 98
twibus::addbyte: 114
twibus::addbyte: 97
twibus::addbyte: 122
twibus::addbyte: 105
twibus::addbyte: 111
twibus::send()
```

Scope (I2C) output:
[image]
---
```
; "t2" test macro
; This macro send the string "Marlin"
; It uses multiple M155 commands with one B<base 10> arg
M155 A63  ; Target slave address
M155 B77  ; M
M155 B97  ; a
M155 B114 ; r
M155 B108 ; l
M155 B105 ; i
M155 B110 ; n
M155 S1   ; Send the current buffer
```

Serial output:
```
>>> t2
SENDING:M155 A63
SENDING:M155 B77
SENDING:M155 B97
SENDING:M155 B114
SENDING:M155 B108
SENDING:M155 B105
SENDING:M155 B110
SENDING:M155 S1
twibus::sendto: 63
twibus::addbyte: 77
twibus::addbyte: 97
twibus::addbyte: 114
twibus::addbyte: 108
twibus::addbyte: 105
twibus::addbyte: 110
twibus::send()
```

Scope (I2C) output:
[image]

Roxy-3DPrintBoard, 2016-03-24:
Does the wrong thing happen if this syntax is used?

M155 B 106 B 98 B 114 B 97 B 122 B 105 B 111

And does this work?

M155 B106B98B114B97B122B105B111

The current GCode parser is OK with both of those variants...

jbrazio, 2016-03-24:
Seems to work flawlessly for both test conditions. I do not do any space check because the combination of `strtol()` and the `default` case statement does it for us.

**Serial output**
```
>>> M155 A63
SENDING:M155 A63
twibus::sendto: 63
>>> M155 B 106 B 98 B 114 B 97 B 122 B 105 B 111
SENDING:M155 B 106 B 98 B 114 B 97 B 122 B 105 B 111
twibus::addbyte: 106
twibus::addbyte: 98
twibus::addbyte: 114
twibus::addbyte: 97
twibus::addbyte: 122
twibus::addbyte: 105
twibus::addbyte: 111
>>> M155 S1
SENDING:M155 S1
twibus::send()
```

**Scope output**

[image]

-----

**Serial output**
```
>>> M155 A63
SENDING:M155 A63
twibus::sendto: 63
>>> M155 B106B98B114B97B122B105B111
SENDING:M155 B106B98B114B97B122B105B111
twibus::addbyte: 106
twibus::addbyte: 98
twibus::addbyte: 114
twibus::addbyte: 97
twibus::addbyte: 122
twibus::addbyte: 105
twibus::addbyte: 111
>>> M155 S1
SENDING:M155 S1
twibus::send()
```

**Scope output**

[image]

jbrazio, 2016-03-25:
Latest code base: 66db6d751f878268e9251be175205d38fa72b968

`M156` is now implemented, it allows to request `X bytes` from a I2C address, in the example bellow I'm from Marlin requesting `5 bytes` to the slave Arduino Nano board with the I2C address `63`.

```
>>> M156 A63 B5
SENDING:M156 A63 B5
twibus::sendto: 63
twibus::reqbytes(): 5
echo:I2C-reply: from:63 bytes:5 data:hello
```

You can see the relayed reply `echo:I2C-reply: from:63 bytes:5 data:hello`.

cabbagecreek, 2016-03-25:
I have been working on the testrig for the filamentsensor..  It reveals alot of secrets:
* The retraction of filament has a greater "slippage" than the extrusion.
*The filament is not moving on constant speed:
- Could be filamentsensor thets needs to be improved = more focus
-The sensor is missing movements of the filament.
* The accuracy of repeated test can be within +-0.25mm
This was my results at 04:05 

I am VERY impressed of the work done on the I2C communication.  @jbrazio 
Today I will have my sister kid to watch over.. I hope that I can make a Youtube video of how the filament sensor is working

jbrazio, 2016-03-25:
Great, let's see the outcome.

mkeyno, 2016-04-03:
dear All can you look at this [pull topic](https://github.com/MagoKimbra/MarlinKimbra/pull/113#issuecomment-204879376) for adding the laser to Marlin along with dynamically  Tools Height Control (THC) feature, I think this topic  with using I2C request from cutting module to set the tools in proper Height  during the printing is close to  implementation THC  feature 
really appreciated to have your comments or join it

jbrazio, 2016-04-17:
Will close this issue, the [experimental] TWI feature is now merged into RCBugFix and will be released with RC6.

closed 2016-04-17 by jbrazio