MarlinFirmware/Marlin #2896: Feature: Movable extruder plate for delta printer

fraternl, 2016-01-09:
My apologies beforehand if this is not the place to ask, but I wouldn't know where....

I have an idea, which either is not original or is not a good idea at all.

When I look at delta printers most of them don't have a direct drive, but use bowden. This is done to reduce the weight. When these printers are high this means long bowdens and this has a negative impact on the E-control.
I thought of a moving extruder-plate that would "hover" at a constant  height above the print head keeping the distance between the extruders and the printer more or less the same.
The bowdens can then be much shorter.

Has this been done before and is it a good idea?
Does Marlin already have a way to drive this extruder-plate?
Would the team be willing to make such an enhancement if it doesn't exist yet?

KiteLab, 2016-01-09:
How about to mount the extruder-motor on one of the already existent tower sliders? The distance to the effector would always be constant and no software extension is needed.

Wackerbarth, 2016-01-09:
That's an interesting concept. I'm not sure how you would accomplish this, mechanically, without a significant increase in the cost and size of the overall printer. Depending on the number of additional actuators involved, it might, or might not, be within the scope of the capabilities of a machine driven by this firmware. I think that we would need much more detail of the mechanism to determine how we might control it.

Wackerbarth, 2016-01-09:
@Kitelab -- The purpose of the Bowden Tube feed is to reduce the mass that moves with the head movement. Any "solution" that uses the existing actuators to move the extruder motor just defeats that aspect of the Bowden Tube design.

fraternl, 2016-01-09:
True... Thought of that too, but then the weight of the extruders would have an impact on the movement of the print head.
It would of course be less....

KiteLab, 2016-01-09:
You'd only need __one__ stronger stepper

fraternl, 2016-01-09:
Can I conclude that the idea is original (well, it is original as no-one told me)?
I mean.. hasn't it been done before?
The plate should just follow layer height and the offset to the printer could just be mechanically solved.
The Z-motor could be on the same plate as the extruder motors.

Wackerbarth, 2016-01-09:
"Stronger" actuators only marginally addresses the issue of inertia. You also have issues with the elasticity of the drive mechanism, etc. In general, I don't think that your approach would be successful.

fraternl, 2016-01-09:
@Wackerbarth   you are commenting on Kitelab, not on me, are you?

KiteLab, 2016-01-09:
Because of the 4-axis limit in Marlin a 4th tower is difficult.

Wackerbarth, 2016-01-09:
@fraternl -- You need to figure out how you would suspend this platform within the framework. Off hand, I don't see how you can do so without interfering with the head suspension.  (My first remark was addressed to your original proposal. The rest, prior to this one, were addressing @KiteLab)

fraternl, 2016-01-09:
The Z-height is there....

fraternl, 2016-01-09:
But I still don't have a confirmation if the concept of it is a good idea. I can read between the lines that you think it is. 
The implementation of it is only interesting if there is something to gain.

Wackerbarth, 2016-01-09:
@fraternl -- On the contrary, I have serious questions about the feasibility of the concept from a mechanical perspective -- totally ignoring whether the existing Marlin could be adapted to become its controller.

brainscan, 2016-01-09:
What about using mutley's flexi drive, light as a Bowden but still direct drive.

fraternl, 2016-01-09:
But that still is the implementation (this time mechanically)....
I also want some confirmation if there is something to gain in having a reasonably short bowden that goes almost vertically to the print head....

@brainscan I already ordered  a Flex3drive, but that also has its movement limits. And is not cheap if you want 3 filaments

KiteLab, 2016-01-09:
Theoretically a good idea, but if the afford/win ration can be good is questionable.and totally depends on the implementation.
A horizontal direction of the bowden tube is a bad idea because the material flow then has a 90° angle.

fraternl, 2016-01-09:
Maybe it should be made autonomous... completely outside of the print controller....

fraternl, 2016-01-09:
@KiteLab   Of course the distance should be that far that the angle never exceeds a certain limit....

KiteLab, 2016-01-09:
> Maybe it should be made autonomous... completely outside of the print controller....
Good luck while homing/probing/manual_z-moves.

fraternl, 2016-01-09:
Oh...  and the extruders could be in a gimbal always "looking" at the print head

KiteLab, 2016-01-09:
Keep win/afford in mind.

KiteLab, 2016-01-09:
For a delta system with 3 or more extruders it could make sense to move the bed instead of the effector. (seen at MakerFaire 2015 Honnover)
https://www.youtube.com/watch?v=zDhpmgzuWzA

fraternl, 2016-01-09:
That solves the extruder problem, but it does introduce a new one....
The object you're printing gets heavier and heavier.....
An Eiffeltower will probably print fine, but other objects can get problematic, me think....

It seems as someone thought of this before.... It's called a "flying extruder"
Someone posted this link after I suggested this on facebook.
I haven't read it yet:

http://www.fabbaloo.com/blog/2015/11/15/the-flying-extruder-the-best-of-bowden-and-direct-3d-printer-extruders

fraternl, 2016-01-09:
And a mechanical one exists as well....
http://imgur.com/0zNnNo7

thinkyhead, 2016-03-16:
> a "flying extruder"

I saw one of those at a user group meeting in Seattle. It works!

The idea of a moving extruder mount is interesting. It would be straightforward to add code for it if you needed it for a custom delta. But it's not going to end up as a standard feature in Marlin until there's actual hardware and more people are using it.

closed 2016-03-16 by thinkyhead