MarlinFirmware/Marlin #2897: 'G28 XYZ' not moving Z probe to bed for Z homing, G29 not working, Z_SAFE_HOMING configured.

ShaunDwyer, 2016-01-10:
Hi All,

I'm not sure if this is a bug of if Ive got something odd in my configuration thats screwing things up...
I've had a look at the code and added some debugging of my own and can't figure out whats going on.

I'm Running Marlin 1.1.0-RC3 

I've Just Added an IR Z probe that is offset from the nozzle: http://wiki.aus3d.com.au/IR_Z_Probe
I've got positioned infront of the nozzle. To make use of that, I've enabled auto bed levelling and Z_SAFE_HOMING: 

Relevant configuration:
#define AUTO_BED_LEVELING_FEATURE
#define Z_MIN_PROBE_REPEATABILITY_TEST
...
#define AUTO_BED_LEVELING_GRID

 #if ENABLED(AUTO_BED_LEVELING_GRID)

   #define LEFT_PROBE_BED_POSITION 40
   #define RIGHT_PROBE_BED_POSITION 140
   #define FRONT_PROBE_BED_POSITION 40
   #define BACK_PROBE_BED_POSITION 140

   #define MIN_PROBE_EDGE 10

   // Set the number of grid points per dimension.
   // You probably don't need more than 3 (squared=9).
   #define AUTO_BED_LEVELING_GRID_POINTS 2
...
#define X_PROBE_OFFSET_FROM_EXTRUDER -5
#define Y_PROBE_OFFSET_FROM_EXTRUDER -30
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.95

#define Z_RAISE_BEFORE_HOMING 4

#define XY_TRAVEL_SPEED 8000

#define Z_RAISE_BEFORE_PROBING 5
#define Z_RAISE_BETWEEN_PROBINGS 5
#define Z_RAISE_AFTER_PROBING 5
...
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)

  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).

#endif

Also have bed dimensions defined as:

#define X_MIN_POS 0
#define Y_MIN_POS 0
#define X_MAX_POS 180
#define Y_MAX_POS 180

Between each of the tests below I've tried with and without running M84 to forget the XYZ position.

Test1 - home XY together, then home Z seperately, probe offset check/abort works fine.
G28 XY    # works fine
G28 Z     # works fine, aborts and warns if probe XY offset puts probe out of bed area
G0 X5Y30  # Move to meet offsets for probe
G28 Z     # works fine

Test2 - home XYZ together - doesn't work as expected given Z_SAFE_HOMING, even though the code indicates it should.
G28 XYZ     # X homes, Y homes, position shown on the LCD "changes" to be X95 Y120.
looks to be the result of:
destination[X_AXIS] = round(Z_SAFE_HOMING_X_POINT - X_PROBE_OFFSET_FROM_EXTRUDER);
destination[Y_AXIS] = round(Z_SAFE_HOMING_Y_POINT - Y_PROBE_OFFSET_FROM_EXTRUDER);
NOTE: X+Y Carriage does not physically move
Z then starts to home with the probe outside the bed area. - at this point i put something under the sensor to stop it going down any further.
XY is physically 0,0, but LCD shows X95 Y120

I notice there are a few other threads on homing issues, but I couldn't find any that exactly fit my case.

I'm also having issues with G29 doing weird things, but I've not tested that as much yet or looked at the code.

the behaviour of G29 from a very quick test is:

G28 XY	# Home XY - works fine
G29		# Z starts homing at 0,0 (sensor outside of bed), 
XY carriages haven't moved, but LCD reads X145 Y170, but XY is physically at 0,0
I put something under the Z probe to stop it moving down outside the bed, and the Z motors make a GRRR noise like its backing of and going down a very short distance repeatedly

Bed Level correction matrix comes back in the console as:
Recv: Bed Level Correction Matrix:
Recv: +1.000000 +0.000000 -0.000000
Recv: +0.000000 +1.000000 +0.000000
Recv: +0.000000 -0.000000 +1.000000

Cheers!
-Shaun

KiteLab, 2016-01-10:
Please send a M119 to your printer and the result to us.
__No endstop shall be triggered when unused.__
If so reverse ?_???_ENDSTOP_INVERTING.

ShaunDwyer, 2016-01-10:
Hi KiteLab,

You're 100% right... this fixed my problem.

Before configuring as you say above:
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: x_max: TRIGGERED
Recv: y_min: open
Recv: y_max: TRIGGERED
Recv: z_min: open
Recv: z_max: TRIGGERED
Recv: ok

I then uncommented:
#define DISABLE_MAX_ENDSTOPS
which does the undef/define -1 all the max pins, and now:

Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: ok

Now G28 and G29 works as expected.

Thankyou, I really appreciate your help!

Cheers!
-Shaun

closed 2016-01-10 by ShaunDwyer

nkuck, 2016-06-04:
I am having maybe a similar issue. I can print fine and can do a G28 serial homing command. I have installed Marlin FW on a Prusa machine and unable to get the G29 code to respond. When I send an M119 I get: SENT:M119, RECEIVED:OK, RECEIVED: x_min:L, y_min:L, z_min:H.
I suspect that is has to do with the homing function disabled unless the nozzle is outside or inside the plater area. Can anyone suggest a problem? Thanks.

thinkyhead, 2016-06-05:
@nkuck What version of Marlin? We aren't patching older versions, so we recommend you update to `RCBugFix` if you're seeing any issues. That branch will be the 1.1.0 release, after we do a little more testing and patching.

nkuck, 2016-06-05:
Thanks for the info on the update/patch. I loaded it but see no difference. I will review the config file in the morning to make sure I haven't forgotten anything.

thinkyhead, 2016-06-05:
What kind of bed leveling are you using? We have `AUTO_BED_LEVELING_FEATURE` and `MESH_BED_LEVELING` available.

nkuck, 2016-06-05:
I have not installed the sensor yet. Waiting for it to arrive. Just trying to get Marlin up and running well, first.

nkuck, 2016-06-05:
I rechecked the config file and am seeing nothing that looks out of place. I'm probably not looking in the right section. One thing I discovered today: the endstops all show L when the bed is manually moved from the stops, but the y stop also reads L when activated. The switch checked out fine. I suspect that might be a reason, but do I need to invert that one? Seems like that might cause problems as the printer works fine otherwise. Thanks for any help.

nkuck, 2016-06-05:
I got the y endstop readjusted to work, but still unable to configure Marlin to allow G29 to work. Probably still doing something wrong.

nkuck, 2016-06-06:
I have AUTO_BED_LEVELING_FEATURE enabled.

thinkyhead, 2016-06-07:
If you send `G29` from Pronterface (or another host) does Marlin send back an error message?

nkuck, 2016-06-07:
No. Very confusing. In any host program, Repeater, Pronteface, Simplfy3D, I just get an “OK” response but no movement.

thinkyhead, 2016-06-07:
@nkuck How strange. If you successfully flashed Marlin onto your machine with `AUTO_BED_LEVELING` enabled, you should most definitely get some kind of response. Before `G28` it should send back an error that you need to home first. And if you have homed, then it should start trying to probe the bed.

If `G28` works, that implies that your endstops are correctly configured. But perhaps they are not. Do you get all the right endstop results when you test with `M119`?

nkuck, 2016-06-07:
I have found an occasional error with my y endstop after homing but not with the x or z. The y switch, for some reason, triggers but then backs away a little to make it read L after M119. Gotta figure that out. But I manually hold it enabled and still no G29 performance. I tried uploading Repetier and Marlin with no change. Totally stuck. 

Live simply!
You can only take with you what you have given.

nkuck, 2016-06-07:
Also, not sure that I ever successfully flashed the FW as the reset jumper is supposedly to be on, but the only way I can ever get it to say "upload done" is to leave the numper off. 

Live simply!
You can only take with you what you have given.

thinkyhead, 2016-06-08:
@nkuck You can tell whether you flashed successfully by looking at the output of `M115`. If the output matches the `STRING_CONFIG_H_AUTHOR` that you set in your `Configuration.h` then you can tell it flashed successfully.

nkuck, 2016-06-08:
Then, sadly, my suspicion was correct. I am not able to upload firmware to my printer.
The information does not change. What next???

thinkyhead, 2016-06-08:
> What next???

@nkuck Use the power of Google to research how to flash your particular board, find others who have encountered the same problem and solved it. Perhaps in a forum. Is your board a Melzi?

nkuck, 2016-06-08:
Yes, a Melzi board. I have posted on three or four forums. Nothing works yet. 

Live simply!
You can only take with you what you have given.

nkuck, 2016-06-08:
I've narrowed down the problem to not being able to upload the firmware to the Melzi board. It never happens, hence G29 is just "talking to the hand". But I am stumped about how to upload to the board. Tried everything posted.

Blue-Marlin, 2016-06-08:
Is Marlin growing too big, with G29, for your Melzi?

nkuck, 2016-06-08:
Not sure what you mean, but I was looking for some serious help on getting Marlin to load onto my Melzi board. 

Live simply!
You can only take with you what you have given.

thinkyhead, 2016-06-08:
What size does Arduino report when you build Marlin? The default configuration comes to under 60K. But with more options added, it can grow pretty big. @AnHardt is suggesting that your Melzi might only have 64K and the compiled Marlin binary might be too large for it.

nkuck, 2016-06-08:
Here is how it compiles:

Sketch uses 113,190 bytes (87%) of program storage space. Maximum is 130,048 bytes.
Global variables use 4,263 bytes (26%) of dynamic memory, leaving 12,121 bytes for local variables. Maximum is 16,384 bytes.

Under 60K. I have not been able to upload any firmware, not just Marlin.

thinkyhead, 2016-06-08:
Looks like it should fit!

thinkyhead, 2016-06-08:
Ha!

nkuck, 2016-06-09:
Is there any issue caused by the fact that the board is still physically installed within the printer? I've tried with the power pin set to USB and to Power. Does anything need to be disconnected? I'm about to buy a different board and start from scratch.

brainscan, 2016-06-09:
Do you get any sort of error message when you try to upload?

nkuck, 2016-06-09:
Depends if I leave the reset jumper in place. If installed, I get the sync error but the uploading message is still displayed and will stay that way forever. If I remove it, the upload finishes with no error but the actual upload didn't really happen. 

Live simply so others can simply live!

thinkyhead, 2016-06-09:
Arduino IDE has a sub-menu to select "Programmers" that I rarely touch. But have you tried the different options there to see if they do any differently?

nkuck, 2016-06-09:
I tried a couple but not all. 

Live simply!

nkuck, 2016-06-29:
I have tried all of the options but I'm still not able to load a different firmware to the Sanduino/Melzi board. Can you please provide more detailed directions. Could the bootloader be damaged? The printer still works fine. 

Live simply!
You can only take with you what you have given.

On Thu, Jun 9, 2016 at 7:55 PM -0400, "Noel Kuck" <noelkuck@gmail.com> wrote:

I tried a couple but not all. 

Live simply!

joecabezas, 2017-07-25:
maybe too late, for had same issue, and it happened that accidentally disconected the Z probe pins (black and white, BLTouch) from the arduino, then without realizing that, tried to home Z and failes with "G28 Z forbidden", then I realized the cable was unplug, plugged it back, same issue, then changed polarity and worked back to normal, thanks