MarlinFirmware/Marlin #2898: Question about the transformation of auto bed leveling

BelovedTutu, 2016-01-10:
Hello, I'm using the Marlin v1.0.2 with auto bed leveling. The command G29 and the auto bed leveling process have no problem, however when I'm printing a rectangular calibration cube, the vertical 'wall' of the cube incline toward the lowest point of the print bed, transforming the cube to a parallelepiped. It is as though even when the base of the print was did perfectly ( I have no issue printing the first layers and the last ones, they are all parallel to the print surface), the object was print vertically to the horizontal surface and not to the bed surface (which is tilted on purpose). Further simulation with the transformation algorithm in matlab give exactly the same issue.

Digging into the code of G29, I eventually arrived at this function in the file vector_3.cpp:
matrix_3x3 matrix_3x3::create_look_at(vector_3 target)
{
    vector_3 z_row = target.get_normal();
    vector_3 y_row = vector_3(0, 1, -target.y/target.z).get_normal();
    vector_3 x_row = vector_3::cross(y_row, z_row).get_normal();

    //x_row.debug("x_row");
    //y_row.debug("y_row");
    //z_row.debug("z_row");

 
     // create the matrix already correctly transposed
    matrix_3x3 rot = matrix_3x3::create_from_rows(x_row, y_row, z_row);

    rot.debug("rot");
    return rot;
}
This function is used by the G29 command to create the equation of the bed level plan after measuring. The problem is I don't understand the logic behind the line vector_3 y_row = vector_3(0, 1, -target.y/target.z).get_normal();. Why would we impose the elements of the y_row like that? Indeed it's perpendicular to z_row, but I just can't see why it's like that.
Thank you a lot for your help

AnHardt, 2016-01-11:
Leaving the math aside for now.
Is there a small possibility your printers coordinate system is not right handed?

BelovedTutu, 2016-01-11:
No, my system is not right handed

AnHardt, 2016-01-11:
Than that is the reason for correcting in the wrong direction.
It works at 0-level but than corrects x/y to the wrong direction.

BelovedTutu, 2016-01-11:
Thank you very much for your answer :) Now I feel like a chump for not noticing that :))

Any suggestion to how to change the correction to a left handed system?

KiteLab, 2016-01-12:
No, I would never do that nonsense.

Roxy-3DPrintBoard, 2016-01-12:
> Any suggestion to how to change the correction to a left handed system?

You can invert any axis in the Configuration.h file.   It doesn't matter where you put your origin.  Once you do, make sure moving in the +X and +Y direction moves away from the origin.   And of course, the same thing is true about Z.   If the nozzle is at home, moving any axis in the + direction should move it away from the origin.

BelovedTutu, 2016-01-12:
Thank you very much for all the helps :) Guess I'll try what you suggest :)

Good day to you all, :)

closed 2016-03-11 by thinkyhead