MarlinFirmware/Marlin #2901: X Probe Positions shifts by Y-Probe-Offset (V1.0.2 ABL)

cavemanatwork, 2016-01-12:
My Printer shifts all measurements on the x axis by the value of the Y-Probe-Offset.
The error is reproducible and stable.

// Travel limits after homing
#define X_MAX_POS 364
#define X_MIN_POS 0
#define Y_MAX_POS 312
#define Y_MIN_POS 0
#define Z_MAX_POS 275
#define Z_MIN_POS 0

// set the rectangle in which to probe
#define LEFT_PROBE_BED_POSITION 17
    #define RIGHT_PROBE_BED_POSITION 357
    #define BACK_PROBE_BED_POSITION 274
    #define FRONT_PROBE_BED_POSITION 1

#define X_PROBE_OFFSET_FROM_EXTRUDER 0
  #define Y_PROBE_OFFSET_FROM_EXTRUDER 29
  #define Z_PROBE_OFFSET_FROM_EXTRUDER 0

In this case the Printer starts the X Probe exactly at X=46 (17+29), Y=1.
The distances between the measurements match exactly, but just too far right.
The result is that the Printhead runs in the right endstop.

I played with the Y_Probe_Offset Values, the x Probe are always exactly shifted by the Y_Probe_Offset.

It's enough for me if you tell me what I need to change in the code.

Thanks a lot.
Best Regards
Michael

Here are my complete mechanical Settings:

//===========================================================================
//=============================Mechanical Settings===========================
//===========================================================================

// Uncomment the following line to enable CoreXY kinematics
// #define COREXY

// coarse Endstop Settings
#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#ifndef ENDSTOPPULLUPS
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  // #define ENDSTOPPULLUP_XMAX
  // #define ENDSTOPPULLUP_YMAX
  // #define ENDSTOPPULLUP_ZMAX
  // #define ENDSTOPPULLUP_XMIN
  // #define ENDSTOPPULLUP_YMIN
  // #define ENDSTOPPULLUP_ZMIN
#endif

#ifdef ENDSTOPPULLUPS
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
#endif

// The pullups are needed if you directly connect a mechanical endswitch between the signal and ground pins.
const bool X_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
[... 172 lines cut ...]
//#define MANUAL_Z_HOME_POS 402 // For delta: Distance between nozzle and print surface after homing.

//// MOVEMENT SETTINGS
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
#define HOMING_FEEDRATE {50*60, 50*60, 7*60, 5}  // set the homing speeds (mm/min)

// default settings

#define DEFAULT_AXIS_STEPS_PER_UNIT   {100,100,800,90}  // default steps per unit for Ultimaker
#define DEFAULT_MAX_FEEDRATE          {65, 65, 10, 8}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {900,900,200,90}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          700    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  700   // X, Y, Z and E max acceleration in mm/s^2 for retracts

// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
//#define EXTRUDER_OFFSET_X {0.0, -55.7} // (in mm) for each extruder, offset of the hotend on the X axis
//#define EXTRUDER_OFFSET_Y {0.0, 0.5}  // (in mm) for each extruder, offset of the hotend on the Y axis

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK                6.0    // (mm/sec)
#define DEFAULT_ZJERK                 2.0     // (mm/sec)
#define DEFAULT_EJERK                 20.0    // (mm/sec)

cavemanatwork, 2016-01-12:
Supplement:

At first i'm doing a G28 then a G29.
After the G28 the Display show me the right Position of the Nozzle.
When i start the G29, the Display show me for the first Probe Position the X Value "17", but that's not true, in reality the Nozzle ist exactly at X 46.
This is exactly LEFT_PROBE_BED_POSITION + Y_PROBE_OFFSET_FROM_EXTRUDER

thinkyhead, 2016-03-11:
Hi @cavemanatwork, sorry this issue has been lingering for so long without any reply.

If you haven't already moved on, please try the latest `RCBugFix` and see if you still have any issues. You will need to update your configurations. Since the config files have been altered a lot since 1.0.2, it may be simplest to start with the clean configurations and apply your changes.

closed 2016-04-27 by jbrazio